Koenkk/zigbee2mqtt #10685: [Problem]: Legrand devices with squares or \u0000

Tonio16, 2022-01-09:
### What happened?

In th description of Legrand devices there are squares or \u0000 for device name, maufacturer name et firmware date.

`{
    "power": 13,
    "state": "ON",
    "device": {
        "applicationVersion": 0,
        "dateCode": " \u000020200827\u0000\u0000\u0000\u0000\u0000",
        "friendlyName": "Télérupteur ch parents",
        "hardwareVersion": 2,
        "ieeeAddr": "0x00047400008384e4",
        "manufacturerID": 4129,
        "manufacturerName": " Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000",
        "model": "FC80RC",
        "networkAddress": 46288,
        "powerSource": "Mains (single phase)",
        "softwareBuildID": "0025",
        "stackVersion": 66,
        "type": "Router",
        "zclVersion": 2
    },
    "linkquality": 178
}`

### What did you expect to happen?

I expected to havn't them.

### How to reproduce it (minimal and precise)

To pair some Legrand devices. I have several of them, either routers or enddevices, and all them have the same behavior.

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20211217

### Adapter

Sonoff CC2652P

### Debug log

` MQTT publish: topic 'zigbee2mqtt/Module dressing', payload '{"device":{"applicationVersion":0,"dateCode":" \u000020200708\u0000\u0000\u0000\u0000\u0000","friendlyName":"Module dressing","hardwareVersion":6,"ieeeAddr":"0x00047400000db744","manufacturerID":4129,"manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","model":"064888","networkAddress":47110,"powerSource":"Mains (single phase)","softwareBuildID":"001c","stackVersion":66,"type":"Router","zclVersion":2},"linkquality":54,"state":"ON"}'`

sjorge, 2022-01-09:
Sadly that’s exactly what these devices send over the mesh, they have a bunch of `nul` at the end.

Not much z2m can do about it, it’s a device firmware issue.

Tonio16, 2022-01-10:
> Sadly that’s exactly what these devices send over the mesh, they have a bunch of `nul` at the end.
> 
> Not much z2m can do about it, it’s a device firmware issue.

Ok, thanks. I didn't have it with other zigbee softwares, so I thought that it could have been a bug.

Tx

Antoine

closed 2022-01-10 by Tonio16