Koenkk/zigbee2mqtt #10686: BlitzWolf BW-IS3 - Battery

Jens-Wymeersch, 2022-01-09:
### Is your feature request related to a problem? Please describe

Most of my endpoint devices have battery level indication. Allowing me to foresee a problem besides these PIRs.

### Describe the solution you'd like

I would like to get some kind of idea about the battery level.

### Describe alternatives you've considered

There is no way to know the actual status of the battery without exposing this.

### Additional context

The PIRs themselves work great - not a lot of false positives, instead sonoff PIRs. Hope you can help

robert-pajak, 2022-01-12:
:+1: for this feature request

Jens-Wymeersch, 2022-02-12:
@Koenkk is there anyway to capture this value at all ?

Jens-Wymeersch, 2022-02-12:
unstale

Koenkk, 2022-02-12:
could you provide the data/database.db entry of this device?

Jens-Wymeersch, 2022-02-12:
@Koenkk Thank you for looking at it. Here you go :
`{"id":63,"type":"EndDevice","ieeeAddr":"0x847127fffe0d00a4","nwkAddr":16631,"manufId":0,"manufName":"_TYST11_i5j6ifxj","powerSource":"Battery","modelId":"5j6ifxj\u0000","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[0,3],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"5j6ifxj\u0000","manufacturerName":"_TYST11_i5j6ifxj","powerSource":3,"zclVersion":3,"appVersion":73,"stackVersion":0,"hwVersion":1,"dateCode":"20180727"}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":73,"stackVersion":0,"hwVersion":1,"dateCode":"20180727","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1644652694594,"defaultSendRequestWhen":"immediate"}`

Koenkk, 2022-02-12:
This device doesn't support the standard battery cluster (which provides the battery %). @alwinzwets do you know whether battery is supported with the tuya gateway?

alwinzwets, 2022-02-12:
Unfortunately I don't own a tuya gateway so I don't know.

Koenkk, 2022-02-12:
@alwinzwets could you add a `console.log('DPVALUE',dpValue)` below https://github.com/Koenkk/zigbee-herdsman-converters/blob/9fd817d0d2e2da92e316b40309c6b9ff64cbe3b4/converters/fromZigbee.js#L4575 to see wether any other dp is logged besides 1027?

alwinzwets, 2022-02-16:
I don't have this sensor in my active setup anymore. I tried to find it but I moved twice in the meantime and can't find it anymore sorry :(

closed 2022-03-26 by github-actions[bot]

Jens-Wymeersch, 2022-05-08:
unstale