Koenkk/zigbee2mqtt #10690: [Problem]: Rootless Docker not working

thueske, 2022-01-09:
### What happened?

I tried running Zigbee2MQTT as a rootless Docker container. Unfortunately, the container does not start correctly because it does not have access to the corresponding Zigbee device.

### What did you expect to happen?

I expected the container to start with no root privileges or errors.

### How to reproduce it (minimal and precise)

Start the compose file:
```yaml
version: "3.5"
services:
  bridge:
    image: koenkk/zigbee2mqtt
    restart: unless-stopped
    environment:
      - ZIGBEE2MQTT_CONFIG_MQTT_SERVER=mqtt://broker:1883
      - ZIGBEE2MQTT_CONFIG_FRONTEND_PORT=8080
      - ZIGBEE2MQTT_CONFIG_HOMEASSISTANT=true
      - TZ=Europe/Berlin
    volumes:
      - /run/udev:/run/udev:ro
      - bridge_data:/app/data
    networks:
      default:
    group_add:
      - dialout
    user: 1000:1000
    devices:
      - /dev/ttyACM0
```

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

0x26720700

### Adapter

Conbee II

### Debug log

Container:
```
/dev $ ls -lah
crw-rw----    1 root     dialout   166,   0 Jan  9 21:16 ttyACM0

/app $ test -w /dev/ttyACM0 && echo success || echo failure
failure
```

```
smarthome-bridge-1  | 2022-01-09T20:14:50.440720251Z Zigbee2MQTT:info  2022-01-09 21:14:50: Logging to console and directory: '/app/data/log/2022-01-09.21-14-35' filename: log.txt
smarthome-bridge-1  | 2022-01-09T20:14:50.626601810Z Zigbee2MQTT:info  2022-01-09 21:14:50: Starting Zigbee2MQTT version 1.22.2 (commit #414c51f)
smarthome-bridge-1  | 2022-01-09T20:14:50.647804760Z Zigbee2MQTT:info  2022-01-09 21:14:50: Starting zigbee-herdsman (0.13.188)
smarthome-bridge-1  | 2022-01-09T20:14:54.160397538Z CREATED DECONZ ADAPTER
smarthome-bridge-1  | 2022-01-09T20:14:54.228855177Z Zigbee2MQTT:error 2022-01-09 21:14:54: Error while starting zigbee-herdsman
smarthome-bridge-1  | 2022-01-09T20:14:54.239459861Z Zigbee2MQTT:error 2022-01-09 21:14:54: Failed to start zigbee
smarthome-bridge-1  | 2022-01-09T20:14:54.248763592Z Zigbee2MQTT:error 2022-01-09 21:14:54: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
smarthome-bridge-1  | 2022-01-09T20:14:54.258128194Z Zigbee2MQTT:error 2022-01-09 21:14:54: Exiting...
smarthome-bridge-1  | 2022-01-09T20:14:54.981823885Z Zigbee2MQTT:error 2022-01-09 21:14:54: Error: Error while opening serialport 'Error: Error: Permission denied, cannot open /dev/ttyACM0'
```

Host-System:
```
➜  smarthome git:(master) l  /dev/ttyACM0
crw-rw---- 1 root dialout 166, 0 Jan  9 21:17 /dev/ttyACM0

➜  smarthome git:(master) id
uid=1000(bananapi) gid=1000(thueske) groups=1000(bananapi),5(tty),6(disk),20(dialout),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),102(systemd-journal),106(input),110(netdev),112(ssh),998(docker)

➜  smarthome git:(master) test -w /dev/ttyACM0 && echo success || echo failure
success
```

ruifung, 2022-01-10:
I fail to see how this is a problem with Zigbee2MQTT
Isn't this more of a matter of you not configuring your docker container / permissions properly?

Like, if you want to run your container as nonroot, it's up to you to grant the container's effective UID permissions access to the serial port, like maybe by using ACLs.

thueske, 2022-01-10:
Sorry, but I don't get your message. The permissions on the host system match as seen in the debug logs. Or did I miss something?

thueske, 2022-01-10:
It actually looks like the group is not set in the container. `group_add` does not work as described in the documentation.

jfcabral, 2022-02-05:
Hello, I'm having the same issue as @thueske:

`Permission denied, cannot open /dev/ttyACM0`

In my case, I can confirm that the `group_add` is working just fine, so https://github.com/docker/compose/pull/8987 seems to have fixed it 😄 

After some troubleshooting and research on the web, I've found that my issue is actually caused due to this situation: https://github.com/moby/moby/issues/43019 as I am running [Docker daemon as a non-root user](https://docs.docker.com/engine/security/rootless/)

So as @ruifung has mentioned, it's not really a problem of Zigbee2MQTT.

Until it gets fixed, I'll be using this guide https://www.xmodulo.com/change-usb-device-permission-linux.html to set the permissions of the device to 666 on boot, so that later Zigbee2MQTT can access it!

I am doing this reply as I believe it could be helpful for someone having the same issue.

For reference, my specs are:
* Ubuntu 20.04.3 LTS
* Linux arm-64 5.9.0
* Docker version 20.10.12, build e91ed57
* Docker Compose version v2.2.3
* zigbee2mqtt 1.23.0

connelhooley, 2025-07-28:
I also had a lot of problems getting my dongle to work in a rootless container, my set-up varied a little to the one in the original post in that I'm using podman quadlets on Fedora.

I'll post the 3 small things I had to do to get my dongle to work as this issue comes up high in search results:

1. Add my non-root user who runs the container to the `dialout` group.
2. Add the following to my quadlet: `GroupAdd=keep-groups`.
3. Run the following command so that SE Linux stopped blocking access: `sudo setsebool -P container_use_devices=true`.

Below is an explanation of what these do, I'm still a bit a newb figuring these things out but hopefully they'll help rather than confuse!

For point 1, as is well documented above, your user needs to be in the dialout group as this group owns the device.

For point 2, this will give the primary process in the container the same GIDs that the user running the container has. So in my case, my non-root user running the container is in the dailout group, and this group has a GID of `18` on the host OS.

If I run `ls -la /dev/zigbee` inside the container, I see:

```
crw-rw----    1 nobody   nobody    166,   0 Jul 28 21:26 /dev/zigbee
```

If I run `ls -la -n /dev/zigbee` to see the GIDs for the device, I see:

```
crw-rw----    1 65534    65534     166,   0 Jul 28 21:26 /dev/zigbee
```

If I run `id` in the container I see:

```
uid=0(root) gid=0(root) groups=65534(nobody),65534(nobody),0(root)
```

So even though the GID for the device is marked as `nobody`, my user has that group and now has permission to read/write to the device.

For my example above, podman is mapping the GID `65534` inside the container, to `18` on the host OS, so this process can access device.

Don't get fooled by the mismatch in numbers here, the group being `nobody` instead of `dialout` is fine.

For point 3, I found the SE Linux line here:
https://docs.podman.io/en/latest/markdown/podman-run.1.html#device-host-device-container-device-permissions

I was banging my head against this one for so long, I had a working config for quite a while, but did not realise that SE Linux was causing the issue.