Koenkk/zigbee2mqtt #10691: [New device support]: Tuya Light Controller TS0501B

andydir87, 2022-01-10:
### Link

https://de.aliexpress.com/item/1005003612460571.html?spm=a2g0o.9042311.0.0.76924c4dhEQgZf

### Database entry

{"id":12,"type":"Router","ieeeAddr":"0xa4c138403d5e0c44","nwkAddr":57487,"manufId":4417,"manufName":"_TZ3210_wuheofsg","powerSource":"Mains (single phase)","modelId":"TS0501B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[3,4,5,6,4096,8,768,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":91,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1641793753342,"useImplicitCheckin":true}

### Comments

Unfortunately I could not add / create the device with the instruction "supporting new device".
Thanks a lot for any help.

### Supported color modes

single color

Danit2, 2022-01-10:
Try this as an external converter:
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const globalStore = require('zigbee-herdsman-converters/lib/store');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition =[
        {
        fingerprint: [{modelID: 'TS0501B', manufacturerName: '_TZ3000_4whigl8i'},
            {modelID: 'TS0501B', manufacturerName: '_TZ3210_4whigl8i'},
            {modelID: 'TS0501B', manufacturerName: '_TZ3210_9q49basr'},
            {modelID: 'TS0501B', manufacturerName: '_TZ3210_grnwgegn'},
            {modelID: 'TS0501B', manufacturerName: '_TZ3210_wuheofsg'}],
        model: 'TS0501B',
        description: 'Zigbee light',
        vendor: 'TuYa',
        extend: extend.light_onoff_brightness(),
        },
]

module.exports = definition;`

When it works, i will add it.

Danit2, 2022-01-10:
OK i have done the PR.
You can close this.

closed 2022-01-11 by andydir87