Koenkk/zigbee2mqtt #10704: [Problem]: CC2652P router won't pair to Z2M again after being flashed a few times with different firmwares (works fine in ZHA)

mortenmoulder, 2022-01-10:
### What happened?

I got a Z-Bee Duo CC2652P stick that I used as a router in ZHA. I then wanted to play with Zigbee2MQTT, so I flashed it with the latest 3.x.0 router firmware and paired it in Z2M. Once I was done playing with it as a router, I grabbed PTVO's tool and played around with that (turning on the LED for example).

I then wanted to use it as a router again, and I decided to flash it with Z-Stack 3.x.0 router firmware again. This is where it completely stopped working in Z2M, because no matter what I do, I cannot get it to pair at all. Not even if I stand next to my Sonoff CC2652P coordinator while trying to pair it.

I looked in the database.db file and there was no entry for it (I only have 2 devices paired in Z2M). I looked in configuration.yaml but only two devices showed up. I tried restarting Z2M many times.

If I then go into ZHA and I add a new device, it's instantly added as a new router. Within 5 seconds or so. I then remove it from Home Assistant, unplug the router from USB, permit join in Z2M, and then plug it back in. Nothing. It simply will not show it anywhere. Nothing in the logs either.

So it went like this:

```
Flash with Z-Stack 3.x.0 router -> ZHA
Flash with Z-Stack 3.x.0 router -> Zigbee2MQTT
Force remove from Zigbee2MQTT
Flash with PTVO custom firmware -> Zigbee2MQTT
Force remove from Zigbee2MQTT
Flash Z-Stack 3.x.0 router to device -> Zigbee2MQTT does not find it
```

### What did you expect to happen?

I would expect the Z-Duo CC2652P to join the Z2M network as a router through my Sonoff CC2652P dongle.

### How to reproduce it (minimal and precise)

1. Flash Z-Stack 3.x.0 router firmware on Z-Duo CC2652P stick
2. Permit join (All) in Z2M
3. Nothing happens

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

99999999 (using the 20 dBm coordinator firmware)

### Adapter

Sonoff CC2652P dongle as coordinator

### Debug log

```
debug 2022-01-10 22:40:27: Loaded state from file /app/data/state.json
info  2022-01-10 22:40:27: Logging to console and directory: '/app/data/log/2022-01-10.22-40-27' filename: log.txt
debug 2022-01-10 22:40:27: Removing old log directory '/app/data/log/2022-01-04.06-05-53'
info  2022-01-10 22:40:27: Starting Zigbee2MQTT version 1.22.2 (commit #414c51f)
info  2022-01-10 22:40:27: Starting zigbee-herdsman (0.13.188)
debug 2022-01-10 22:40:27: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0e248dc2c119ec1199ee76e5f01c6278-if00-port0"}}'
info  2022-01-10 22:40:28: zigbee-herdsman started (resumed)
info  2022-01-10 22:40:28: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":99999999,"transportrev":2},"type":"zStack3x0"}'
debug 2022-01-10 22:40:28: Zigbee network parameters: {"channel":11,"extendedPanID":"0x00124b00258b772a","panID":6754}
info  2022-01-10 22:40:28: Set transmit power to '20'
info  2022-01-10 22:40:28: Currently 3 devices are joined:
info  2022-01-10 22:40:28: Ikea Shortcut button (0xb4e3f9fffe756fed): E1812 - IKEA TRADFRI shortcut button (EndDevice)
info  2022-01-10 22:40:28: Ikea Shortcut button 2 (0x0c4314fffe4e88be): E1812 - IKEA TRADFRI shortcut button (EndDevice)
info  2022-01-10 22:40:28: Verpan Pantop (0xf0d1b80000187423): 4058075208421 - LEDVANCE SMART+ candle E14 tunable white (Router)
info  2022-01-10 22:40:28: Zigbee: disabling joining new devices.
info  2022-01-10 22:40:28: Connecting to MQTT server at mqtt://192.168.1.56
debug 2022-01-10 22:40:28: Using MQTT anonymous login
info  2022-01-10 22:40:28: Connected to MQTT server
info  2022-01-10 22:40:28: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
debug 2022-01-10 22:40:28: Received MQTT message on 'zigbee2mqtt/bridge/info' with data '{"commit":"c2b5229","config":{"advanced":{"adapter_concurrent":null,"adapter_delay":null,"availability_blacklist":[],"availability_blocklist":[],"availability_passlist":[],"availability_whitelist":[],"cache_state":true,"cache_state_persistent":true,"cache_state_send_on_startup":true,"channel":11,"elapsed":false,"ext_pan_id":[221,221,221,221,221,221,221,221],"homeassistant_discovery_topic":"homeassistant","homeassistant_legacy_entity_attributes":false,"homeassistant_legacy_triggers":false,"homeassistant_status_topic":"hass/status","ikea_ota_use_test_url":false,"last_seen":"disable","legacy_api":false,"log_directory":"/app/data/log/%TIMESTAMP%","log_file":"log.txt","log_level":"debug","log_output":["console","file"],"log_rotation":true,"log_symlink_current":false,"log_syslog":{"app_name":"Zigbee2MQTT","eol":"/n","host":"localhost","localhost":"localhost","path":"/dev/log","pid":"process.pid","port":123,"protocol":"tcp4","type":"5424"},"pan_id":6754,"report":false,"soft_reset_timeout":0,"timestamp_format":"YYYY-MM-DD HH:mm:ss"},"ban":[],"blocklist":[],"device_options":{"legacy":false},"devices":{"0x0c4314fffe4e88be":{"friendly_name":"Ikea Shortcut button 2"},"0xb4e3f9fffe756fed":{"friendly_name":"Ikea Shortcut button"},"0xf0d1b80000187423":{"friendly_name":"Verpan Pantop"}},"experimental":{"new_api":true,"output":"json","transmit_power":20},"external_converters":[],"frontend":{"host":"0.0.0.0","port":9442,"url":"https://z2m.i7.dk"},"groups":{},"homeassistant":true,"map_options":{"graphviz":{"colors":{"fill":{"coordinator":"#e04e5d","enddevice":"#fff8ce","router":"#4ea3e0"},"font":{"coordinator":"#ffffff","enddevice":"#000000","router":"#ffffff"},"line":{"active":"#009900","inactive":"#994444"}}}},"mqtt":{"base_topic":"zigbee2mqtt","force_disable_retain":false,"include_device_information":true,"server":"mqtt://192.168.1.56"},"ota":{"disable_automatic_update_check":false,"update_check_interval":1440},"passlist":[],"permit_join":false,"serial":{"disable_led":false,"port":"/dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0e248dc2c119ec1199ee76e5f01c6278-if00-port0"},"whitelist":[]},"config_schema":{"definitions":{"device":{"properties":{"debounce":{"description":"Debounces messages of this device","title":"Debounce","type":"number"},"debounce_ignore":{"description":"Protects unique payload values of specified payload properties from overriding within debounce time","examples":["action"],"items":{"type":"string"},"title":"Ignore debounce","type":"array"},"filtered_attributes":{"description":"Filter attributes from publish payload.","examples":["temperature","battery","action"],"items":{"type":"string"},"title":"Filtered publish attributes","type":"array"},"filtered_optimistic":{"description":"Filter attributes from optimistic publish payload when calling /set. (This has no effect if optimistic is set to false).","examples":["color_mode","color_temp","color"],"items":{"type":"string"},"title":"Filtered optimistic attributes","type":"array"},"friendly_name":{"description":"Used in the MQTT topic of a device. By default this is the device ID","readOnly":true,"title":"Friendly name","type":"string"},"icon":{"description":"The user-defined device icon for the frontend. It can be a link to an image (not a path to a file) or base64 encoded data URL like: image/svg+xml;base64,PHN2ZyB3aW....R0aD","title":"Icon","type":"string"},"optimistic":{"default":true,"description":"Publish optimistic state after set","title":"Optimistic","type":"boolean"},"qos":{"description":"QoS level for MQTT messages of this device","title":"QoS","type":"number"},"retain":{"description":"Retain MQTT messages of this device","title":"Retain","type":"boolean"},"retention":{"description":"Sets the MQTT Message Expiry in seconds, Make sure to set mqtt.version to 5","title":"Retention","type":"number"}},"required":["friendly_name"],"type":"object"},"group":{"properties":{"devices":{"items":{"type":"string"},"type":"array"},"filtered_attributes":{"items":{"type":"string"},"type":"array"},"friendly_name":{"type":"string"},"optimistic":{"type":"boolean"},"qos":{"type":"number"},"retain":{"type":"boolean"}},"required":["friendly_name"],"type":"object"}},"properties":{"advanced":{"properties":{"adapter_concurrent":{"description":"Adapter concurrency (e.g. 2 for CC2531 or 16 for CC26X2R1) (default: null, uses recommended value)","requiresRestart [... cut ...]
debug 2022-01-10 22:42:04: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"device":null,"time":254,"transaction":"jr5jn-3","value":true}'
info  2022-01-10 22:42:04: Zigbee: allowing new devices to join.
info  2022-01-10 22:42:04: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"jr5jn-3"}'
debug 2022-01-10 22:45:27: Saving state to file /app/data/state.json
debug 2022-01-10 22:50:27: Saving state to file /app/data/state.json

```

mortenmoulder, 2022-01-10:
I just tried flashing the Z-Bee Duo CC2652P with PTVO's firmware again, this time setting the status LED to "Sending reports" on DIO_7, and it just keeps flashing every 1 second, [which means](https://ptvo.info/zigbee-configurable-firmware-features/general-info/) "the router is connecting to a network".

However, it is not connecting to my Sonoff CC2652P coordinator.

[image]

mortenmoulder, 2022-01-10:
And now I just tried switching from `Device type: Router` to `Device type: End device without routing`.

What the heck is going on. I changed nothing except the device type and it pairs immediately. Why won't my device pair as a router? Is there some "underlying" check in Zigbee2MQTT, that won't pair devices based on a certain criteria or something?

dsantos21, 2022-01-18:
Just to add a bit to this: I've had that same problem, but it seems to be caused by have changed firmware back and forth between router and coordinator (some experiments with zzh and sonoff dongle plus).

At same point I had this problem. Not only it didn't pair anymore as router, but my zigbee network would soon crash (had to restart the coordinator).

Checking all logs I could and trying flash again, I realized that both devices (coordinator and router) were using the same serial number (IEEE address). Tried change it in flash programmer and it worked again.

I don't know how it could happen, but that solved for me. I still have some problem with my network, although not related (crash a few times per week), so I disabled that router to try and pin point the problem (still crashing 🙃).

bojanpotocnik, 2022-01-20:
I have similar issue - I have new ZBMINI modules with CC2652P1F chips.

Selecting Board Type `CC2652R1` in _PTVO. Firmware Configuration 1.6.0.2_ causes the board to pair immediately, however selecting `CC2652P/CC1352P2` causes Z2M to not pair, despite the module transmitting, as seen with Smart RF Sniffer.

mortenmoulder, 2022-02-21:
Commenting to keep it open.

closed 2022-03-01 by github-actions[bot]

Aralox, 2024-11-28:
I had this terrible problem too, and this was the only link I could find that described it happening. I was developing router code on an esp32c6, and z2m suddenly decided to completely ignore my device, with absolutely nothing in the logs, etc. My device behaved as though it was connected to a network. I tried changing the MAC address as described above (although the situation was different and I did not have a clash), but to no avail. What fixed it for me was clearing the zb_storage partition on the device, which also happens when esp_zb_factory_reset() is called.
I hope this helps someone, as it was a nightmare to debug.