Koenkk/zigbee2mqtt #10706: [New device support]: ADEO/Lexman Smart Plug 10A - LDSENK01F

FlorianDarras, 2022-01-10:
### Link

https://www.leroymerlin.fr/produits/electricite-domotique/domotique-et-objets-connectes/domotique/solutions-de-commande/prise-connectee-10a-2300w-lexman-82428893.html

### Database entry

{"id":37,"type":"Router","ieeeAddr":"0xcc86ecfffe2abb0a","nwkAddr":52738,"manufId":4727,"manufName":"ADEO","powerSource":"Mains (single phase)","modelId":"LDSENK01F","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,8,2821,4096,64642],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"LDSENK01F","manufacturerName":"ADEO","powerSource":1,"zclVersion":3,"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20200323-175","swBuildId":"2.8"}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00212effff07207b","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":1,"dateCode":"20200323-175","swBuildId":"2.8","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1466202208},"lastSeen":1641856195010,"useImplicitCheckin":true}

### Comments

Hello
LDSENK02F is already present in ADEO.js converters, but not this one LDSENK01F.
They are almost the same but 01F does not expose energy/power.
External converter is fully functional.
Thanks for your great work

### External converter

```shell
/**
Device '0xcc86ecfffe25cda8'
with Zigbee model 'LDSENK01F'
and manufacturer name 'ADEO'
*/

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['LDSENK01F'],
    model: 'LDSENK01F',
    vendor: 'ADEO',
    description: '10A EU smart plug',
    fromZigbee: [fz.on_off],
    toZigbee: [tz.on_off],
    exposes: [e.switch()],
    // The configure method below is needed to make the device reports on/off state changes
    // when the device is controlled manually through the button on it.
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(endpoint);
    },
};

module.exports = definition;
```

Koenkk, 2022-01-11:
added, thanks!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-11 by Koenkk

jojeju9428, 2023-03-22:
hello,
Is the 16A model compatible with zigbee2mqtt?
does the power consumption work?
[](https://www.leroymerlin.fr/produits/electricite-domotique/domotique-et-objets-connectes/domotique/solutions-de-commande/prise-connectee-16a-3680w-repeteur-lexman-82428895.html)
Thanks