Koenkk/zigbee2mqtt #10707: [Problem]: URC4450BC0-X-R (XHK1-UE) Xfinity Alarm Keypad support broken

keito, 2022-01-11:
### What happened?

Since a recent version of zigbee2mqtt (likely since Koenkk/zigbee-herdsman-converters@4391877), the XHK1-UE alarm keypad no longer responds to arm/disarm/other actions. It results in topics like the following (notice null action, action_code) regardless of the action taken on the keypad:

```
info MQTT publish: topic 'zigbee2mqtt/Keypad1', payload '{"action":null,"action_code":null,"action_transaction":null,"action_zone":null,"battery":100,"battery_low":false,"contact":true,"linkquality":90,"occupancy":false,"presence":null,"tamper":false,"temperature":20,"voltage":7000}'
```

When I revert part of the change made in the commit above (under the URC4450BC0-X-R device in zigbee-herdsman-converters/devices/universal_electronics_inc.js, replace `fromZigbee: [fz.command_arm_with_transaction` with `fromZigbee: [fz.command_arm`), the keypads work correctly again.

It doesn't appear that the XHK1-UE transmits a transaction ID. For example, with debug logging on, here are some examples of messages received from them:

```
Received Zigbee message from 'Keypad1', type 'commandArm', cluster 'ssIasAce', data '{"armmode":3,"code”:”1234”,”zoneid":23}' from endpoint 1 with groupID 0
Received Zigbee message from 'Keypad1', type 'commandArm', cluster 'ssIasAce', data '{"armmode":1,"code”:”1234”,”zoneid":23}' from endpoint 1 with groupID 0
Received Zigbee message from 'Keypad1', type 'commandArm', cluster 'ssIasAce', data '{"armmode":0,"code”:”1234”,”zoneid":23}' from endpoint 1 with groupID 0
```

So at least with the keypads that I have, the change made to the URC4450BC0-X-R device in the commit above appear incorrect.

### What did you expect to happen?

N/A - see above

### How to reproduce it (minimal and precise)

Connect a URC4450BC0-X-R (XHK1-UE) Xfinity alarm with zigbee2mqtt 1.22.2 and perform actions like arm away, disarm, and arm night.

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20210120

### Adapter

zzh!

### Debug log

N/A - see above

keito, 2022-01-11:
cc @robertalexa - since this is relevant to your commit from August. I'm curious if you encountered any XHK1-UEs that /did/ transmit the transaction ID/required this to parse their messages properly? The 2 that I have don't. Thanks!

Koenkk, 2022-01-11:
Duplicate of #10504

closed 2022-01-11 by Koenkk