Koenkk/zigbee2mqtt #10708: [New device support]: Miboxer FUT089Z (TS1002)

denv3r, 2022-01-11:
### Link

http://www.futlight.com/light/xqy/fut089z.html

### Database entry

{"id":5,"type":"EndDevice","ieeeAddr":"0x540f57fffe0f3ebd","nwkAddr":7944,"manufId":4098,"manufName":"_TZ3000_xwh1e22x","powerSource":"Battery","modelId":"TS1002","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,0,1,3,4,4096,61184,57346],"outClusterList":[25,10,0,3,4,5,6,8,4096,57346],"clusters":{"genBasic":{"attributes":{"modelId":"TS1002","manufacturerName":"_TZ3000_xwh1e22x","powerSource":3,"zclVersion":3,"appVersion":75,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":75,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1641840473136,"useImplicitCheckin":true}

### Comments

I tried to follow the supporting new device page but got stuck at "Once finished, restart Zigbee2MQTT and trigger some actions on the device. You will see messages like:". No messages are returned. I see that template is applied because the name of the device is changed. However, there are no messages.

rubenverhoef, 2022-01-21:
Same for me, I've paired the remote but did not get any data from it.

It has a cluster `manuSpecificTuya` so i also added `fz.tuya_data_point_dump` to `fromZigbee` in my "device.js". But there is no file `tuya.dump.txt` created.

I'm pressing each button and do not see anything at the log. Does anyone has an idea to get more information out of the remote?

denv3r, 2022-02-21:
Not stale

lorenz, 2022-03-22:
I have dumped the serial lines between the two small microcontrollers and the EFR32MG21 running the ZigBee stack and I see data being exchanged when a button or slider is used but the data fits no publicly documented protocol. The protocol looks similar-ish to the Tuya protocol (but it's too far from it to actually learn something from it) and the device identifies as a Tuya, but it doesn't actually use a Tuya module internally.

Defining ZigBee groups is also not doing anything.

thierry-rhone, 2022-04-11:
This remote use non standard communication.
It's works only with miboxer ZB-Box 1 gateway (impossible to pair it with a standard tuya gateway)
In Tuya iot platform, I tried to debug it. (with ZB-Box 1)
It report nothing except when you learn a new button  (nothinq when you press a learned button
[image]

lorenz, 2022-04-11:
@thierry-rhone Do you have the Miboxer gateway? If yes, could you sniff packets while you're pairing and configuring it using the official app?

thierry-rhone, 2022-04-12:
Yes I have the Mixboxer gateway. 
how can I sniff the packets other than with the tuya iot platform?

lorenz, 2022-04-12:
Do you have a ZigBee USB adapter based on a TI chip? Something like a CC2531 stick or a Sonoff USB stick?

For the CC2531 there's a guide at https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

thierry-rhone, 2022-04-12:
I have Zigbee usb adapter but if I flash it as a sniffer will it continue to work as a coordinator?

Impact123, 2022-04-12:
> New revision of ZBOSS sniffer can be downloaded from [DSR IoT](https://dsr-iot.com/downloads) web-site.

[image]

thierry-rhone, 2022-04-12:
I dont' if it can help
pairing remote to miboxer gateway
```
remote pairing

Frame 77: 68 bytes on wire (544 bits), 66 bytes captured (528 bits) on interface \\.\pipe\zboss_sniffer_COM4, id 0
IEEE 802.15.4 Data, Dst: Broadcast, Src: 0xb9de
ZigBee Network Layer Command, Dst: Broadcast, Src: 0xb9de
    Frame Control Field: 0x1209, Frame Type: Command, Discover Route: Suppress, Security, Extended Source Command
    Destination: 0xfffc
    Source: 0xb9de
    Radius: 1
    Sequence Number: 39
    Extended Source: SiliconL_ff:fe:0f:ef:44 (84:2e:14:ff:fe:0f:ef:44)
    ZigBee Security Header
        Security Control Field: 0x28, Key Id: Network Key, Extended Nonce
            ...0 1... = Key Id: Network Key (0x1)
            ..1. .... = Extended Nonce: True
        Frame Counter: 3507470
        Extended Source: SiliconL_ff:fe:0f:ef:44 (84:2e:14:ff:fe:0f:ef:44)
        Key Sequence Number: 0
        Message Integrity Code: 4c3c00cb
        [Expert Info (Warning/Undecoded): Encrypted Payload]
            [Encrypted Payload]
            [Severity level: Warning]
            [Group: Undecoded]
    Data (23 bytes)
        Data: 351c359a9180ed7abce0c236f4a7dcb64d56ad3495b306
        [Length: 23]
```

lorenz, 2022-04-12:
Can you pair it and then configure some things using the official app and then export the file from Wireshark and upload it here? Just the pairing packet doesn't help.

lorenz, 2022-04-12:
You do use Wireshark, right? Can you just save the whole session (File->Save) and upload the file here?

kirovilya, 2022-04-12:
Hello to all. I do not have a miboxer gateway, but there is a Thua. this remote also works with it.
it connects successfully, but in order to start controlling the light bulbs, a group is created, a light bulb is added to the group along with one of the remote control buttons.
[miboxer_fut089z.zip](https://github.com/Koenkk/zigbee2mqtt/files/8475555/miboxer_fut089z.zip)

thierry-rhone, 2022-04-12:
@kirovilya what network key are you using (how you found it). you mean Tuya? (I didn't find Thua). For me didn't works with a standard Tuya gateway
do you only have a remote control and a led control on your network? . I see a lot of traffic.

Here is the file that corresponds to my previous post
[WS-Zigbee.zip](https://github.com/Koenkk/zigbee2mqtt/files/8475843/WS-Zigbee.zip)

kirovilya, 2022-04-12:
@thierry-rhone used standart key 5A:69:67:42:65:65:41:6C:6C:69:61:6E:63:65:30:39 - look here https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#_2-installing-required-tools 
yes, I mistyped - used Tuya gateway.

lorenz, 2022-04-12:
@thierry-rhone In your trace there is pretty much no actual communication. Could you remove the remote from the Miboxer gateway, then start capturing, add it again and configure it (all while capturing)? This should tell us the commands needed to make it work.

thierry-rhone, 2022-04-12:
File description
1 no devices in my zigbee networks
2 pairing  FUT036Z Miboxer single color LED controler
3 pairing FUT089Z remote
4 with tuya application, creating the zone 1 group lighting with selection of LED controler
5 check of operations
   - push 'on button 1' (led turns off)
   - push 'off button 1' (led turns on)
   - manipulate the brightness slider multiple time (variation of led brightness)
   - 
[WS-Zigbee-2.zip](https://github.com/Koenkk/zigbee2mqtt/files/8476662/WS-Zigbee-2.zip)

lorenz, 2022-04-12:
I got at least one button/zone to work by literally joining the remote into a ZigBee group (yes, the remote) and also joining a device into the group. You need to do that just after the remote has been powered on as otherwise it doesn't respond to the add group request. You also need to mash the on button on the desired zone, but I'm not 100% sure how to reliably do it. Another way to link the remote is to use TouchLink, at least if your lights support it. Pressing the On button of the desired zone until the remote blinks and moving it right next to the light should bind it assuming TouchLink support.
From the traces above it might also be necessary to send the magic Tuya initialization/unlock sequence, see https://github.com/Koenkk/zigbee-herdsman-converters/blob/10836909dcaebdfe5695cf7f2d3c2b25224be25e/devices/tuya.js#L1277. All my tests were with this active.
Overall it seems like their ZigBee implementation is super sketchy and buggy. I haven't yet found a way to reliably map anything and it seems to be completely broken with CCT-style controllers.

thierry-rhone, 2022-04-13:
@lorenz Do you think I can help you more with other logs (because gateway is not mine, I would like to know if I should return it)

lorenz, 2022-04-13:
@thierry-rhone Sadly your previous trace is pretty much empty. Most likely you are sniffing on the wrong channel. You should be seeing hundreds of ZigBee packets for what you described you were doing (see @kirovilya's capture for an example). Also it would be ideal if you used the Miboxer app instead of the Tuya app. If you could confirm that you are sniffing on the right channel and then redo what you described in your last post (but with the Miboxer app) that would be very helpful. We still don't know how to reliably assign groups to buttons.

thierry-rhone, 2022-04-14:
When I join the Mibox Led controler with standard tuya zigbee gateway,
I can see in the log 
```
468	10.640737	0x4960	0x0000	ZigBee	58	Request Key
469	10.642975			IEEE 802.15.4	5	Ack
470	10.645247	0x4bc3	0xc138	IEEE 802.15.4	12	Data Request
471	10.646106			IEEE 802.15.4	5	Ack
472	10.649310	0x0000	0x4960	ZigBee	90	Transport Key
473	10.652543			IEEE 802.15.4	5	Ack
474	10.660300	0x4960	0x0000	ZigBee	65	Verify Key
475	10.662742			IEEE 802.15.4	5	Ack
476	10.668861	0x0000	0x4960	ZigBee	67	Confirm Key, SUCCESS
477	10.671349			IEEE 802.15.4	5	Ack
``` 
I replaced in wireshark the Trust Center link key by the transport key that I've read.
When I switch on/off I see zigbee commands
But I am not able to decrypt commands ! Where did I go wrong?

---------------------------------------------------------

When I join the Mibox Led controler with Mibox zigbee gateway,
I don't see any of the previous lines (file [WS-Zigbee-2.zip] in my previous post)
and when I switch on/off I don't see any zigbee command

I think the communication between mibox devices is not Zigbee 

the log of the remote in tuya iot platform logs are empty. If you press buttons nothing happens.

lorenz, 2022-04-14:
I still suspect that your Miboxer Gateway is on a different channel than you're capturing. It is possible (but IMO unlikely) that the remote talks something else between itself and the gateway as I got it to do something over ZigBee. Why whould they bother to implement much of ZigBee if they were configuring it using a proprietary protocol.

thierry-rhone, 2022-04-14:
I tried all the channels. I received data only on the 11

thierry-rhone, 2022-04-16:
I have something new  !

Mibox gateway works on canal 15
I linked the remote control and 2 single led controler to the gateway and I extracted
the transport key.

Linking of the remote
[WS-Zigbee-3.zip](https://github.com/Koenkk/zigbee2mqtt/files/8499506/WS-Zigbee-3.zip)

build 3 zones
Zone 1 controls led controler 1
Zone 2 controls led controler 2
Zome 8 controls led controlers 1 and 2

Sequence
press button 1 on - off twice
manipulate brighness slider 1
pres button 2 on - off twice
manipulate slider 2
press button 1 on off twice
manipulate slider
[WS-Zigbee-4.zip](https://github.com/Koenkk/zigbee2mqtt/files/8499507/WS-Zigbee-4.zip)

thierry-rhone, 2022-04-16:
The log when you build the Zone 1 of the remote including a Led controler
[WS-Zigbee-5.zip](https://github.com/Koenkk/zigbee2mqtt/files/8499576/WS-Zigbee-5.zip)

lorenz, 2022-04-16:
These traces look much better! I think I have identified two commands of interest, both in WS-Zigbee-3. One is Packet 172 containing a proprietary group management command (0xf0). The payload encoding is unknown, but look at what happens when you format it as groups of three bytes:
```
08 c4 27
01 d8 27
02 ec 27
03 00 28
04 14 28
05 28 28
06 3c 28
07 50 28
08
```
I'm 95% sure the first byte is the zone identifier and the following two bytes are the little-endian group identifier. The remote actually responds with a `0x81 Unsupported Cluster Command` but I suspect this to just be lazy programming. There is one caviat which is that the group IDs in subsequent traces do not match 100%. But since the source identifiers also don't match I'm not sure if this was all captured in a single session. There might also be some obfuscation at play.

The reason I'm pretty sure that this is the case is because I cannot see any other commands which could possibly be carrying group association data and I'm pretty sure that the groups are not hardcoded because if I try with the same groups on my network nothing happens.

The other interesting command is Packet 196, which is a single-byte unknown command (0xf0) which gets ACKed by the remote. I have no idea what it does, it might be some kind of activation command.

thierry-rhone, 2022-04-16:
Hi @lorenz,
All of this is complicated for me.
Do you want me to record other logs.
In that case tell me what you would like.
I think it is no longer necessary to make pairing logs?
A detail: it is not possible to link the remote with a switch.

thierry-rhone, 2022-04-17:
> ```
> 08 c4 27
> 01 d8 27
> 02 ec 27
> 03 00 28
> 04 14 28
> 05 28 28
> 06 3c 28
> 07 50 28
> 08
> ```
I don't know if it helps
have you seen that the difference is always 0x14

thierry-rhone, 2022-04-17:
I made a log complete in the same session

- Joining the Led Controler 1
- Joining the led controler 2
- Joining the remote
[zigbee join.zip](https://github.com/Koenkk/zigbee2mqtt/files/8501890/zigbee.join.zip)

we find 
- line 661 reset to factory default
- line 665 remove all group
- line 772 the unknown command ( acked at line 726)
- line 835 key confirmation

- Link remote with Led controler 1 on zone 1
- Link remote with Led controler 2 on zone 2
- Link remote with the 2 Led controlers on zone 3
[zigbee link remote leds.zip](https://github.com/Koenkk/zigbee2mqtt/files/8501932/zigbee.link.remote.leds.zip)
The difference between group number is 14

on Zone 1
- on 
- off 
- brighness
- color intensity
- chromatic color
On zone 2
- on 
- off
- Brighness
- Color intensenty
- Chromatic wheel
On zone 3
- on 
- off
- Brighness
- Color intensenty
- Chromatic wheel
[Zigbee - send commands.zip](https://github.com/Koenkk/zigbee2mqtt/files/8501956/Zigbee.-.send.commands.zip)

@lorenz 
is it possible to send the unknown command to the remote?

lorenz, 2022-04-18:
Thanks for the full trace. I think I have finally cracked the encoding of the unknown instruction. It's three bytes per zone, but not arranged in the way I thought initially. The first byte is the number of zones as a uint8. Every following three bytes represent one zone and are arranged as a uint16le with the group id followed by a uint8 zone id.

```
08
84 2b 01
98 2b 02
ac 2b 03
c0 2b 04
d4 2b 05
e8 2b 06
fc 2b 07
10 2c 08
```
This gives us

| Zone ID | Group ID |
:- | :-----------
1 | 2b84
2 | 2b98
3 | 2bac
4 | 2bc0
5 | 2bd4
6 | 2be8
7 | 2bfc
8 | 2c10

which matches perfectly with what we're seeing.

@Koenkk Basically to make this device work we need to send a custom command (0xf0) to the Groups cluster (0x0004) with a custom binary payload. This command essentially configures all 8 group bindings this remote supports in one go. We'd need guidance on two points:
1. How to actually emit a custom command with a binary payload. I haven't found any examples of this yet.
2. How we should make this configurable in Zigbee2MQTT. This is essentially a special case of a binding, with multiple being configured at once and no support for binding specific clusters only. Also the remote doesn't seem to accept commands after some time unless you replug the battery so this might only work on initial initialisation. We might have to do what the official gateway/app are doing and just pick 8 unused groups at interview/configuration time.

Koenkk, 2022-04-18:
@lorenz this can be done by adding the command to https://github.com/Koenkk/zigbee-herdsman/blob/f4f7b1b3df3a423dc8010e0ed830d0d72d5afa3a/src/zcl/definition/cluster.ts#L172.  For the payload you can use the buffer data type so that you can feed in any buffer in `configure()`, e.g.: https://github.com/Koenkk/zigbee-herdsman/blob/f4f7b1b3df3a423dc8010e0ed830d0d72d5afa3a/src/zcl/definition/cluster.ts#L1133

thierry-rhone, 2022-04-20:
@Koenkk 
If I understand correctly, I have to add the lines after line 213 to the file cluster.ts: 
```
            customMiboxCmd: {
                ID: 240,
                parameters: [
                    {name: 'groupid', type: DataType.uint16},
                    {name: 'multiplezones', type: DataType.BUFFER},
                ],
            },
```
I don't know how to use the cluster.ts file to continue building my converter. do i have to make a pull request ?

thierry-rhone, 2022-04-21:
cluster.ts
```
            miboxBulkGroupSend: {
                ID: 240,
                parameters: [
                    {name: 'groupid', type: DataType.uint16},
                    {name: 'miboxgroups', type: BuffaloZclDataType.BUFFER},

                ],
            }
```
Is this right ?

it is not possible to debug new commands (custom command) without modifying cluster.ts
I don't know how to use my forked repository
should I make a pull request?

closed 2022-04-26 by Koenkk

samuelthng, 2023-01-08:
Hello, adding to this, there is actually another function when you double click the I button.

In the manuals, it describes that the remote is able to map double click "I" button to an action in the native app. Would that be supported?

T-vK, 2023-10-12:
I am currently working a user extension which adds proper support for all the controls of the remote: https://github.com/Koenkk/zigbee2mqtt-user-extensions/pull/3
If you paste the code of this user extension in the extensions tab in Zigbe2MQTT, the all the controls will show up in Home Assistant, no Zigbee groups required. Be aware that I still have a some bugs to fix.

@samuelthng I could add support for double presses through the extension.

T-vK, 2023-10-14:
I got the extension merged, fixed a few bugs and [updated the documentation](https://www.zigbee2mqtt.io/devices/FUT089Z.html#exposing-the-controls) with detailed instructions on how to expose the controls along with examples on how to assign the brightness slider and the on/off buttons to lights or smartplugs in Home Assistant.

I'm not sure how I'll add the double press functionality. The problem is that the first press would send a normal button press event anf the second button press would send the buttton double press event.
I could add logic to wait for another button press after the first button press and then only send the double press event, but that would introduce an artificial delay to all single press events, making the remote feel less responsive.
For now you could add custom logic inside a Home Assistant Automation to detect double presses.

mycanaletto, 2023-11-19:
Thank you for your superb work!

I did notice one annoying thing though! When I switch ZHA to pairing mode for another device, it loses the remote control...

Then I have to re-pair it and everything works again...

T-vK, 2023-11-20:
Thanks for the feedback. I probably need to add another listener, maybe `onDeviceJoined`. At the moment I'm only listening to the `onDeviceAnnounce` event.

Maybe this would help:
```
        this.eventBus.onDeviceJoined(this, (data) => {
            const device = data.device;
            if (this._deviceIsSupportedRemote(device)) {
                this._setupDeviceDiscovery(device);
            }
        });
```

mycanaletto, 2023-11-29:
In fact, I've now noticed that z2m actually loses the remote control from time to time, without me doing anything in z2m...

mycanaletto, 2023-12-18:
> Thanks for the feedback. I probably need to add another listener, maybe `onDeviceJoined`. At the moment I'm only listening to the `onDeviceAnnounce` event.

I'm not sure where to put this?

Has anyone made a more complete extension?

Merci beaucoup !

darknessblade, 2024-05-11:
Is there any update on this?

I have gotten a Miboxer FUT098Z, but I can't see any ZHA events

mycanaletto, 2024-05-11:
> I have gotten a Miboxer FUT098Z, but I can't see any ZHA events

We've just about reached Z2M, but I've never managed ZHA....

darknessblade, 2024-05-11:
> > I have gotten a Miboxer FUT098Z, but I can't see any ZHA events
> 
> We've just about reached Z2M, but I've never managed ZHA....

AAH OK, then the "Zigbee Compatibility repo should be adjusted"
https://zigbee.blakadder.com/MiBoxer_FUT089Z.html

mycanaletto, 2024-05-11:
> AAH OK, then the "Zigbee Compatibility repo should be adjusted"

It may have evolved, tell me if you manage to make it work in ZHA, I'm interested!

darknessblade, 2024-05-11:
> > AAH OK, then the "Zigbee Compatibility repo should be adjusted"
> 
> It may have evolved, tell me if you manage to make it work in ZHA, I'm interested!

Will do, Already send a mail to the manufacturer to ask for the Zigbee cluster/quirk information.
Am Hoping they will give it. 
Since a slim remote with 22+ buttons is quite nice to have. [the only one coming close is the DIY one, which in my opinion is too bulky]

I had hoped to use it for curtain control [4 curtains per layer, with 4 settings per curtain like **100% XX% 0%** and **STOP**]
then use the RGB/CCT buttons to control which layer I am on.

daxcore, 2024-10-03:
maybe somebody else is interessted into this. I wrote an automation for the fut089z for lights. Just replace `mylight` and mqtt `topic`.

```
alias: _test2
description: ""
trigger:
  - platform: mqtt
    topic: zigbee2mqtt/0xf082c0fffeb174ef
condition: []
action:
  - variables:
      mylight: light.0xa81710b98bb20000
  - if:
      - condition: template
        value_template: "{{ trigger.payload_json.action_group == 101 }}"
    then:
      - if:
          - condition: template
            value_template: "{{ trigger.payload_json.action == 'move_to_hue_and_saturation' }}"
        then:
          - action: light.turn_on
            data:
              hs_color:
                - "{{ trigger.payload_json.action_hue*360/254 }}"
                - "{{ (trigger.payload_json.action_saturation-170)/84*100 }}"
            target:
              entity_id: "{{ mylight }}"
      - if:
          - condition: template
            value_template: "{{ trigger.payload_json.action == 'brightness_move_to_level' }}"
        then:
          - action: light.turn_on
            data:
              brightness: "{{ trigger.payload_json.action_level }}"
            target:
              entity_id: "{{ mylight }}"
      - if:
          - condition: template
            value_template: "{{ trigger.payload_json.action == 'color_temperature_move' }}"
        then:
          - action: light.turn_on
            data:
              color_temp: "{{ trigger.payload_json.action_color_temperature }}"
            target:
              entity_id: "{{ mylight }}"
      - if:
          - condition: template
            value_template: "{{ trigger.payload_json.action == 'on' }}"
        then:
          - action: light.turn_on
            target:
              entity_id: "{{ mylight }}"
      - if:
          - condition: template
            value_template: "{{ trigger.payload_json.action == 'off' }}"
        then:
          - action: light.turn_off
            target:
              entity_id: "{{ mylight }}"
mode: single

```

mightyred078, 2025-01-28:
> maybe somebody else is interessted into this. I wrote an automation for the fut089z for lights. Just replace `mylight` and mqtt `topic`.
>
>
>
> ```
>
> alias: _test2
>
> description: ""
>
> trigger:
>
>   - [...]

Hey, thanks for the automation! Any idea how to convert this into an automation for all groups (101-108)?

mycanaletto, 2025-01-28:
You can do this with input_select.
At the time, I had worked on the subject without managing the color.
https://www.canaletto.fr/post/home-assistant-and-mi-boxer

mightyred078, 2025-01-28:
> You can do this with input_select.
> At the time, I had worked on the subject without managing the color.
> https://www.canaletto.fr/post/home-assistant-and-mi-boxer

Hey, I've actually been using your automation, but am also trying to see if can use the colour controls as well. Do you mean input_select for different lights and action groups?

mycanaletto, 2025-01-28:
> Do you mean input_select for different lights and action groups?

Yes, input_select lets you switch from one group to another

mightyred078, 2025-01-28:
> > Do you mean input_select for different lights and action groups?
> 
> Yes, input_select lets you switch from one group to another
> 
> 

Hmm, let me try first and see if I'm successful. Thanks!

bjornarfjelldal, 2025-01-29:
Hi! I am trying to add the extension from the docs (https://www.zigbee2mqtt.io/devices/FUT089Z.html), getting the error "logging.info is not a function" when saving. Is it incomplete, is the docs not updated on what is the best practice? Anything to do with Z2M 2.0?

BDNHA77, 2025-04-08:
Here is the code that creates color, brightness, and temperature topics for zigbee2 mqtt. This can be seen in the explorer. Each of the values changes in the topics when manipulating the remote control (first press the ON button, then the color or brightness). But I don't have sensors created in HA, what to add and where, or who can fix the code, please help, this is from file: /homeassistant/zigbee2mqtt/external_extensions/miboxer-fut089z-controls-exposer.js

const DISCOVERY_PREFIX = 'homeassistant';
const VERSION = "1.0.4-stable";
const NAME = "miboxer-fut089z/controls-exposer";

class MiboxerFut089zControlsExposer {
constructor(zigbee, mqtt, state, publishEntityState, eventBus, settings, logger) {
this.zigbee = zigbee;
this.mqtt = mqtt;
this.state = state;
this.publishEntityState = publishEntityState;
this.eventBus = eventBus;
this.settings = settings;
this.logger = logger;

    this.previousValues = {
        brightness: {},
        hue: {},
        saturation: {}
    };
}

_setupDeviceDiscovery(device) {
    const ieeeAddr = device.zh._ieeeAddr;
    const friendlyName = device.zh.friendlyName || `MiBoxer_${ieeeAddr.slice(-4)}`;
    const deviceInfo = {
        identifiers: [`zigbee2mqtt_${ieeeAddr}`],
        name: friendlyName,
        manufacturer: 'MiBoxer',
        model: 'FUT089Z',
        via_device: 'zigbee2mqtt'
    };

    // Brightness sensor
    this._publishDiscovery(
        'sensor',
        `${ieeeAddr}_brightness`,
        {
            name: `${friendlyName} Brightness`,
            state_topic: `zigbee2mqtt/${ieeeAddr}/brightness/state`,
            unit_of_measurement: '%',
            value_template: '{{ value_json.brightness }}',
            device: deviceInfo
        }
    );

    // Hue sensor
    this._publishDiscovery(
        'sensor',
        `${ieeeAddr}_hue`,
        {
            name: `${friendlyName} Hue`,
            state_topic: `zigbee2mqtt/${ieeeAddr}/hue/state`,
            unit_of_measurement: '°',
            value_template: '{{ value_json.hue }}',
            device: deviceInfo
        }
    );

    // Saturation sensor
    this._publishDiscovery(
        'sensor',
        `${ieeeAddr}_saturation`,
        {
            name: `${friendlyName} Saturation`,
            state_topic: `zigbee2mqtt/${ieeeAddr}/saturation/state`,
            unit_of_measurement: '%',
            value_template: '{{ value_json.saturation }}',
            device: deviceInfo
[... 109 lines cut ...]
                // Brightness change
                this._handleBrightnessChange(ieeeAddr, zone, msgData.level);
            }
            else if (cluster === 'lightingColorCtrl' && type === 'commandMoveToHueAndSaturation') {
                // Hue and saturation change
                this._handleHueChange(
                    ieeeAddr,
                    zone,
                    msgData.hue ?? msgData.action_hue,
                    msgData.saturation ?? msgData.action_saturation
                );
            }
        } catch(e) {
            this.logger.error(`[${NAME}] Error processing message: ${e.message}`);
            this.logger.debug(e.stack);
        }
    });
}

stop() {
    this.eventBus.removeListeners(this);
}
}

module.exports = MiboxerFut089zControlsExposer;

BDNHA77, 2025-04-13:
для Zigbee2MQTT. Вот рабочий код. Для правильной работы изменить Адрес IEEE на имя pult. Можно было вместо pult автоматическую подстановку имени, но в ходе проверок, когда не всё работало, поменял и оставил pult. Почему-то не работает создание сенсоров по зонам 1-8, хотя в каком-то из вариантов работало. Но кнопки можно обработать в автоматизации. Вставить это в файл /homeassistant/zigbee2mqtt/external_extensions/miboxer-fut089z-controls-exposer.js:

const DISCOVERY_PREFIX = 'homeassistant';
const VERSION = "1.0.4-stable";
const NAME = "miboxer-fut089z/controls-exposer";

class MiboxerFut089zControlsExposer {
    constructor(zigbee, mqtt, state, publishEntityState, eventBus, settings, logger) {
        this.zigbee = zigbee;
        this.mqtt = mqtt;
        this.state = state;
        this.publishEntityState = publishEntityState;
        this.eventBus = eventBus;
        this.settings = settings;
        this.logger = logger;

        this.previousValues = {
            brightness: null,
            hue: null,
            saturation: null
        };
    }

    _setupDeviceDiscovery() {
        const deviceInfo = {
            identifiers: ["pult"],
            name: "pult",
            manufacturer: 'MiBoxer',
            model: 'FUT089Z'
        };

        // Brightness sensor
        this._publishDiscovery(
            'sensor',
            'pult_brightness',
            {
                name: "pult Brightness",
                state_topic: "pult/brightness",
                unit_of_measurement: '%',
                device: deviceInfo
            }
        );

        // Hue sensor
        this._publishDiscovery(
            'sensor',
            'pult_hue',
            {
                name: "pult Hue",
                state_topic: "pult/hue",
                unit_of_measurement: '°',
                device: deviceInfo
            }
        );

        // Saturation sensor
        this._publishDiscovery(
            'sensor',
            'pult_saturation',
            {
                name: "pult Saturation",
                state_topic: "pult/saturation",
                unit_of_measurement: '%',
                device: deviceInfo
            }
        );
    }

    _publishDiscovery(component, objectId, config) {
        const topic = `${DISCOVERY_PREFIX}/${component}/${objectId}/config`;
        this.mqtt.publish(topic, JSON.stringify(config), { retain: true });
    }

    _handleBrightnessChange(level) {
        const brightness = Math.round((level - 2) / 252 * 100);
        if (brightness !== this.previousValues.brightness) {
            this.mqtt.publish(
                "pult/brightness",
                brightness.toString(),
                { retain: true }
            );
            this.previousValues.brightness = brightness;
        }
    }

    _handleHueChange(hue, saturation) {
        if (hue !== undefined && hue !== this.previousValues.hue) {
            this.mqtt.publish(
                "pult/hue",
                hue.toString(),
                { retain: true }
            );
            this.previousValues.hue = hue;
        }

        const saturationPercent = Math.round((saturation / 254) * 100);
        if (saturation !== undefined && saturationPercent !== this.previousValues.saturation) {
            this.mqtt.publish(
                "pult/saturation",
                saturationPercent.toString(),
                { retain: true }
            );
            this.previousValues.saturation = saturationPercent;
        }
    }

    start() {
        this._setupDeviceDiscovery();
        
        this.eventBus.onDeviceMessage(this, (data) => {
            try {
                const { type, groupID, cluster, data: msgData } = data;
                
                const zone = groupID >= 101 && groupID <= 108 ? groupID - 100 : null;
                if (!zone || zone < 1 || zone > 8) return;

                if (cluster === 'genOnOff') {
                    const action = type === 'commandOn' ? 'on' : 'off';
                    this.mqtt.publish(
                        "pult/action",
                        JSON.stringify({
                            action: `button_group_${zone}_${action}`,
                            zone
                        })
                    );
                }
                else if (cluster === 'genLevelCtrl' && type === 'commandMoveToLevel') {
                    this._handleBrightnessChange(msgData.level);
                }
                else if (cluster === 'lightingColorCtrl' && type === 'commandMoveToHueAndSaturation') {
                    this._handleHueChange(
                        msgData.hue ?? msgData.action_hue,
                        msgData.saturation ?? msgData.action_saturation
                    );
                }
            } catch(e) {
                this.logger.error(`[${NAME}] Error: ${e.message}`);
            }
        });
    }

    stop() {
        this.eventBus.removeListeners(this);
    }
}

module.exports = MiboxerFut089zControlsExposer;

BDNHA77, 2025-04-13:
сенсоры по коду выше почему-то не создались в HA, хотя топики создаются. Поэтому добавьте в основной файл configuration.js:

mqtt:
  sensor:
    - name: "pult_brightness"
      state_topic: "zigbee2mqtt/pult/brightness"
      unit_of_measurement: "°"
      unique_id: "pult_brightness"
      device:
        identifiers: ["zigbee2mqtt_pult"]
    - name: "pult_hue"
      state_topic: "zigbee2mqtt/pult/hue"
      unit_of_measurement: "°"
      unique_id: "pult_hue"
      device:
        identifiers: ["zigbee2mqtt_pult"]
    - name: "pult_saturation"
      state_topic: "zigbee2mqtt/pult/saturation"
      unit_of_measurement: "°"
      unique_id: "pult_saturation"
      device:
        identifiers: ["zigbee2mqtt_pult"]

BDNHA77, 2025-04-13:
Ну и наконец automatisations.yaml:
пример управления hue, saturations, brightness, Button 108
цвет ленты можно наворотить, чтобы использовать весь цветовой спектр, но HA начинает подтупливать и пульт тормозит

- id: simple_hue_to_rgb
  alias: Простое преобразование Hue в RGB
  trigger:
  - platform: state
    entity_id: sensor.hue
  action:
  - service: light.turn_on
    target:
      entity_id: light.0xa4c138a6c8a21bc9
    data:
      rgb_color: '{% set hue_value = states(''sensor.hue'') | float %} {% set hue_normalized
        = (hue_value / 255 * 360) % 360 %} {% set sector = (hue_normalized / 60) |
        int %} {% set f = (hue_normalized / 60) - sector %} {% set p = 0 %} {% set
        q = ((1 - f) * 255) | round(0) %} {% set t = (f * 255) | round(0) %} {% if
        sector == 0 %}[255, {{ t }}, {{ p }}] {% elif sector == 1 %}[{{ q }}, 255,
        {{ p }}] {% elif sector == 2 %}[{{ p }}, 255, {{ t }}] {% elif sector == 3
        %}[{{ p }}, {{ q }}, 255] {% elif sector == 4 %}[{{ t }}, {{ p }}, 255] {%
        else %}[255, {{ p }}, {{ q }}] {% endif %}'
      brightness_pct: 100
  mode: single
- id: brightness
  alias: Преобразование Hue в RGB (защищенное)
  trigger:
  - platform: state
    entity_id:
    - sensor.none_brightness
  action:
  - service: light.turn_on
    target:
      entity_id: light.0xa4c138a6c8a21bc9
    data_template:
      brightness_pct: '{{ states(''sensor.none_brightness'')|float(100) }}'
  mode: restart
- id: saturation
  alias: MiBoxerRemote1_BrightnessSlider2
  description: Управление цветовой температурой через пульт
  trigger:
  - platform: state
    entity_id:
    - sensor.none_saturation
  condition: []
  action:
  - service: light.turn_on
    target:
      entity_id: light.0xa4c138a6c8a21bc9
    data_template:
      color_temp_kelvin: '{% set x = states(''sensor.none_saturation'') | float %}
        {{ (6500 - 3500 * (x - 70) / 30) | round(0) }}'
  mode: restart

# - id: 'lenta_ON'
#   alias: "Zone 8 Action 8_ON"
#   trigger:
#     - platform: mqtt
#       topic: "zigbee2mqtt/pult"
#   condition:
#     - condition: and
#       conditions:
#         - condition: template
#           value_template: "{{ trigger.payload_json.action_group == 108 }}"
#         - condition: template
#           value_template: "{{ trigger.payload_json.action == 'on' }}"
#   actions:
#   - type: turn_on
#     device_id: e661e93e3e77f70e573ab063ba6f13d0
#     entity_id: b80a993a78fd28bb22a7b1397daa3970
#     domain: light
#     brightness_pct: 100
#   mode: single

daxcore, 2025-04-19:
> Hey, thanks for the automation! Any idea how to convert this into an automation for all groups (101-108)?

using blueprints!

```
blueprint:
  name: Miboxer Remote
  description: ""
  domain: automation
  author: daxcore
  input:
    group_number:
      name: Group
      default: 101
      selector:
        number:
          min: 101
          max: 107
    light_target:
      name: Light
      selector:
        entity:
          domain:
            - light

triggers:
  - topic: zigbee2mqtt/0xf082c0fffeb174ef
    trigger: mqtt
conditions: []
actions:
  - variables:
      mylight: !input light_target
      group: !input group_number
  - if:
      - condition: template
        value_template: "{{ trigger.payload_json.action_group | int(0) == group | int(0) }}"
    then:
      - if:
          - condition: template
            value_template: "{{ trigger.payload_json.action == 'move_to_hue_and_saturation' }}"
        then:
          - action: light.turn_on
            data:
              hs_color:
                - "{{ trigger.payload_json.action_hue*360/254 }}"
                - "{{ (trigger.payload_json.action_saturation-170)/84*100 }}"
            target:
              entity_id: "{{ mylight }}"
      - if:
          - condition: template
            value_template: "{{ trigger.payload_json.action == 'brightness_move_to_level' }}"
        then:
          - action: light.turn_on
            data:
              brightness: "{{ trigger.payload_json.action_level }}"
            target:
              entity_id: "{{ mylight }}"
      - if:
          - condition: template
            value_template: "{{ trigger.payload_json.action == 'color_temperature_move' }}"
        then:
          - action: light.turn_on
            data:
              color_temp: "{{ trigger.payload_json.action_color_temperature }}"
            target:
              entity_id: "{{ mylight }}"
      - if:
          - condition: template
            value_template: "{{ trigger.payload_json.action == 'on' }}"
        then:
          - action: light.turn_on
            target:
              entity_id: "{{ mylight }}"
      - if:
          - condition: template
            value_template: "{{ trigger.payload_json.action == 'off' }}"
        then:
          - action: light.turn_off
            target:
              entity_id: "{{ mylight }}"
mode: single
```

MichaelWeilbaecher, 2025-04-25:
Hi@all
any Solutions about that -->

Hi! I am trying to add the extension from the docs (https://www.zigbee2mqtt.io/devices/FUT089Z.html and https://github.com/Koenkk/zigbee2mqtt-user-extensions/blob/main/unstable/miboxer-fut089z/controls-exposer.js), getting the error "logging.info is not a function" when saving. Is it incomplete, is the docs not updated or what is the best practice? Anything to do with Z2M 2.0?

I have the same problem as like my predecessor

Thanks

ewaldc, 2025-07-12:
To make it work:

- Keep the name "miboxer-fut089z-control-exposer.js" as per the install instructions (in alignment with the NAME constant)
- Change "logger.info" to "this.logger.info" (and move to the start() function where is really belongs) as the "logger" object only exists in the context of the MiboxerFut089zControlsExposer class.