Koenkk/zigbee2mqtt #10709: [Problem]: Aqara SSM-U01 energy resetting

McGiverGim, 2022-01-11:
### What happened?

I have an Aqara SSM-U01, with neutral, and power monitoring. The strange thing is that the energy looks like that:

[image]

The result is that the total energy count in Home Assistant is really wrong with this spikes.

What I see in the logs (attached below) is that the `energy` field goes up and down, but for example the `consumption` one seems estable and correct. I don't know if both are the same data, or one of them is calculated in a different way.

One solution is to publish the `consumption` instead of energy, or revisit why the `energy` is wrong.

### What did you expect to happen?

The energy grow continuously.  Not to go down and up.

### How to reproduce it (minimal and precise)

Use the Aqara SSM-U01 device.

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20211217

### Adapter

Egony E72

### Debug log

Here is one of the "spikes". The energy field goes down and up again.
```log
...
info  2022-01-11 08:57:55: MQTT publish: topic 'zigbee2mqtt/rele_despacho', payload '{"consumption":0.42,"current":0.2114,"device_temperature":38,"energy":0.42,"last_seen":"2022-01-11T08:57:55+01:00","linkquality":98,"power":47.5,"power_outage_memory":false,"state":"ON","switch_type":"toggle","temperature":39,"voltage":229.3}'
info  2022-01-11 08:58:01: MQTT publish: topic 'zigbee2mqtt/rele_despacho', payload '{"consumption":0.42,"current":0.2114,"device_temperature":38,"energy":0.34,"last_seen":"2022-01-11T08:58:01+01:00","linkquality":98,"power":47.5,"power_outage_memory":false,"state":"ON","switch_type":"toggle","temperature":39,"voltage":229.3}'
...
info  2022-01-11 09:01:29: MQTT publish: topic 'zigbee2mqtt/rele_despacho', payload '{"consumption":0.42,"current":0.2114,"device_temperature":40,"energy":0.34,"last_seen":"2022-01-11T09:01:29+01:00","linkquality":98,"power":0,"power_outage_memory":false,"state":"ON","switch_type":"toggle","temperature":39,"voltage":229.3}'
info  2022-01-11 09:01:50: MQTT publish: topic 'zigbee2mqtt/rele_despacho', payload '{"consumption":0.42,"current":0.2113,"device_temperature":40,"energy":0.42,"last_seen":"2022-01-11T09:01:50+01:00","linkquality":102,"power":47.32,"power_outage_memory":false,"state":"ON","switch_type":"toggle","temperature":40,"voltage":22
...
```

McGiverGim, 2022-01-11:
This is more strange... Maybe I'm wrong, because I don't understand exactly what the code does, but it seems that consumption and energy must be the same value:

https://github.com/Koenkk/zigbee-herdsman-converters/blob/4586475fca68966b98f0ed75a2c3e7085ef52303/converters/fromZigbee.js#L5194-L5197

But the metering modifies the value too...
https://github.com/Koenkk/zigbee-herdsman-converters/blob/4586475fca68966b98f0ed75a2c3e7085ef52303/converters/fromZigbee.js#L653

Maybe the `opple` method works right and the `metering` not and overwrites the value?

McGiverGim, 2022-01-11:
Here is the log with a spike in debug mode, sorry, my earlier one was in info mode.

As suspected, `energy` goes down when received a `seMetering`:
```log
debug 2022-01-11 12:56:52: Received Zigbee message from 'rele_despacho', type 'attributeReport', cluster 'genDeviceTempCfg', data '{"currentTemperature":40}' from endpoint 1 with groupID 0
info  2022-01-11 12:56:52: MQTT publish: topic 'zigbee2mqtt/rele_despacho', payload '{"consumption":0.6,"current":0.2093,"device_temperature":40,"energy":0.6,"last_seen":"2022-01-11T12:56:52+01:00","linkquality":65,"power":47.12,"power_outage_memory":false,"state":"ON","switch_type":"toggle","temperature":39,"voltage":229.7}'
debug 2022-01-11 12:56:56: Received Zigbee message from 'rele_despacho', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,340]}' from endpoint 1 with groupID 0
info  2022-01-11 12:56:56: MQTT publish: topic 'zigbee2mqtt/rele_despacho', payload '{"consumption":0.6,"current":0.2093,"device_temperature":40,"energy":0.34,"last_seen":"2022-01-11T12:56:56+01:00","linkquality":65,"power":47.12,"power_outage_memory":false,"state":"ON","switch_type":"toggle","temperature":39,"voltage":229.7}'
```
and returns to the correct value with `aqaraOpple`:
```log
debug 2022-01-11 13:01:05: Received Zigbee message from 'rele_despacho', type 'attributeReport', cluster 'genDeviceTempCfg', data '{"currentTemperature":40}' from endpoint 1 with groupID 0
info  2022-01-11 13:01:05: MQTT publish: topic 'zigbee2mqtt/rele_despacho', payload '{"consumption":0.6,"current":0.2093,"device_temperature":40,"energy":0.34,"last_seen":"2022-01-11T13:01:05+01:00","linkquality":65,"power":0,"power_outage_memory":false,"state":"ON","switch_type":"toggle","temperature":39,"voltage":229.7}'
debug 2022-01-11 13:01:14: Received Zigbee message from 'rele_despacho', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[100,16,1,3,40,40,152,57,204,78,60,66,149,57,110,249,25,63,150,57,172,33,16,69,151,57,158,74,80,67,5,33,4,0,154,32,0,11,32,0,8,33,21,1,9,33,1,12],"type":"Buffer"}}' from endpoint 1 with groupID 0
info  2022-01-11 13:01:14: MQTT publish: topic 'zigbee2mqtt/rele_despacho', payload '{"consumption":0.6,"current":0.2083,"device_temperature":40,"energy":0.6,"last_seen":"2022-01-11T13:01:14+01:00","linkquality":65,"power":47.08,"power_outage_memory":false,"state":"ON","switch_type":"toggle","temperature":40,"voltage":230.6}'
```

Koenkk, 2022-01-11:
Removed the metering converter, can you check if this is fixed now?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

McGiverGim, 2022-01-12:
Thanks @Koenkk for the fast fix! Testing it and all seems ok now. I will continue with an eye over it the next hours.

McGiverGim, 2022-01-12:
@Koenkk It seems it happens similar with power. In the graphs is difficult to see, but:

The `aqaraOpple` publishes the `power`:
```log
debug 2022-01-12 08:15:01: Received Zigbee message from 'rele_despacho', type 'attributeReport', cluster 'genDeviceTempCfg', data '{"currentTemperature":39}' from endpoint 1 with groupID 0
info  2022-01-12 08:15:01: MQTT publish: topic 'zigbee2mqtt/rele_despacho', payload '{"consumption":0.77,"current":0.215,"device_temperature":39,"energy":0.77,"last_seen":"2022-01-12T08:15:01+01:00","linkquality":61,"power":0,"power_outage_memory":false,"state":"ON","switch_type":"toggle","temperature":36,"voltage":227.6}'
debug 2022-01-12 08:15:05: Received Zigbee message from 'rele_despacho', type 'attributeReport', cluster 'aqaraOpple', data '{"247":{"data":[100,16,1,3,40,38,152,57,185,100,62,66,149,57,171,68,69,63,150,57,108,15,14,69,151,57,188,166,85,67,5,33,4,0,154,32,0,11,32,0,8,33,21,1,9,33,1,12],"type":"Buffer"}}' from endpoint 1 with groupID 0
info  2022-01-12 08:15:05: MQTT publish: topic 'zigbee2mqtt/rele_despacho', payload '{"consumption":0.77,"current":0.2137,"device_temperature":39,"energy":0.77,"last_seen":"2022-01-12T08:15:05+01:00","linkquality":61,"power":47.6,"power_outage_memory":false,"state":"ON","switch_type":"toggle","temperature":38,"voltage":227.3}'
```
but the `haElectricalMeasurement` resets to zero:
```log
debug 2022-01-12 08:21:35: Received Zigbee message from 'rele_despacho', type 'attributeReport', cluster 'genDeviceTempCfg', data '{"currentTemperature":40}' from endpoint 1 with groupID 0
info  2022-01-12 08:21:35: MQTT publish: topic 'zigbee2mqtt/rele_despacho', payload '{"consumption":0.78,"current":0.2143,"device_temperature":40,"energy":0.78,"last_seen":"2022-01-12T08:21:35+01:00","linkquality":61,"power":47.56,"power_outage_memory":false,"state":"ON","switch_type":"toggle","temperature":39,"voltage":226.3}'
debug 2022-01-12 08:21:43: Received Zigbee message from 'rele_despacho', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
info  2022-01-12 08:21:43: MQTT publish: topic 'zigbee2mqtt/rele_despacho', payload '{"consumption":0.78,"current":0.2143,"device_temperature":40,"energy":0.78,"last_seen":"2022-01-12T08:21:43+01:00","linkquality":61,"power":0,"power_outage_memory":false,"state":"ON","switch_type":"toggle","temperature":39,"voltage":226.3}'
```

You can see the state `ON` in all the logs, so it must be 47-48 W in all of them.

Koenkk, 2022-01-12:
@McGiverGim also ignored `haElectricalMeasurement` now, could you check if the `current` and `voltage` values are still ok?

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

McGiverGim, 2022-01-13:
@Koenkk tested and it seems to change the `current` and `voltage` without problem. 

Looking at it I've detected several "strange" things:
- I don't know if it has the correct "refreshing" time, the `aqaraOpple` is received only once about each 5-6 minutes. I suppose this is not configurable?
- If I power off, wait 20 seconds, and power on again, I don't receive any `aqaraOpple` message, so the power does not return to zero. I will try with longer periods today later.
- I continue seeing the `haElectricalMeasurement` and the `seMetering` in the reporting tab, but maybe it simply is a "backup" of the old state before the update?
[image]

Regards!

McGiverGim, 2022-01-13:
I think it seems we are ignoring `genAnalogInput`, that is sent after each ON-OFF and seems to contain the power value:
```log
debug 2022-01-13 08:42:36: Received MQTT message on 'zigbee2mqtt/rele_despacho/set' with data 'OFF'
debug 2022-01-13 08:42:36: Publishing 'set' 'state' to 'rele_despacho'
info  2022-01-13 08:42:36: MQTT publish: topic 'zigbee2mqtt/rele_despacho', payload '{"consumption":1.21,"current":0.2135,"device_temperature":40,"energy":1.21,"last_seen":"2022-01-13T08:42:36+01:00","linkquality":65,"power":47.28,"power_outage_memory":false,"state":"ON","switch_type":"toggle","temperature":40,"voltage":225.9}'
info  2022-01-13 08:42:36: MQTT publish: topic 'zigbee2mqtt/rele_despacho', payload '{"consumption":1.21,"current":0.2135,"device_temperature":40,"energy":1.21,"last_seen":"2022-01-13T08:42:36+01:00","linkquality":65,"power":47.28,"power_outage_memory":false,"state":"OFF","switch_type":"toggle","temperature":40,"voltage":225.9}'
debug 2022-01-13 08:42:36: Received Zigbee message from 'rele_despacho', type 'attributeReport', cluster 'genOnOff', data '{"245":117440530,"onOff":0}' from endpoint 1 with groupID 0
debug 2022-01-13 08:42:36: Received Zigbee message from 'rele_despacho', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2022-01-13 08:42:36: MQTT publish: topic 'zigbee2mqtt/rele_despacho', payload '{"consumption":1.21,"current":0.2135,"device_temperature":40,"energy":1.21,"last_seen":"2022-01-13T08:42:36+01:00","linkquality":65,"power":47.28,"power_outage_memory":false,"state":"OFF","switch_type":"toggle","temperature":40,"voltage":225.9}'
info  2022-01-13 08:42:36: MQTT publish: topic 'zigbee2mqtt/rele_despacho', payload '{"consumption":1.21,"current":0.2135,"device_temperature":40,"energy":1.21,"last_seen":"2022-01-13T08:42:36+01:00","linkquality":65,"power":47.28,"power_outage_memory":false,"state":"OFF","switch_type":"toggle","temperature":40,"voltage":225.9}'
debug 2022-01-13 08:42:36: Received Zigbee message from 'rele_despacho', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":0}' from endpoint 21 with groupID 0
info  2022-01-13 08:42:36: MQTT publish: topic 'zigbee2mqtt/rele_despacho', payload '{"consumption":1.21,"current":0.2135,"device_temperature":40,"energy":1.21,"last_seen":"2022-01-13T08:42:36+01:00","linkquality":65,"power":47.28,"power_outage_memory":false,"state":"OFF","switch_type":"toggle","temperature":40,"voltage":225.9}'
debug 2022-01-13 08:42:47: Received Zigbee message from 'rele_despacho', type 'attributeReport', cluster 'genDeviceTempCfg', data '{"currentTemperature":41}' from endpoint 1 with groupID 0
info  2022-01-13 08:42:47: MQTT publish: topic 'zigbee2mqtt/rele_despacho', payload '{"consumption":1.21,"current":0.2135,"device_temperature":41,"energy":1.21,"last_seen":"2022-01-13T08:42:47+01:00","linkquality":65,"power":47.28,"power_outage_memory":false,"state":"OFF","switch_type":"toggle","temperature":40,"voltage":225.9}'
debug 2022-01-13 08:42:57: Received Zigbee message from 'rele_despacho', type 'attributeReport', cluster 'genDeviceTempCfg', data '{"currentTemperature":40}' from endpoint 1 with groupID 0
info  2022-01-13 08:42:57: MQTT publish: topic 'zigbee2mqtt/rele_despacho', payload '{"consumption":1.21,"current":0.2135,"device_temperature":40,"energy":1.21,"last_seen":"2022-01-13T08:42:57+01:00","linkquality":65,"power":47.28,"power_outage_memory":false,"state":"OFF","switch_type":"toggle","temperature":40,"voltage":225.9}'
debug 2022-01-13 08:43:07: Received Zigbee message from 'rele_despacho', type 'attributeReport', cluster 'genDeviceTempCfg', data '{"currentTemperature":40}' from endpoint 1 with groupID 0
info  2022-01-13 08:43:07: MQTT publish: topic 'zigbee2mqtt/rele_despacho', payload '{"consumption":1.21,"current":0.2135,"device_temperature":40,"energy":1.21,"last_seen":"2022-01-13T08:43:07+01:00","linkquality":69,"power":47.28,"power_outage_memory":false,"state":"OFF","switch_type":"toggle","temperature":40,"voltage":225.9}'
debug 2022-01-13 08:43:12: Received Zigbee message from 'rele_despacho', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2022-01-13 08:43:12: MQTT publish: topic 'zigbee2mqtt/rele_despacho', payload '{"consumption":1.21,"current":0.2135,"device_temperature":40,"energy":1.21,"last_seen":"2022-01-13T08:43:12+01:00","linkquality":65,"power":47.28,"power_outage_memory":false,"state":"ON","switch_type":"toggle","temperature":40,"voltage":225.9}'
debug 2022-01-13 08:43:12: Received Zigbee message from 'rele_despacho', type 'attributeReport', cluster 'genOnOff', data '{"245":54377472,"onOff":1}' from endpoint 1 with groupID 0
info  2022-01-13 08:43:12: MQTT publish: topic 'zigbee2mqtt/rele_despacho', payload '{"consumption":1.21,"current":0.2135,"device_temperature":40,"energy":1.21,"last_seen":"2022-01-13T08:43:12+01:00","linkquality":65,"power":47.28,"power_outage_memory":false,"state":"ON","switch_type":"toggle","temperature":40,"voltage":225.9}'
debug 2022-01-13 08:43:15: Received Zigbee message from 'rele_despacho', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":47.493995666503906}' from endpoint 21 with groupID 0
info  2022-01-13 08:43:15: MQTT publish: topic 'zigbee2mqtt/rele_despacho', payload '{"consumption":1.21,"current":0.2135,"device_temperature":40,"energy":1.21,"last_seen":"2022-01-13T08:43:15+01:00","linkquality":65,"power":47.28,"power_outage_memory":false,"state":"ON","switch_type":"toggle","temperature":40,"voltage":225.9}'
```
It seems the converter exists:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/4586475fca68966b98f0ed75a2c3e7085ef52303/converters/fromZigbee.js#L5014-L5020

Koenkk, 2022-01-13:
> I don't know if it has the correct "refreshing" time, the aqaraOpple is received only once about each 5-6 minutes. I suppose this is not configurable?

I don't expect is

> I continue seeing the haElectricalMeasurement and the seMetering in the reporting tab, but maybe it simply is a "backup" of the old state before the update?

Removed it now, so it will not appear for new devices, for already paired devices it will still be there. (it is not a problem)

> Another "strange" thing is that the temperature genDeviceTempCfg is sent each 10 seconds. Is more important for Aqara to report temperature and not power for example?

Changed this to 5 mins now

> I think it seems we are ignoring genAnalogInput, that is sent after each ON-OFF and seems to contain the power value:

Added a converter for this now

McGiverGim, 2022-01-14:
Thanks @Koenkk ! It seeems to work ok now. It shows the on/off, the voltage and current (but only updated once each 5-6 minutes) shows the correct power when on/off, etc.

One question, by curiosity. When I turn on or off the switch, the device sends two messages, one with the or (or off) message, and the other with the on (or off) and another "245" attribute:
```
debug 2022-01-14 08:09:39: Received Zigbee message from 'rele_despacho', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2022-01-14 08:09:39: MQTT publish: topic 'zigbee2mqtt/rele_despacho', payload '{"consumption":1.64,"current":0.2154,"device_temperature":35,"energy":1.64,"last_seen":"2022-01-14T08:09:39+01:00","linkquality":51,"power":0,"power_outage_memory":false,"state":"ON","switch_type":"toggle","temperature":35,"voltage":226.9}'
debug 2022-01-14 08:09:39: Received Zigbee message from 'rele_despacho', type 'attributeReport', cluster 'genOnOff', data '{"245":54377472,"onOff":1}' from endpoint 1 with groupID 0
info  2022-01-14 08:09:39: MQTT publish: topic 'zigbee2mqtt/rele_despacho', payload '{"consumption":1.64,"current":0.2154,"device_temperature":35,"energy":1.64,"last_seen":"2022-01-14T08:09:39+01:00","linkquality":51,"power":0,"power_outage_memory":false,"state":"ON","switch_type":"toggle","temperature":35,"voltage":226.9}'
```

Do you know what is this 245 attribute?

Apart of this, this issue can be closed, The device seems to work ok now.

Koenkk, 2022-01-14:
> Do you know what is this 245 attribute?

No clue, definitely Xiaomi specific, looks like a random number to me.

closed 2022-01-14 by Koenkk

vvaltson, 2022-02-06:
After this update none of my SSM-U01's report consumed energy or power anymore.

> MQTT publish: topic 'zigbee2mqtt/garage_extension', payload '{"device_temperature":30,"energy":null,"linkquality":108,"power":null,"power_outage_memory":true,"state":"ON","switch_type":"toggle"}'

inode64, 2022-05-11:
I have the same problem, after updating to any version higher than 1.23 the power is not shown
I have tried even with the latest version 1.25.1
But there is no way for it to work, I have erased the device and rejoined it but none of this has worked
Please, what more information do you need to solve it?

a greeting

-----
Zigbee2MQTT:info  2022-05-11 19:05:35: Configuring '0x04cf8cdf3c8e8762'
Zigbee2MQTT:info  2022-05-11 19:05:35: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"Aqara single switch module T1 (with neutral)","friendly_name":"0x04cf8cdf3c8e8762","model":"SSM-U01","supported":true,"vendor":"Xiaomi"},"type":"pairing"}'
Zigbee2MQTT:info  2022-05-11 19:05:35: MQTT publish: topic 'homeassistant/switch/0x04cf8cdf3c8e8762/switch/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0x04cf8cdf3c8e8762/set","device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c8e8762"],"manufacturer":"Xiaomi","model":"Aqara single switch module T1 (with neutral) (SSM-U01)","name":"0x04cf8cdf3c8e8762","sw_version":""},"json_attributes_topic":"zigbee2mqtt/0x04cf8cdf3c8e8762","name":"0x04cf8cdf3c8e8762","payload_off":"OFF","payload_on":"ON","state_topic":"zigbee2mqtt/0x04cf8cdf3c8e8762","unique_id":"0x04cf8cdf3c8e8762_switch_zigbee2mqtt","value_template":"{{ value_json.state }}"}'
Zigbee2MQTT:info  2022-05-11 19:05:35: MQTT publish: topic 'homeassistant/sensor/0x04cf8cdf3c8e8762/energy/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c8e8762"],"manufacturer":"Xiaomi","model":"Aqara single switch module T1 (with neutral) (SSM-U01)","name":"0x04cf8cdf3c8e8762","sw_version":""},"device_class":"energy","enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/0x04cf8cdf3c8e8762","name":"0x04cf8cdf3c8e8762 energy","state_class":"total_increasing","state_topic":"zigbee2mqtt/0x04cf8cdf3c8e8762","unique_id":"0x04cf8cdf3c8e8762_energy_zigbee2mqtt","unit_of_measurement":"kWh","value_template":"{{ value_json.energy }}"}'
Zigbee2MQTT:info  2022-05-11 19:05:35: MQTT publish: topic 'homeassistant/sensor/0x04cf8cdf3c8e8762/power/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c8e8762"],"manufacturer":"Xiaomi","model":"Aqara single switch module T1 (with neutral) (SSM-U01)","name":"0x04cf8cdf3c8e8762","sw_version":""},"device_class":"power","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x04cf8cdf3c8e8762","name":"0x04cf8cdf3c8e8762 power","state_class":"measurement","state_topic":"zigbee2mqtt/0x04cf8cdf3c8e8762","unique_id":"0x04cf8cdf3c8e8762_power_zigbee2mqtt","unit_of_measurement":"W","value_template":"{{ value_json.power }}"}'
Zigbee2MQTT:info  2022-05-11 19:05:35: MQTT publish: topic 'homeassistant/sensor/0x04cf8cdf3c8e8762/device_temperature/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c8e8762"],"manufacturer":"Xiaomi","model":"Aqara single switch module T1 (with neutral) (SSM-U01)","name":"0x04cf8cdf3c8e8762","sw_version":""},"device_class":"temperature","enabled_by_default":true,"entity_category":"diagnostic","json_attributes_topic":"zigbee2mqtt/0x04cf8cdf3c8e8762","name":"0x04cf8cdf3c8e8762 device temperature","state_class":"measurement","state_topic":"zigbee2mqtt/0x04cf8cdf3c8e8762","unique_id":"0x04cf8cdf3c8e8762_device_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.device_temperature }}"}'
Zigbee2MQTT:info  2022-05-11 19:05:35: MQTT publish: topic 'homeassistant/switch/0x04cf8cdf3c8e8762/switch_power_outage_memory/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"command_topic":"zigbee2mqtt/0x04cf8cdf3c8e8762/set/power_outage_memory","device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c8e8762"],"manufacturer":"Xiaomi","model":"Aqara single switch module T1 (with neutral) (SSM-U01)","name":"0x04cf8cdf3c8e8762","sw_version":""},"entity_category":"config","icon":"mdi:memory","json_attributes_topic":"zigbee2mqtt/0x04cf8cdf3c8e8762","name":"0x04cf8cdf3c8e8762 power_outage_memory","payload_off":"false","payload_on":"true","state_topic":"zigbee2mqtt/0x04cf8cdf3c8e8762","unique_id":"0x04cf8cdf3c8e8762_switch_power_outage_memory_zigbee2mqtt","value_template":"{% if value_json.power_outage_memory %} true {% else %} false {% endif %}"}'
Zigbee2MQTT:info  2022-05-11 19:05:35: MQTT publish: topic 'homeassistant/sensor/0x04cf8cdf3c8e8762/power_outage_count/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c8e8762"],"manufacturer":"Xiaomi","model":"Aqara single switch module T1 (with neutral) (SSM-U01)","name":"0x04cf8cdf3c8e8762","sw_version":""},"enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/0x04cf8cdf3c8e8762","name":"0x04cf8cdf3c8e8762 power outage count","state_topic":"zigbee2mqtt/0x04cf8cdf3c8e8762","unique_id":"0x04cf8cdf3c8e8762_power_outage_count_zigbee2mqtt","value_template":"{{ value_json.power_outage_count }}"}'
Zigbee2MQTT:info  2022-05-11 19:05:35: MQTT publish: topic 'homeassistant/sensor/0x04cf8cdf3c8e8762/switch_type/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x04cf8cdf3c8e8762"],"manufacturer":"Xiaomi","model":"Aqara single sw [... cut ...]
Zigbee2MQTT:info  2022-05-11 19:06:20: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c8e8762', payload '{"current":null,"device_temperature":null,"energy":null,"linkquality":255,"power":null,"power_outage_count":null,"power_outage_memory":null,"state":"ON","switch_type":null,"temperature":null,"voltage":null}'
Zigbee2MQTT:info  2022-05-11 19:10:34: MQTT publish: topic 'zigbee2mqtt/0x04cf8cdf3c8e8762', payload '{"current":null,"device_temperature":29,"energy":null,"linkquality":255,"power":null,"power_outage_count":null,"power_outage_memory":null,"state":"ON","switch_type":null,"temperature":null,"voltage":null}'

McGiverGim, 2022-05-11:
@inode64  I think is better to open a new issue than replying to a closed one. It will be read for more people.

When you open it, we need a "debug" log of the device to see what messages are your device sending.