Koenkk/zigbee2mqtt #10710: [Problem]: Aquara Bulb ZNLDP12LM doesn't transition correctly on ha.turn_on()

RubenKelevra, 2022-01-11:
### What happened?

If I set the transition in the Zigbee2MQTT settings to say 30, it will dim very slowly as expected.

But if the transition in Zigbee2MQTT settings is zero, and I send a turn_on command (light change request in HA) with a configured transition period it will instantly switch to the requested stated ignoring the requested transition period.

I think we somehow overwrite the requested transition period from the command with the 0 from the settings somewhere, since it does transition longer if the settings are changed.

### What did you expect to happen?

Lorem ipsum dolor sit amet, consetetur sadipscing elitr, sed diam nonumy eirmod tempor invidunt ut labore et dolore magna aliquyam erat, sed diam voluptua. At vero eos et accusam et justo duo dolores et ea rebum.

### How to reproduce it (minimal and precise)

1. Change a light with ha's turn_on command with a transition period
2. Observe issue

### Zigbee2MQTT version

1.22.1-1

### Adapter firmware version

unknown

### Adapter

Slaesh's CC2652RB

### Debug log

Lorem ipsum dolor sit amet, consetetur sadipscing elitr, sed diam nonumy eirmod tempor invidunt ut labore et dolore magna aliquyam erat, sed diam voluptua. At vero eos et accusam et justo duo dolores et ea rebum.

Koenkk, 2022-01-11:
The message transition should be preferred over the options transition, code: https://github.com/Koenkk/zigbee-herdsman-converters/blob/ca9398a429714bc335fde2e3fea2407e2340e46c/lib/utils.js#L335

Could you provide the herdsman debug log when turning the bulb on with a transition in the message?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

RubenKelevra, 2022-01-12:
> The message transition should be preferred over the options transition, code: https://github.com/Koenkk/zigbee-herdsman-converters/blob/ca9398a429714bc335fde2e3fea2407e2340e46c/lib/utils.js#L335 [...]

Is there a guide how to get to the console on the docker container on HA OS? 🤔

RubenKelevra, 2022-01-12:
I have no idea why, since I changed nothing, but now it's working?

RubenKelevra, 2022-01-12:
I mean I did a reboot of the HA host yesterday, maybe it`s just a HA bug :)

closed 2022-01-12 by RubenKelevra