Koenkk/zigbee2mqtt #10713: [New device support]: Pernio Power Link (16A)

drako86, 2022-01-11:
### Link

https://perenio.com/catalog/power-plug-power-link-white-pehpl01

### Database entry

{"id":20,"type":"Unknown","ieeeAddr":"0xf4ce364f116817e5","nwkAddr":51908,"epList":[],"endpoints":{},"interviewCompleted":false,"meta":{},"lastSeen":1641907856033,"useImplicitCheckin":true}

### Comments

Hi, I tried to add this socket, but I realized that it is not yet compatible (while other products from the same manufacturer do).
so I would like to try to integrate it and help someone who maybe buys it

sorry if i'm wrong something and sorry for my bad english

oyunets, 2022-01-12:
External converter with limited functionality

`
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['PEHPL0X'],
    model: 'PEHPL0X',
    vendor: 'Perenio',
    description: 'Perenio Power Link',
    fromZigbee: [fz.on_off, fz.metering],
    toZigbee: [tz.on_off],
    exposes: [e.switch(), e.energy()],
    configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint1 = device.getEndpoint(1);
            await reporting.bind(endpoint1, coordinatorEndpoint, ['genOnOff', 'seMetering']);
            await reporting.onOff(endpoint1);
            endpoint1.saveClusterAttributeKeyValue('seMetering', {divisor: 100, multiplier: 1});
            await reporting.currentSummDelivered(endpoint1);
    },
};

module.exports = definition;
`

drako86, 2022-01-12:
hello, first of all thank you!
but why "limited functionality" ?
can I send you some data to improve the code?
will be added in the next version of Z2M?

oyunets, 2022-01-12:
External converter supports: onOff and power meterring functionality only, 
Original application supports in additonal : voltage metering, on/off timers, power limits, voltage limits, default states.

drako86, 2022-01-12:
I tried to add the file, but I still don't see the usable socket and Z2M keeps telling me it's not supported

oyunets, 2022-01-12:
Take a look at https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-adding-your-device

drako86, 2022-01-12:
Yes, I have already followed that guide, the problem is that the socket does not expose 

zigbeeModel
Model
Vendor 

so I don't think it reads the .js file

solution ???

drako86, 2022-01-13:
ok, perfect now i see plug ... but it's possible Power ?

Jahx, 2022-02-11:
Hi @drako86! May I ask you, were you able to implement an external converter for the plug?

KonsGit, 2022-03-15:
External converter work fine, but have limited functionality
Please, add full support for this socket in the next version of Z2M

closed 2022-04-23 by github-actions[bot]