Koenkk/zigbee2mqtt #10714: [Problem]: Supported device not supported

cantfindanam, 2022-01-11:
### What happened?

Supported device Philips 3261031P6 is not recognized as supported. Before it worked in ZHA and Deconz just fine with the Respbee2 module.

### What did you expect to happen?

Well it should be recognized as supported right? ;)

### How to reproduce it (minimal and precise)

Happens every time when adding the device to my Zigbee network.

### Zigbee2MQTT version

1.22.2 

### Adapter firmware version

0x26520700

### Adapter

ConBee2/RaspBee2

### Debug log

Info Device '0x001788010b4db982' joined
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788010b4db982","ieee_address":"0x001788010b4db982"},"type":"device_joined"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x001788010b4db982"},"type":"device_connected"}'
Info Starting interview of '0x001788010b4db982'
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x001788010b4db982","ieee_address":"0x001788010b4db982","status":"started"},"type":"device_interview"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x001788010b4db982"},"type":"pairing"}'
Debug Received Zigbee message from '0x001788010b4db982', type 'readResponse', cluster 'genBasic', data '{"hwVersion":0}' from endpoint 11 with groupID null
Debug Skipping message, definition is undefined and still interviewing
Info Successfully interviewed '0x001788010b4db982', device has successfully been paired
Warning Device '0x001788010b4db982' with Zigbee model '929003055001' and manufacturer name 'Philips' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x001788010b4db982","ieee_address":"0x001788010b4db982","status":"successful","supported":false},"type":"device_interview"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0x001788010b4db982","supported":false},"type":"pairing"}'

Koenkk, 2022-01-11:
Please provide the color temp range, then I can add it: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

Yamakaky, 2022-01-17:
I have the same light and confirm that this converter works. I have the [aluminium model](https://www.philips-hue.com/fr-fr/p/hue-white-ambiance-plafonnier-being/8719514341111) with ID 929003055201, so I had to add it to the list. Maybe different frame colors will have different ids ?

@Koenkk on my device, I get `{"colorTempPhysicalMin":153,"colorTempPhysicalMax":454}`.

cantfindanam, 2022-01-17:
@Koenkk sorry was quite busy in the past days. 
I also get : {"colorTempPhysicalMin":153,"colorTempPhysicalMax":454}.
I think the IDs in the philips.py are not the Zigbee ID's but the model numbers. So I suspect that non of the Being configs work atm.

@Yamakaky yes the IDs are different. I have the white model.

Koenkk, 2022-01-18:
@cantfindanam thanks, added `929003055001`, 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

@Yamakaky if your device is not supported in the latest-dev please fill in : https://github.com/Koenkk/zigbee2mqtt/issues/new?assignees=&labels=new+device+support&template=new_device_support.yaml&title=%5BNew+device+support%5D%3A+

closed 2022-01-18 by Koenkk

Yamakaky, 2022-01-18:
I'm on the 21.10 release of home assistant so i can't update for now I think...