Koenkk/zigbee2mqtt #10733: [Update device support]: DANALOCK + DANAPAD : Pin code

XLDom, 2022-01-12:
### Is your feature request related to a problem? Please describe

Hello,

when we connect DANAPAD to the DANALOCK, we can have 20 pin code.

### Describe the solution you'd like

Here is pool request for the definition file : https://github.com/Koenkk/zigbee-herdsman-converters/pull/3665
Here is pool request for the documentation : https://github.com/Koenkk/zigbee2mqtt.io/pull/1135
(you can correct my French English if needed into the documentation lol)
All is tested and run.

### Describe alternatives you've considered

nothing

### Additional context

nothing

Koenkk, 2022-01-12:
merged, thanks!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-12 by Koenkk

smulle48, 2022-02-07:
How does this work?

I got both Danalock and Danapad, but I can't get the pincode to work. If I add a pincode with the MQTT message or in the gut, it does not show up in the Danalock app, and does not work on the Danapad.

If I add a pincode in the Danalock app, the pincode is not shown in the mqtt response.

smulle48, 2022-02-08:
> ### Is your feature request related to a problem? Please describe
> Hello,
> 
> when we connect DANAPAD to the DANALOCK, we can have 20 pin code.
> 
> ### Describe the solution you'd like
> Here is pool request for the definition file : [Koenkk/zigbee-herdsman-converters#3665](https://github.com/Koenkk/zigbee-herdsman-converters/pull/3665) Here is pool request for the documentation : [Koenkk/zigbee2mqtt.io#1135](https://github.com/Koenkk/zigbee2mqtt.io/pull/1135) (you can correct my French English if needed into the documentation lol) All is tested and run.
> 
> can you please validate the request on master ?

How does this work?

I got both Danalock and Danapad, but I can't get the pincode to work. If I add a pincode with the MQTT message or in the gut, it does not show up in the Danalock app, and does not work on the Danapad.

If I add a pincode in the Danalock app, the pincode is not shown in the mqtt response.