Koenkk/zigbee2mqtt #10737: [Problem]: Cant get Z2M to work on Debian Bullseye headless.

lgLindstrom, 2022-01-12:
### What happened?

I have a PC with Debian Bullseye running, headless.  Docker and Docker-Compose is also installed. Installation works as expected with hello-world. I am running on ThinkCentre PC

My docker-compose file looks like 

```
version: '3.8'
services:
  zigbee2mqtt:
    container_name: zigbee2mqtt
    restart: unless-stopped
    image: koenkk/zigbee2mqtt
    volumes:
      - ./zigbee2mqtt-data:/app/data
      - /run/udev:/run/udev:ro
    ports:
      - 8080:8080
    environment:
      - TZ=Europe/Berlin
    devices:
      - /dev/ttyACM0:/dev/ttUSB0
```

### What did you expect to happen?

When starting  the container I hade expected that the zigbee-herdsman should connect properly. Instead I get the following error.

```
zigbee2mqtt      | Zigbee2MQTT:info  2022-01-12 09:56:14: Logging to console and directory: '/app/data/log/2022-01-12.09-56-14' filename: log.txt
zigbee2mqtt      | Zigbee2MQTT:info  2022-01-12 09:56:14: Starting Zigbee2MQTT version 1.22.2 (commit #414c51f)
zigbee2mqtt      | Zigbee2MQTT:info  2022-01-12 09:56:14: Starting zigbee-herdsman (0.13.188)
zigbee2mqtt      | Zigbee2MQTT:error 2022-01-12 09:56:14: Error while starting zigbee-herdsman
zigbee2mqtt      | Zigbee2MQTT:error 2022-01-12 09:56:14: Failed to start zigbee
zigbee2mqtt      | Zigbee2MQTT:error 2022-01-12 09:56:14: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
zigbee2mqtt      | Zigbee2MQTT:error 2022-01-12 09:56:14: Exiting...
zigbee2mqtt      | Zigbee2MQTT:error 2022-01-12 09:56:14: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyACM0'
zigbee2mqtt      |     at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:146:28)
zigbee2mqtt      |     at SerialPort._error (/app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
zigbee2mqtt      |     at /app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12
zigbee2mqtt exited with code 1
```

Kind of obvious error message,,, but the USB stick is there.

### How to reproduce it (minimal and precise)

Copy and run docker-compose file in Debian Bullseye setup.

### Zigbee2MQTT version

1.22.2 (commit #414c51f)

### Adapter firmware version

My adapter worked during last summer when I used it in a RPi 4

### Adapter

My adapter worked during last summer when I used it in a RPi 4

### Debug log

info  2022-01-12 15:42:38: Logging to console and directory: '/app/data/log/2022-01-12.15-42-38' filename: log.txt
info  2022-01-12 15:42:38: Starting Zigbee2MQTT version 1.22.2 (commit #414c51f)
info  2022-01-12 15:42:38: Starting zigbee-herdsman (0.13.188)
error 2022-01-12 15:42:39: Error while starting zigbee-herdsman

Koenkk, 2022-01-12:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/discussions/10692

closed 2022-01-12 by Koenkk

lgLindstrom, 2022-01-13:
Thanks.
I don't know how many times I read that line without seeing anything....