Koenkk/zigbee2mqtt #10740: [Problem]: Hue wall switch binding issue

rhvs, 2022-01-12:
### What happened?

The Hue wall switch binds succesfully to a lamp, however it doesn't operate the lamp as expected (nothing happens). 

### What did you expect to happen?

I expected that (as I did before with the Hue dimmer switches) after binding, I would be able to operate the light directly.

### How to reproduce it (minimal and precise)

[image]

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20211217

### Adapter

Slaesh's CC2652RB stick

### Debug log

-

Tonio16, 2022-01-14:
Hello

Did you try without the coordinator in the binding list, like for the 324131092621?
https://www.zigbee2mqtt.io/devices/324131092621.html#philips-324131092621

Antoine

leoluk, 2022-01-15:
Unbinding the coordinator doesn't seem to do anything :/

Sniffer shows that binding a light get a "Success" reply and unbinding twice returns "No Entry" the second time - so it does update its binding table, but never sends any commands.

The device simply doesn't seem to support binding.

Related threads:

- https://github.com/Koenkk/zigbee-herdsman-converters/issues/2393
- https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4566

rhvs, 2022-01-16:
Not sure if it's related, but a dimmer switch has two endpoint (attached screenshots of the dimmer switch clusters and wall switch (1 endpoint). When binding the dimmer switch, the second endpoint is used.

[image]
[image]

Koenkk, 2022-01-16:
Could you check if this issue has been fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('../lib/tuya');

const definition = {
    zigbeeModel: ['RDM001', '9290030171'],
    model: '929003017102',
    vendor: 'Philips',
    description: 'Hue wall switch module CUSTOM',
    fromZigbee: [fz.battery, fz.hue_wall_switch_device_mode, fz.hue_wall_switch],
    exposes: [e.battery(), e.action(['left_press', 'left_press_release', 'right_press', 'right_press_release']),
        exposes.enum('device_mode', ea.ALL, ['single_rocker', 'single_push_button', 'dual_rocker', 'dual_push_button'])],
    toZigbee: [tz.hue_wall_switch_device_mode],
    configure: async (device, coordinatorEndpoint, logger) => {
        // const endpoint = device.getEndpoint(1);
        // await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'genOnOff', 'manuSpecificPhilips']);
        // await reporting.batteryPercentageRemaining(endpoint);
        // const options = {manufacturerCode: 0x100B, disableDefaultResponse: true};
        // await endpoint.write('genBasic', {0x0034: {value: 0, type: 48}}, options);
    },
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, go to the frontend, verify that the description of the device ends with CUSTOM (this means that external converter has been loaded correctly)
- now factory reset the hue wall switch (repair it)
- try to bind it.
- does it work now?

WernerRH21, 2022-01-19:
[image]
[image]
[image]

leoluk, 2022-02-06:
@Koenkk

Thanks! So, this is interesting.

Added the external converter, paired the switch and confirmed that it loads the CUSTOM converter.

The device continues to ignore the binding config - it maintains a binding table and I can bind/unbind things, it just doesn't do anything. However, when looking with a sniffer, you can see that it actually _does_ send OnOff broadcasts regardless of binds, but to a hardcoded group `0xd31d` (I haven't tested whether this happens only with the custom converter):

[image]

And indeed, creating a group with that ID and adding lights to it works perfectly! Even dimming works (with `single_push_button`).

When setting mode to `dual_rocker`, both channels work, but use the same group ID (not terribly useful). Presumably, the Hue bridge uses vendor-specific commands to properly configure these as well as things like scenes.

Switching lights on/off via the switch also won't update the entity status in z2m, presumably because it doesn't know anything about the "binding" and hasn't set up the right reportings. For my use case, this is the only thing that's missing.

Worth trying to remove the custom converter? What is the commented-out code supposed to do originally (other than configuring reporting battery status reporting, which indeed doesn't work)?

(actually switched from zha to z2m so I can try this - I'm definitely keeping z2m, it's so smooth - thanks!)

Koenkk, 2022-02-07:
@leoluk 

> Worth trying to remove the custom converter? What is the commented-out code supposed to do originally (other than configuring battery status reporting, which indeed doesn't work)?

Yes please check if this doesn't change the binding behaviour.

leoluk, 2022-02-07:
Can confirm that without the custom converter, it's back to sending key presses to the coordinator:

[image]

Koenkk, 2022-02-08:
@leoluk can you switch back to the external converter and remove the comments line-by-line to see what statement is stopping the switch from sending commands to the group? (https://github.com/Koenkk/zigbee2mqtt/issues/10740#issuecomment-1013833937)

leoluk, 2022-02-09:
@Koenkk Binding the reporting endpoints makes the difference. With this config, it still sends group commands:

```
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        //await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'genOnOff', 'manuSpecificPhilips']);
        await reporting.batteryPercentageRemaining(endpoint);
        const options = {manufacturerCode: 0x100B, disableDefaultResponse: true};
        await endpoint.write('genBasic', {0x0034: {value: 0, type: 48}}, options);
    },
```

Koenkk, 2022-02-09:
@leoluk can you find out which cluster? Start with `await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);

kegelmeier, 2022-03-08:
Anyhow, I can support here? 

Also trying to bind the Hue Wall Switch with a Hue Light Group with success, however, no effect while pressing.

cyrille74500, 2022-03-25:
same problem....

trollix, 2022-03-25:
Hello, same problem. I can Bind, but no effect when action

Daanoz, 2022-03-25:
From the other thread, it looks like the group address changes on each reset, anyone knows an easy way to get the last known group address, or do we need a sniffer for it?

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4566#issuecomment-1030950892

Daanoz, 2022-03-26:
Right, so it hooked up to one a group, and the switch can indeed turn on an off a group.

Ended up with these reports in the convertor
` await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'genOnOff'/*, 'manuSpecificPhilips'*/]);`

After removing the device from the network, resetting and rejoining. The following messages can be observer in debug mode on the logging when pressing a button:

[image]

The `15204` group id is the group that's gets toggled in this case, but it changes on every reset of the switch. There is also no difference between switch 1 and 2; both seem to trigger the same message towards the `15204` group id.

Daanoz, 2022-04-10:
So, I started sniffing on the Hue Bridge. Some key take away's:
- The Hue bridge doesn't use binding as far as I can tell, If I simply unplug the bridge the switch stops working. All messages from the switch also have as destination the bridge.
- There is a difference in payload for the p1 port and the p2 port, but not if it's connecting or disconnecting.

| Action | Sequence |
| -------| ---------- |
| Connect P1 | 0100003000210000 -> 50ms -> 0100003002210100 |
| Disconnect P1 | 0100003000210000 -> 50ms -> 0100003002210100 |
| Connect P2 | 0200003000210000 -> 50ms -> 0200003002210100 |
| Disconnect P2 | 0200003000210000 -> 50ms -> 0200003002210100 |

Payload doesn't change between single rocker and dual rocker mode; except that in single rocker mode the p2 port simply doesn't do anything.

As the bridge doesn't do binding, will it even ever work?

leoluk, 2022-05-11:
Still relevant

martijn-brant, 2022-06-28:
Are there any updates to this? I would like to remove ControllerX from my setup and just have these Hue modules talk natively via bindings.

moritz31, 2022-08-11:
have the same problem with ioBroker and zigbee .... @leoluk is there an easy way to find out the groupId, which is random and need to be found first

JBS5, 2022-11-17:
Any updates on the direct binding possibilities of the Hue Wall Switch?

bavo, 2022-11-18:
I can confirm this working for me on the Sonoff CC2652 stick
* zigbee2mqtt version:     1.28.2
* Coordinator type: zStack3x0
* Coordinator revision 20220219

Wall switch has a direct binding to a group of 3 hue spots

steve41200, 2022-11-18:
It always doesn't work for me.

Have u the directbinding AND the coordinator ? or just the directbinding ?
I try in the hue to bind with checkbox On/off

bavo, 2022-11-18:
See my edit, I was tricked by my Home Assistant setup into thinking this worked. It does not ....  :disappointed:

Tscherno, 2022-11-20:
Are the Hue Wall Switch bindable or not?

rhvs, 2022-11-20:
> Are the Hue Wall Switch bindable or not? 

With all the comments above this qualifies for the lazy Sunday morning award.

treeman, 2022-11-28:
This is still a problem by default, but it's possible to bind them so I'll reiterate the findings:

1. [Add a custom converter](https://github.com/Koenkk/zigbee2mqtt/issues/10740#issuecomment-1013833937).
2. [Find the hardcoded group id](https://github.com/Koenkk/zigbee2mqtt/issues/10740#issuecomment-1079653103).
3. Create a group with the above group id, and add your entities into that group, and bind the switch to that group.

And it should work.

Note that it seems that it's not possible to bind the second port to another group as they use the same group id.

itavero, 2022-11-29:
I would not be surprised if the module also support TouchLink binding. I vaguely recall that this is a requirement for the Zigbee certification (although sometimes manufacturers get around this by using hidden switches or strange button combinations, as they don't really want customers to use it).

Tscherno, 2022-11-29:
> Create a group with the above group id, and add your entities into that group, and bind the switch to that group.

I'm a complete beginner with Z2M. Is there a good starting point to Custom Converters?

Tscherno, 2022-12-02:
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = [...]

When i add this converter, i get the following error during startup:

```
node:internal/modules/cjs/loader:959
  throw err;
  ^
Error: Cannot find module '../lib/tuya'
Require stack:
- /app/dist/util/utils.js
- /app/dist/util/settings.js
- /app/index.js
    at Function.Module._resolveFilename (node:internal/modules/cjs/loader:956:15)
    at Function.Module._load (node:internal/modules/cjs/loader:804:27)
    at Module.require (node:internal/modules/cjs/loader:1028:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at /app/dist/util/externally-loaded.js:8:14
    at Script.runInContext (node:vm:141:12)
    at Script.runInNewContext (node:vm:146:17)
    at Object.runInNewContext (node:vm:306:38)
    at loadModuleFromText (/app/lib/util/utils.ts:148:8)
    at loadModuleFromFile (/app/lib/util/utils.ts:155:12)
```

kegelmeier, 2022-12-04:
I resolved this by also pulling this library externally

Original
`const tuya = require('../lib/tuya');`

External 
`const tuya = require('zigbee-herdsman-converters/lib/tuya');`

larsrhdk, 2022-12-22:
Have same problem, any solution to solve problem in HA.

daniel-jong, 2023-03-28:
Thanks for all the help, I got the group ID by using [these steps](https://github.com/Koenkk/zigbee2mqtt/issues/10740#issuecomment-1328649057) ultimately making these hue wall switches I had laying in a drawer usefull :+1: 

Like said above; the converter has changed a bit since Koen's post of Januari 2022. 
I used the below, that is based on the imports in the [normal converter](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/philips.js)

```javascript
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const ota = require('zigbee-herdsman-converters/lib/ota');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const philips = require('zigbee-herdsman-converters/lib/philips');
const e = exposes.presets;
const ea = exposes.access;

const extend = {switch: require('zigbee-herdsman-converters/lib/extend').switch};

const definition = {
	zigbeeModel: ['RDM001', '9290030171'],
	model: '929003017102',
	vendor: 'Philips',
	description: 'Hue wall switch module CUSTOM',
	fromZigbee: [fz.battery, fz.hue_wall_switch_device_mode, fz.hue_wall_switch],
	exposes: [
		e.battery(), e.action(['left_press', 'left_press_release', 'right_press', 'right_press_release',
			'left_hold', 'left_hold_release', 'right_hold', 'right_hold_release']),
		exposes.enum('device_mode', ea.ALL, ['single_rocker', 'single_push_button', 'dual_rocker', 'dual_push_button'])],
	toZigbee: [tz.hue_wall_switch_device_mode],
	configure: async (device, coordinatorEndpoint, logger) => {
		const endpoint = device.getEndpoint(1);
		// await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'genOnOff', 'manuSpecificPhilips']);
		await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'genOnOff' /*, 'manuSpecificPhilips' */]);
		await reporting.batteryPercentageRemaining(endpoint);
		const options = {manufacturerCode: 0x100B, disableDefaultResponse: true};
		await endpoint.write('genBasic', {0x0034: {value: 0, type: 48}}, options);
	}
};
module.exports = definition;
```

martijn-brant, 2023-03-28:
> Thanks for all the help, I got the group ID by using [these steps](https://github.com/Koenkk/zigbee2mqtt/issues/10740#issuecomment-1328649057) ultimately making these hue wall switches I had laying in a drawer usefull 👍
> 
> Like said above; the converter has changed a bit since Koen's post of Januari 2022. I used the below, that is based on the imports in the [normal converter](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/philips.js)
> 
> ```js
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const ota = require('zigbee-herdsman-converters/lib/ota');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const philips = require('zigbee-herdsman-converters/lib/philips');
> const e = exposes.presets;
> const ea = exposes.access;
> 
> const extend = {switch: require('zigbee-herdsman-converters/lib/extend').switch};
> 
> const definition = {
> 	zigbeeModel: ['RDM001', '9290030171'],
> 	model: '929003017102',
> 	vendor: 'Philips',
> 	description: 'Hue wall switch module CUSTOM',
> 	fromZigbee: [fz.battery, fz.hue_wall_switch_device_mode, fz.hue_wall_switch],
> 	exposes: [
> 		e.battery(), e.action(['left_press', 'left_press_release', 'right_press', 'right_press_release',
> 			'left_hold', 'left_hold_release', 'right_hold', 'right_hold_release']),
> 		exposes.enum('device_mode', ea.ALL, ['single_rocker', 'single_push_button', 'dual_rocker', 'dual_push_button'])],
> 	toZigbee: [tz.hue_wall_switch_device_mode],
> 	configure: async (device, coordinatorEndpoint, logger) => {
> 		const endpoint = device.getEndpoint(1);
> 		// await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'genOnOff', 'manuSpecificPhilips']);
> 		await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'genOnOff' /*, 'manuSpecificPhilips' */]);
> 		await reporting.batteryPercentageRemaining(endpoint);
> 		const options = {manufacturerCode: 0x100B, disableDefaultResponse: true};
> 		await endpoint.write('genBasic', {0x0034: {value: 0, type: 48}}, options);
> 	}
> };
> module.exports = definition;
> ```

Does this mean you can toggle/hold-for-brightness without any controller active? And is it also still bound to the controller? In some cases, I wish to have the switch do the lighting for a room but also control something else like a window cover or ventilation.

martijn-brant, 2023-03-28:
I just tested this, I had to re-pair my Hue wall switch in order to get a valid group ID out of it (would report 0 before re-pairing). This now works with a test lamp group however, both buttons control the same group. Any thoughts on how we can use the 2nd button for a different light/group/automation?

bjornarfjelldal, 2023-03-29:
> I just tested this, I had to re-pair my Hue wall switch in order to get a valid group ID out of it (would report 0 before re-pairing). This now works with a test lamp group however, both buttons [...]

100% this! Binding both inputs separately (they send different signals to Z2M when not bindes) would be absolutely great

JBS5, 2023-03-29:
> Also, it no longer reports to the controller at all, for all switches of the same type. I've used some buttons of some hue wall switches bound to non zigbee entities / automation.

Do you mean that other existing Hue Wall Switches doesn't report to the coordinator when this external convertor is added?

martijn-brant, 2023-03-29:
> Do you mean that other existing Hue Wall Switches doesn't report to the coordinator when this external convertor is added?

Indeed, all wall hue switches of that type will no longer report to the controller directly when you use that code. You'll see some of the traffic in your logs but not the actions in mqtt.

martijn-brant, 2023-06-20:
Has anyone found a way for the Hue in-wall switches to talk peer-to-peer on the Zigbee network (binding?) so that the switches will always work even when the Zigbee controller/switching logic is down? I'll be willing to post up a feature bounty if that's allowed by @Koenkk.

I've had the PSU of my NUC running HA/Z2M get fried and I couldn't switch my lights off until I replaced it. This is of course a big flaw in my HA setup, being reliant on Zigbee, Z2M and some kind of switching logic (appdaemon in my case).

bjornarfjelldal, 2023-06-28:
> 

This is my motivation as well.. :) +1

Koenkk, 2023-06-28:
I didn't read all the posts here before, but:

Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/167e75c57604a6fbd698cf9661e36b24

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m
- re-pair the device (such that it factory resets)
- see if you can now bind it.

ghouscht, 2023-06-28:
> I didn't read all the posts here before, but:
>
> Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/167e75c57604a6fbd698cf9661e36b24
>
>     * [...]

No, does not seem to work in my case but I can see that the external converter is used.

martijn-brant, 2023-06-29:
@Koenkk I've also did some testing. I added the external converter (and added CUSTOMTEST to the name so I can spot the new code in the logs), restarted Z2M, rejoined my test Hue in-wall switch, unbind it from the controller (also tested without this step) and bound the switch to a group with a light. It seems to work but there is no effect to the actual group/light. I've also tried binding it to a light (no effect, no error) and adding it to a group (got an error).

I've put logging on debug and posted it here: https://gist.github.com/martijn-brant/e58dd2be2e742c001466db7d4b98e42f

martijn-brant, 2023-06-29:
I should add: 

Zigbee2MQTT version
1.31.2-dev commit: [61dd1dd](https://github.com/Koenkk/zigbee2mqtt/commit/61dd1dd)
Coordinator type
zStack3x0
Coordinator revision
20221226

ghouscht, 2023-06-30:
> @ghouscht you did re-pair the device?

Yes I did remove the device from the coordinator, factory resetted the device and rejoined it.

Koenkk, 2023-06-30:
I've ordered the device so I can debug the issue locally.

Koenkk, 2023-07-09:
I did some analysis on the behaviour of this device when paired to the Hue bridge. Unfortunately they don't use binding here. The remote sends a manufacturer specific command to the bridge, the bridge then sends a standard Zigbee command to the group:

[image]

The bridge also allows to change the behaviour of the buttons:

[image]

[image]

When changing these nothing gets send to the switch, which means all this logic is handled on the bridge side.

I will continue with checking if we somehow can get binding working when paired to z2m.

martijn-brant, 2023-07-09:
Thanks for your efforts Koen, I had figured as much as the switch is useless when the official Hue Bridge is down. They designed this switch to be as dumb as possible and assume the bridge will work 24/7.

rhdny, 2023-07-09:
> I've had the PSU of my NUC running HA/Z2M get fried and I couldn't switch my lights off until I replaced it. This is of course a big flaw in my HA setup, being reliant on Zigbee, Z2M and some kind of switching logic (appdaemon in my case).

This ^^^

I have a large number of hue lights and while HA is pretty stable, I worry about being on the road for work and the fam unable to control lighting particularly in the remote area we're in. It's further complicated by the fact that I have the bulbs set to return to prior state in the event of a power loss as those happen regularly out here. 

I'm actually a Deconz/Phoscon user with a Conbee II stick and I ended up here while researching this issue. 

I was ultimately able to solve this in a simple manner by using an alternate version of the Phoscon software that they keep available via a link on the help page of the current application. 

It allowed me to bind the hue switch directly to a group of lights. Using the newer hue switch, by default the top button is an on/off toggle and the brightness buttons work as well. The bottom button is not used. I tested the above with HA shut down. 

With HA up, the above events are reported under deconz_events so I can see the actions there and handle them properly (accidental trigger in daylight, shutdown after time without activity, etc). 

The remaining button actions are also available for regular HA use (bottom button, long press on all the buttons). 

So with some quick training of the fam, I'm in business. I was subscribed to this thread and an update today reminded me to circle back and share my journey through this.

martijn-brant, 2023-07-09:
> This ^^^
>
> I have a large number of hue lights and while HA is pretty stable, I worry about being on the road for work and the fam unable to control lighting particularly in the remote area we're in. [...]

Are you referring the hue wall magnet mounted switch or do you mean the in-wall switches? We're talking about the inwalls that have two contact, nothing more.

rhdny, 2023-07-09:
I was speaking of the on-wall vs the in-wall as evidenced by my description of the buttons. 

The tool I referenced above supports the in-wall as well. I haven't bought one as the in-wall has less functionality for my use case and is also battery dependent. If it had neutral support, I'd likely get some. 

I'm traveling a lot but if possible, I may get an in-wall to confirm the above outcome. Will circle back here if so.

bjornarfjelldal, 2023-07-10:
This is awesome dedication, @Koenkk! Thanks a lot for your efforts. While you now have the device, is it possible to also investigate why it, in many cases, depletes the battery in a manner of days, as in #12545 and at least another issue number which I can't find at the moment, as well as https://community.home-assistant.io/t/battery-drain-philips-hue-wall-switch-modules/518501.

They are supposed to last 5 years, not 5 days.. :)

[image]

Koenkk, 2023-07-10:
As written earlier in this thread, it is possible to make this device send toggle commands to Zigbee groups. To enable this some fixes where required. These are available in the latest-dev branch now. I documented how to accomplish the group binding: https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/devices/929003017102.md#directly-control-zigbee-groups

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Note: this will probably break the battery reporting, something in the frontend needs to be fixed first (since `genPowerCfg` shouldn't be unbound). I will fix this later and update the docs.

@bjornarfjelldal regarding battery, please create a new issue for this (to not clutter this topic)

martijn-brant, 2023-07-10:
> As written earlier in this thread, it is possible to make this device send toggle commands to Zigbee groups. To enable this some fixes where required. These are available in the latest-dev branch [...]

To clarify.. this only works for one of the input/side of switch, right?

Koenkk, 2023-07-13:
@martijn-brant yes, I've updated the docs with this. 

> Note: this will probably break the battery reporting, something in the frontend needs to be fixed first (since genPowerCfg shouldn't be unbound). I will fix this later and update the docs.

This will be fixed after https://github.com/nurikk/zigbee2mqtt-frontend/pull/1754 is merged (already updated the docs)

I guess that after merging this issue can be closed.

Koenkk, 2023-07-22:
https://github.com/nurikk/zigbee2mqtt-frontend/pull/1754 is merged now, so I assume I can close this now.

closed 2023-07-22 by Koenkk

burpy, 2023-07-29:
Sorry for reopen this issue.

I migrated from the Philips Hue bridge and a DECONZ Stick with Phoscon to Zigbee2MQTT on a Raspberry with a Zigbee 3.0 Sonoff USB Dongle.

It works very well with the Hue wall switch module and the workaround above to bind the Hue wall switch module with a group of lamps. As you mentioned the binding does not work with two outputs if I use two inputs in the wall switch mondule. Is there a change to get it work in future? At the moment this is the only reason to keep my Philips Hue Bridge alive.

This would be great!

JBS5, 2024-02-01:
The current docs (https://www.zigbee2mqtt.io/devices/929003017102.html#directly-control-zigbee-groups) tells me to add the Hue Wall Switch to the created group, but in the past it was neccessary to create a group with the Group ID which was triggered by the Hue Wall Switch. 

When adding the Hue Wall Switch to the created group, I get an error (even after waking up the device by pressing it one or multiple times).

[image]

Does anyone recognize this?

marazmarci, 2024-02-01:
> The current docs (https://www.zigbee2mqtt.io/devices/929003017102.html#directly-control-zigbee-groups) tells me to add the Hue Wall Switch to the created group, but in the past it was neccessary to [...]

Yes, the same happens for me, but if after this you check what group ID the device sends commands to, you'll see that it is now the group you tried to add the device to 🙂

Please confirm that it works also for you like this, and then I'll update the documentation!

JBS5, 2024-02-01:
> > The current docs (https://www.zigbee2mqtt.io/devices/929003017102.html#directly-control-zigbee-groups) tells me to add the Hue Wall Switch to the created group, but in the past it was neccessary to create a group with the Group ID which was triggered by the Hue Wall Switch.
> > When adding the Hue Wall Switch to the created group, I get an error (even after waking up the device by pressing it one or multiple times).
> > ![chrome_LZOFGi5Y0O](https://private-user-images.githubusercontent.com/25594213/301473806-f95024ea-4abd-47d2-8437-ac75fff4dd4b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY3ODY2ODAsIm5iZiI6MTcwNjc4NjM4MCwicGF0aCI6Ii8yNTU5NDIxMy8zMDE0NzM4MDYtZjk1MDI0ZWEtNGFiZC00N2QyLTg0MzctYWM3NWZmZjRkZDRiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjAxVDExMTk0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFlODg4OWZlNDdkNTQ3MjYzODhjYzE5YzEzOTRjYjc2YzkzZTQyNmQ0YjAwMzRjMjViODE1OGY2NTk1ZDlhMzkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.RrQTzrEAJg8TXxEmwD3UcFYCOOwe3d0jCKheKIN0REY)
> > Does anyone recognize this?
> 
> Yes, the same happens for me, but if after this you check what group ID the device sends commands to, you'll see that it is now the group you tried to add the device to 🙂
> 
> Please confirm that it works also for you like this, and then I'll update the documentation!

Yes, you are right. I do see the groupID of the created group in the log when I trigger the Hue Wall Switch by pressing the button.
But, not all the time, when I press e.g. 10 times, I see ~6-8x the log message like below. Not sure why, yet.

```
debug 2024-02-01 12:48:10: Received Zigbee message from 'Wandschakelaar WC boven', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 15
info  2024-02-01 12:48:10: MQTT publish: topic 'zigbee2mqtt/Wandschakelaar WC boven', payload '{"action":"toggle","action_group":15,"battery":100,"device_mode":null,"linkquality":75}'
info  2024-02-01 12:48:10: MQTT publish: topic 'zigbee2mqtt/Wandschakelaar WC boven', payload '{"action":"","battery":100,"device_mode":null,"linkquality":75}'
info  2024-02-01 12:48:10: MQTT publish: topic 'zigbee2mqtt/Wandschakelaar WC boven/action', payload 'toggle'
```

[3 more replies]