Koenkk/zigbee2mqtt #10741: [Problem]: YSR-MINI-01_rgbcct does not turn on when triggered by HA-scenes

chhofi, 2022-01-12:
### What happened?

When you create two scenes in HA where 

**scene one** just switches the YSR-MINI-01_rgbcct off and 
**scene two** switches YSR-MINI-01_rgbcct on, to blue, 100% brightness, and you trigger first scene one and than scene two than nothing will happen.

When you "manually" turn on and off the light via HA-Dashboard, the light turns on and off.

When looking at the send MQTT-Messages the off command is always the same and works great:

{"state": "OFF"}

When I manually switch on the light via the HA-UI the following message gets send:

{"state": "ON"}

also working great.

When the light is already on I can change the color without any issue:

{"state": "ON", "color": {"x": 0.527, "y": 0.447}}

but what's not possible is switching the light off and than sending an on command with the desired color at ones, like the scene in HA does:

{"state": "ON", "color": {"x": 0.581, "y": 0.245}, "brightness": 254}

-> lights stayes switched off

 

### What did you expect to happen?

YSR-MINI-01_rgbcct turning from off to on with a bright blue color.

### How to reproduce it (minimal and precise)

Create two scenes in HA. 

Scene one: Just turn YSR-MINI-01_rgbcct off
Scene two: Turn YSR-MINI-01_rgbcct on and give it some color

Than just trigger scene one and after that scene two.

### Zigbee2MQTT version

1.22.2-dev

### Adapter firmware version

20201026

### Adapter

Silicon_Labs_slae.sh_cc2652rb_stick

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/7856993/log.txt)

chhofi, 2022-03-15:
Still an issue :/

platte-73, 2022-04-11:
I have the same issue with this device in Domoticz

Can manually switch device on and off but when i put it in a scene device is not switching on

robertklep, 2022-04-12:
Similar issue with YSR-MINI-01_dimmer.

I don't use scenes but sending state and brightness from Home Assistant:
```
service: light.turn_on
entity_id: …
data:
  brightness_pct: 50
```

robertklep, 2022-04-13:
After a bit more experimenting, it looks like this device has some serious flaws.

When I turn off the device, most of the time it seems it reports it's off even though it's not. I'm now trying to turn the device off by turning it on with a brightness of 0, which seems to work better. I'll have to see how well it works in the long run.

robertklep, 2022-04-14:
Here's an extension that seems to fix the issue for me. I only have a YSR-MINI-01_dimmer device so not sure if the other devices have the exact same issue.

```
const stringify = require('json-stable-stringify-without-jsonify');

// Add model names that should be fixed to this array.
const WHITELIST = [ 'YSR-MINI-01_dimmer', 'YSR-MINI-01_rgbcct', 'YSR-MINI-01_wwcw' ];

module.exports = class FixYSRSAIExtension {
  constructor(zigbee, mqtt, state, publishEntityState, eventBus, settings, logger) {
    this.zigbee             = zigbee;
    this.mqtt               = mqtt;
    this.state              = state;
    this.publishEntityState = publishEntityState;
    this.eventBus           = eventBus;
    this.settings           = settings;
    this.logger             = logger;
  }

  start() {
    const baseTopic = this.settings.get().mqtt.base_topic;
    
    this.eventBus.onMQTTMessage(this, ({ topic, message }) => {
      try {
        if (! topic.startsWith(baseTopic)) return;
        
        // extract friendly name
        const friendlyName = topic.match(/([^\/]+)\/set$/)?.[1];
        if (! friendlyName) return;

        // resolve target device and check if it's whitelisted (meaning the message might need to be fixed)
        const device = this.zigbee.resolveEntity(friendlyName);
        if (! device || ! WHITELIST.includes(device._definition?.model)) return;
        
        // check the payload to see if it needs to be fixed
        const payload = JSON.parse(message);
        if (payload.state === 'OFF' && ! ("transition" in payload)) {
          this.logger.info(`[fix-ysrsai-extension] fixing state for "${ friendlyName }"`);
          this.mqtt.onMessage(`${ baseTopic }/${ friendlyName }/set`, stringify({ state: 'OFF', transition: 0 }));
        }
      } catch(e) {
        this.logger.error(`[fix-ysrsai-extension] error: ${ e.stack }`);
      }
    });
  }

  stop() {
    this.eventBus.removeListeners(this);
  }
}
```

Mordevolt, 2022-04-19:
@robertklep Thank you a lot! seems fix behavior of my YSR-MINI-01_rgbcct

platte-73, 2022-04-20:
hello Robert,

I’m using domoticz. Where do I place this file or which file do I have to replace

> Here's an extension that seems to fix the issue for me. I only have a YSR-MINI-01_dimmer device so not sure if the other devices have the exact same issue.
>
> ```
> const stringify = [...]

robertklep, 2022-04-20:
@platte-73 you can load extensions in the zigbee2mqtt web frontend:
[image]

Press the green + button, give the script a name (I named it `fix-ysrsai.js` but I don't think the name matters), paste the script I posted above, and press "Save". It should work automatically.

robertklep, 2022-05-09:
Here's an updated version of the extension that works around another issue I found with the dimmer: when it's on and you want to set the brightness to a value _lower_ than its current value, it'll turn off. The workaround consists of sending `brightness_move: 0` to the device once during the time it's on (no idea why that fixes it, though).

```
const stringify = require('json-stable-stringify-without-jsonify');

// Add model names that should be fixed to this array.
const WHITELIST = [ 'YSR-MINI-01_dimmer', 'YSR-MINI-01_rgbcct', 'YSR-MINI-01_wwcw' ];

module.exports = class FixYSRSAIExtension {
  constructor(zigbee, mqtt, state, publishEntityState, eventBus, settings, logger) {
    this.mqtt          = mqtt;
    this.eventBus      = eventBus;
    this.logger        = logger;
    this.mqttBaseTopic = settings.get().mqtt.base_topic;
    this.fixed         = {};
  }

  start() {
    this.eventBus.onStateChange(this, ({ entity, update, from, to }) => {
      if (! WHITELIST.includes(entity._definition?.model)) return;
      this.logger.debug(`entity=${ entity.name } from='${ stringify(from) }' to='${ stringify(to) }'} update='${ stringify(update) }'`);

      if (to.state === 'ON' && !( entity.name in this.fixed )) {
        this.logger.debug(`[fix-ysrsai-extension] fixing "${ entity.name }"`);
        this.fixed[entity.name] = true;
        this.mqtt.onMessage(`${ this.mqttBaseTopic }/${ entity.name }/set`, stringify({ state : 'ON', brightness_move: 0 }));
        this.mqtt.onMessage(`${ this.mqttBaseTopic }/${ entity.name }/set`, stringify(to));
      } else if (to.state === 'OFF' && to.brightness !== 0) {
        this.logger.debug(`[fix-ysrsai-extension] fixing OFF state for "${ entity.name }"`);
        delete this.fixed[entity.name];
        this.mqtt.onMessage(`${ this.mqttBaseTopic }/${ entity.name }/set`, stringify({ state: 'ON', brightness: 0 }));
      }
    });
  }

  stop() {
    this.eventBus.removeListeners(this);
  }
}
```

platte-73, 2022-06-09:
But mij front end of zigbee2mqtt doesn’t have that option 

[image]

robertklep, 2022-06-10:
@platte-73 that's not the z2m frontend but (I assume) a Domoticz dashboard to manage z2m.

platte-73, 2022-06-23:
> > @platte-73 you can load extensions in the zigbee2mqtt web frontend: [image]
> > Press the green + button, give the script a name (I named it `fix-ysrsai.js` but I don't think the name matters), paste the script I posted above, and press "Save". It should work automatically.
> 
> I use domoticz and don’t have the option load extension.

Code doesn't work for me. Still the same. When I use the slider to turn on and lower the brightness it turns off
My controller is YSR-MINI-01_rgbcct

robertklep, 2022-06-23:
These devices kept giving me issues so I stopped using them. Now using GLEDOPTO RGB+CCT controllers, only a bit more expensive but a lot less quirky.

platte-73, 2022-06-23:
Thank you for your quick reply. I’m going to order also that one

platte-73, 2022-06-26:
Thanks. I also ordered

closed 2022-08-04 by github-actions[bot]