Koenkk/zigbee2mqtt #10742: [New device support]: Schneider Electric - Wiser shutter control - NU350818

cavallaroe, 2022-01-12:
### Link

https://www.se.com/ww/en/product/NU350818/new-unica-wiser-blind-control-switch-white/

### Database entry

{"type":"Router", "ieeeAddr":"0x680ae2fffe116e49", "nwkAddr":"0xF613", "Zigbee model": "NHPB/SHUTTER/1", "Firmware version": "002.004.016 R", "Model": "S520567"}

### Comments

I tried to follow the support new device page but I am not able to follow it step-by-step as I am using the Zigbee2mqtt addon for Home Assistant and it seems to be somehow different there.
The device I am trying to configure is a shutter control swithc from Schneider Electric from the Wiser Unica series, and I am using the Zigbee2mqtt addon (v 1.22.2-1) for Home Assistant installed on a Raspberry Pi with a [CC2652R1 adapter](https://shop.electrolama.com/collections/usb-rf-sticks/products/zzh-multiprotocol-rf-stick).
The device was found by the zigbee2mqtt and successfully paired but it is identified as a [different model](https://www.zigbee2mqtt.io/devices/S520567.html#schneider%2520electric-s520567). The available commands (open, close, stop) work correctly also for this device, but other configurations are not available (duration, led behaviour)

I will be happy to contribute in achieving full support for this device but I will need some help from the community.

fariouche, 2022-01-17:
I'm using them and they work perfectly fine

cavallaroe, 2022-01-18:
Hi fariouche, it is working for me as well. But I would like to set the drive time and turn off the led and I don't see these options in the configuration for the S520567 shutter

fariouche, 2022-01-18:
Oh, I see an error in the integration of the lift duration feature @Koenkk 
In schneider_electric.js, tzLocal.lift_duration was only added to 'PUCK/SHUTTER/1' and missing for 'NHPB/SHUTTER/1'

@cavallaroe , I'm not sure that the led can be controlled

cavallaroe, 2022-01-18:
Thanks fariouche,
The led settings can be modified when connnected via bluetooth to the SE Wiser mobile app, so I expect it can be done with zigbee as well. But I may be wrong

fariouche, 2022-01-19:
I have the wiser hub home touch, and there is no led option. So either the led cannot be controlled over zigbee or they didn't offer the option in the home touch and in that case we have now way to know the command to send

cavallaroe, 2022-01-20:
Ok, I see. So the led settings cannot be changed but the drive duration yes. 
You already identified what is missing, should I add it on my local version or there will be an update?

fariouche, 2022-01-20:
until @Koenkk  add it to the NHPB/SHUTTER/1 device in zigbee-herdsman, you can use an external converter (see https://github.com/Koenkk/zigbee2mqtt/issues/10019) or modify schneider_electric.js in herdsman/devices to add localTz like PUCK/SHUTTER/1

Vebryn, 2022-01-31:
Also interested 👍

Vebryn, 2022-01-31:
> until @Koenkk add it to the NHPB/SHUTTER/1 device in zigbee-herdsman, you can use an external converter (see #10019) or modify schneider_electric.js in herdsman/devices to add localTz like PUCK/SHUTTER/1

Is external converter specified into #10019 compatible with NHPB/SHUTTER/1 ?

fariouche, 2022-01-31:
the converter in #10019 is for NHPB/SHUTTER/1 but seem to be the same as PUCK/SHUTTER/1

closed 2022-04-02 by github-actions[bot]