Koenkk/zigbee2mqtt #10743: [Problem]: Not all Zigbee events get received

ioulian, 2022-01-12:
### What happened?

Hi,

I'm using Z2M addon for HomeAssistant and using Ikea Tradfri devices. I have bound the [remote controller] (https://www.zigbee2mqtt.io/devices/E1524_E1810.html#ikea-e1524%252Fe1810) directly to my 5 [lightbulbs](https://www.zigbee2mqtt.io/devices/LED1545G12.html#ikea-led1545g12) in 1 room (so I can still use the controller without running HomeAssistant).

Everything is bound "correctly". When I use the remote control, the lights do switch on and off (all 5 of them). But the state in the frontend is not updated correctly for at least 1 light. Looking at the debug logs, I'm almost always missing at least 1 light.

### What did you expect to happen?

When turning all lights on, the state should be reflected in the frontend. In the debug log you see that "topic 'zigbee2mqtt/office_top_light_2'" is missing, sometimes it's 3, sometimes it's another one. Same goes when turning off the lights.

Is there a maximum number of devices that can send events simultaneously?

### How to reproduce it (minimal and precise)

- I have added 6 devices to Zigbee2MQTT: Ikea Remote Control and 5 Ikea lights
- I have bound all lights to the Remote control with clusters: "LevelCtrl" and "OnOff".
- Click on the OnOff button on the remote control and watch the "Dashboard" in the frontend (or logs)

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

26720700

### Adapter

Conbee 2

### Debug log

`Debug Received Zigbee message from 'office_remote_control', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/office_remote_control', payload '{"action":"toggle","battery":74,"linkquality":140,"update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/office_remote_control', payload '{"action":"","battery":74,"linkquality":140,"update":{"state":"idle"},"update_available":false}'
Info MQTT publish: topic 'zigbee2mqtt/office_remote_control/action', payload 'toggle'
Debug Received Zigbee message from 'office_stand_light', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/office_stand_light', payload '{"brightness":254,"color_mode":"color_temp","color_temp":454,"linkquality":255,"power_on_behavior":"previous","state":"ON","update":{"state":"idle"},"update_available":false}'
Debug Received Zigbee message from 'office_small_light', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/office_small_light', payload '{"brightness":254,"color_mode":"color_temp","color_temp":454,"linkquality":255,"power_on_behavior":"previous","state":"ON","update":{"state":"idle"},"update_available":false}'
Debug Received Zigbee message from 'office_top_light_1', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/office_top_light_1', payload '{"brightness":254,"color_mode":"color_temp","color_temp":454,"linkquality":255,"power_on_behavior":"previous","state":"ON","update":{"state":"idle"},"update_available":false}'
Debug Received Zigbee message from 'office_top_light_3', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/office_top_light_3', payload '{"brightness":254,"color_mode":"color_temp","color_temp":454,"linkquality":255,"power_on_behavior":"previous","state":"ON","update":{"state":"idle"},"update_available":false}'`

sjorge, 2022-01-12:
Check the reporting tab, reporting may be missing for the one bulb that is not updating

ioulian, 2022-01-13:
Thanks for the reply, but no, reporting is turned on for all the bulbs for OnOff (0, 3600, 0) (> min, max, change)
I have unbound one of the lights, to see if it works better with (5 events = 1 controller + 4 bulbs) and it much better, but still, sometimes one bulb does not refresh it's state = no event get's received.

Maybe Conbee 2 can't handle more than x events in such a short time? Also when I turn off all lights in the room, not all lights get turned off, so sending multiple zigbee events also fails if they are sent simultaneously...

The link quality is 255, as everything is located in one room. So maybe is the adapter too slow? I have found an advanced setting: "Adapter concurrency", but leaving it on null, or setting to 2 or 16 did not help.

ioulian, 2022-01-13:
I have downgraded the conbee 2 stick to a version (26580700) seen here: #https://github.com/Koenkk/zigbee2mqtt/issues/9554#issuecomment-1004030668 but this doesn't change much

ioulian, 2022-01-13:
Update: I've added all my lights in a group (Zigbee2MQTT group) and it mostly works fine now. There are cases some of the lights do not get the correct state, but in 90% of the time it's fine.

closed 2022-02-20 by github-actions[bot]