Koenkk/zigbee2mqtt #10744: [Problem]: Trying to add supported Devices but getting unsupported message

generalpax, 2022-01-12:
### What happened?

I'm trying to add a new Philips Hue Light Bulp to my zigbee2mqtt (LTV002) . Device is joined but unsupported. I checked the herdsman external converters and device should be supported on current version.

[image]

### What did you expect to happen?

Device can be used.

### How to reproduce it (minimal and precise)

- Factory reset device via Hue App on Android.
- Join device to zigbee2mqtt

### Zigbee2MQTT version

1.21.2

### Adapter firmware version

0x26660700

### Adapter

ConBee2/RaspBee2

### Debug log

Zigbee2MQTT:debug 2022-01-12 22:11:17: Received MQTT message on 'homeassistant/sensor/0x00158d0006ec38e9/action_angle/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0006ec38e9"],"manufacturer":"Xiaomi","model":"Mi/Aqara smart home cube (MFKZQ01LM)","name":"0x00158d0006ec38e9","sw_version":"Zigbee2MQTT 1.21.2"},"enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/0x00158d0006ec38e9","name":"0x00158d0006ec38e9 action angle","state_topic":"zigbee2mqtt/0x00158d0006ec38e9","unique_id":"0x00158d0006ec38e9_action_angle_zigbee2mqtt","value_template":"{{ value_json.action_angle }}"}'
Zigbee2MQTT:debug 2022-01-12 22:11:17: Received MQTT message on 'homeassistant/sensor/0x00158d0006ec38e9/action_from_side/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0006ec38e9"],"manufacturer":"Xiaomi","model":"Mi/Aqara smart home cube (MFKZQ01LM)","name":"0x00158d0006ec38e9","sw_version":"Zigbee2MQTT 1.21.2"},"enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/0x00158d0006ec38e9","name":"0x00158d0006ec38e9 action from side","state_topic":"zigbee2mqtt/0x00158d0006ec38e9","unique_id":"0x00158d0006ec38e9_action_from_side_zigbee2mqtt","value_template":"{{ value_json.action_from_side }}"}'
Zigbee2MQTT:debug 2022-01-12 22:11:17: Received MQTT message on 'homeassistant/sensor/0x00158d0006ec38e9/action_side/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0006ec38e9"],"manufacturer":"Xiaomi","model":"Mi/Aqara smart home cube (MFKZQ01LM)","name":"0x00158d0006ec38e9","sw_version":"Zigbee2MQTT 1.21.2"},"enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/0x00158d0006ec38e9","name":"0x00158d0006ec38e9 action side","state_topic":"zigbee2mqtt/0x00158d0006ec38e9","unique_id":"0x00158d0006ec38e9_action_side_zigbee2mqtt","value_template":"{{ value_json.action_side }}"}'
Zigbee2MQTT:debug 2022-01-12 22:11:17: Received MQTT message on 'homeassistant/sensor/0x00158d0006ec38e9/action_to_side/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0006ec38e9"],"manufacturer":"Xiaomi","model":"Mi/Aqara smart home cube (MFKZQ01LM)","name":"0x00158d0006ec38e9","sw_version":"Zigbee2MQTT 1.21.2"},"enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/0x00158d0006ec38e9","name":"0x00158d0006ec38e9 action to side","state_topic":"zigbee2mqtt/0x00158d0006ec38e9","unique_id":"0x00158d0006ec38e9_action_to_side_zigbee2mqtt","value_template":"{{ value_json.action_to_side }}"}'
Zigbee2MQTT:debug 2022-01-12 22:11:17: Received MQTT message on 'homeassistant/sensor/0x00158d0006ec38e9/action/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0006ec38e9"],"manufacturer":"Xiaomi","model":"Mi/Aqara smart home cube (MFKZQ01LM)","name":"0x00158d0006ec38e9","sw_version":"Zigbee2MQTT 1.21.2"},"enabled_by_default":true,"icon":"mdi:gesture-double-tap","json_attributes_topic":"zigbee2mqtt/0x00158d0006ec38e9","name":"0x00158d0006ec38e9 action","state_topic":"zigbee2mqtt/0x00158d0006ec38e9","unique_id":"0x00158d0006ec38e9_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
Zigbee2MQTT:debug 2022-01-12 22:11:17: Received MQTT message on 'homeassistant/sensor/0x00158d0006ec38e9/linkquality/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0006ec38e9"],"manufacturer":"Xiaomi","model":"Mi/Aqara smart home cube (MFKZQ01LM)","name":"0x00158d0006ec38e9","sw_version":"Zigbee2MQTT 1.21.2"},"enabled_by_default":false,"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x00158d0006ec38e9","name":"0x00158d0006ec38e9 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0x00158d0006ec38e9","unique_id":"0x00158d0006ec38e9_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Zigbee2MQTT:debug 2022-01-12 22:11:17: Received MQTT message on 'homeassistant/sensor/0x00158d0007d3c249/battery/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0007d3c249"],"manufacturer":"Xiaomi","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","name":"0x00158d0007d3c249","sw_version":"Zigbee2MQTT 1.21.2"},"device_class":"battery","enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/0x00158d0007d3c249","name":"0x00158d0007d3c249 battery","state_class":"measurement","state_topic":"zigbee2mqtt/0x00158d0007d3c249","unique_id":"0x00158d0007d3c249_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
Zigbee2MQTT:debug 2022-01-12 22:11:17: Received MQTT message on 'homeassistant/sensor/0x00158d0007d3c249/temperature/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0007d3c249"],"manufacturer":"Xiaomi","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","name":"0x00158d0007d3c249","sw_version":"Zigbee2MQTT 1.21.2"},"device_class":"temperature","enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/0x00158d0007d3c249","name":"0x00158d0007d3c249 temperature","state_class":"measurement","state_topic":"zigbee2mqtt/0x00158d0007d3c249","unique_id":"0x00158d0007d3c249_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.temperature }}"}'
Zigbee2MQTT:debug 2022-01-12 22:11:17: Received MQTT message on 'homeassistant/sensor/0x00158d0007d3c249/humidity/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0007d3c249"],"manufacturer":"Xiaomi","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","name":"0x00158d0007d3c249","sw_version":"Zigbee2MQTT 1.21.2"},"device_class":"humidity","enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/0x00158d0007d3c249","name":"0x00158d0007d3c249 humidity","state_class":"measurement","state_topic":"zigbee2mqtt/0x00158d0007d3c249","unique_id":"0x00158d0007d3c249_humidity_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.humidity }}"}'
Zigbee2MQTT:debug 2022-01-12 22:11:17: Received MQTT message on 'homeassistant/sensor/0x00158d0007d3c249/pressure/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0007d3c249"],"manufacturer":"Xiaomi","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","name":"0x00158d0007d3c249","sw_version":"Zigbee2MQTT 1.21.2"},"device_class":"pressure","enabled_by_default":true,"json_attributes_topic":"zigbee2mqtt/0x00158d0007d3c249","name":"0x00158d0007d3c249 pressure","state_class":"measurement","state_topic":"zigbee2mqtt/0x00158d0007d3c249","unique_id":"0x00158d0007d3c249_pressure_zigbee2mqtt","unit_of_measurement":"hPa","value_template":"{{ value_json.pressure }}"}'
Zigbee2MQTT:debug 2022-01-12 22:11:17: Received MQTT message on 'homeassistant/sensor/0x00158d0007d3c249/voltage/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00158d0007d3c249"],"manufacturer":"Xiaomi","model":"Aqara temperature, humidity and pressure sensor (WSDCGQ11LM)","name":"0x00158d0007d3c249","sw_version":"Zigbee2MQTT 1.21.2"},"device_class":"voltage","enabled_by_default":false,"json_attributes_topic":"zigbee2mqtt/0x00158d0007d3c249","name":"0x00158d0007d3c249 voltage","state_class":"measurement","state_topic":"zigbee2mqtt/0x00158d0007d3c249","unique_id":"0x00158d0007d3c249_voltage_zigbee2mqtt","unit_of_measurement":"mV","value_template":"{{ value_json.voltage }}"}'
[... cut ...]
2022-01-12T21:14:14.036Z zigbee-herdsman:deconz:frameParser device state: 10100010
2022-01-12T21:14:14.036Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2022-01-12T21:14:24.032Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 35
2022-01-12T21:14:24.037Z zigbee-herdsman:deconz:frameParser device state: 10100010
2022-01-12T21:14:24.038Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1

generalpax, 2022-01-12:
Zigbee2MQTT:debug 2022-01-12 22:19:05: Received Zigbee message from '0x001788010b806f29', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":16783362,"imageType":276,"manufacturerCode":4107}' from endpoint 11 with groupID null
Zigbee2MQTT:warn  2022-01-12 22:19:05: Received message from unsupported device with Zigbee model 'LTV002' and manufacturer name 'Signify Netherlands B.V.'
Zigbee2MQTT:warn  2022-01-12 22:19:05: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
2022-01-12T21:19:05.692Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2022-01-12T21:19:14.029Z zigbee-herdsman:deconz:driver send write parameter request from queue. seqNr: 82 paramId: 38 param: 600
2022-01-12T21:19:14.032Z zigbee-herdsman:deconz:frameParser write parameter response - parameter id: 38 - status: 0
2022-01-12T21:19:14.039Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 83
2022-01-12T21:19:14.042Z zigbee-herdsman:deconz:frameParser device state: 10100010
2022-01-12T21:19:14.042Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2022-01-12T21:19:15.664Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2022-01-12T21:19:15.666Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2022-01-12T21:19:15.668Z zigbee-herdsman:deconz:driver query aps data indication
2022-01-12T21:19:15.674Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 84
2022-01-12T21:19:15.677Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 84 srcAddr: 0x2516 destAddr: 0x0 profile id: 0x104 cluster id: 0x19 lqi: 255
2022-01-12T21:19:15.678Z zigbee-herdsman:deconz:frameParser response payload: 1,162,1,0,11,16,20,1,2,24,0,1
2022-01-12T21:19:15.681Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":162,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"fieldControl":0,"manufacturerCode":4107,"imageType":276,"fileVersion":16783362},"Command":{"ID":1,"response":2,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35}],"name":"queryNextImageRequest"}},"address":9494,"endpoint":11,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:debug 2022-01-12 22:19:15: Received Zigbee message from '0x001788010b806f29', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":16783362,"imageType":276,"manufacturerCode":4107}' from endpoint 11 with groupID null
Zigbee2MQTT:warn  2022-01-12 22:19:15: Received message from unsupported device with Zigbee model 'LTV002' and manufacturer name 'Signify Netherlands B.V.'
Zigbee2MQTT:warn  2022-01-12 22:19:15: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
2022-01-12T21:19:15.694Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2022-01-12T21:19:24.036Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 85
2022-01-12T21:19:24.038Z zigbee-herdsman:deconz:frameParser device state: 10100010
2022-01-12T21:19:24.038Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2022-01-12T21:19:25.667Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2022-01-12T21:19:25.668Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2022-01-12T21:19:25.670Z zigbee-herdsman:deconz:driver query aps data indication
2022-01-12T21:19:25.675Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 86
2022-01-12T21:19:25.678Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 86 srcAddr: 0x2516 destAddr: 0x0 profile id: 0x104 cluster id: 0x19 lqi: 255
2022-01-12T21:19:25.679Z zigbee-herdsman:deconz:frameParser response payload: 1,163,1,0,11,16,20,1,2,24,0,1
2022-01-12T21:19:25.683Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":163,"manufacturerCode":null,"commandIdentifier":1},"Payload":{"fieldControl":0,"manufacturerCode":4107,"imageType":276,"fileVersion":16783362},"Command":{"ID":1,"response":2,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35}],"name":"queryNextImageRequest"}},"address":9494,"endpoint":11,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}'
Zigbee2MQTT:debug 2022-01-12 22:19:25: Received Zigbee message from '0x001788010b806f29', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":16783362,"imageType":276,"manufacturerCode":4107}' from endpoint 11 with groupID null
Zigbee2MQTT:warn  2022-01-12 22:19:25: Received message from unsupported device with Zigbee model 'LTV002' and manufacturer name 'Signify Netherlands B.V.'
Zigbee2MQTT:warn  2022-01-12 22:19:25: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
2022-01-12T21:19:25.696Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1

Koenkk, 2022-01-13:
This device was added in 1.22.0, you are on 1.21.2 so update.

closed 2022-01-13 by Koenkk