Koenkk/zigbee2mqtt #10745: [New device support]: Gledopto GL-B-003P

denevecon, 2022-01-12:
### Link

https://www.gledopto.eu/led-e27-bulb-zigbee-st64-cct-color-temperature-eng

### Database entry

{"id":5,"type":"Router","ieeeAddr":"0xa4c138757c955e3b","nwkAddr":63673,"manufId":4687,"manufName":"GLEDOPTO","powerSource":"Mains (single phase)","modelId":"GL-B-003P","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"GL-B-003P","manufacturerName":"GLEDOPTO","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":2,"hwVersion":0,"dateCode":"20211015","swBuildId":"10051203"}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":134}},"lightingColorCtrl":{"attributes":{"colorMode":2,"colorTemperature":495,"colorTempPhysicalMin":158,"colorTempPhysicalMax":495}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":0,"dateCode":"20211015","swBuildId":"10051203","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1642022045032,"useImplicitCheckin":true}

### Comments

Device '0xa4c138757c955e3b' with Zigbee model 'GL-B-003P' and manufacturer name 'GLEDOPTO' is NOT supported, please etc.

Other models in same family GL-B-002P GL-B-004P

Did not succeed to fetch the temperature range. 

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['GL-B-003P'],
    model: 'GL-B-003P',
    vendor: 'Gledopto',
    description: 'Zigbee 7w E26/E27 Bulb RGB+CCT (pro)',
    // Note that fromZigbee, toZigbee and exposes are missing here since we use extend here.
    // Extend contains a default set of fromZigbee/toZigbee converters and expose for common device types.
    // The following extends are available:
    // - extend.switch
    // - extend.light_onoff_brightness
    // - extend.light_onoff_brightness_colortemp
    // - extend.light_onoff_brightness_color
    // - extend.light_onoff_brightness_colortemp_color
    extend: extend.light_onoff_brightness_colortemp(),
};

module.exports = definition;
```

### Supported color modes

colortemp

Koenkk, 2022-01-13:
Added, I assumed that the color temp range is the same as other gledopto devices (`colorTempRange: [155, 495]`)

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-13 by Koenkk

denevecon, 2022-01-13:
Thanks! I'll create 2 separate new device support issues to accommodate the 2 other new filament bulbs.

bsunderhus, 2025-05-28:
I'm having issues binding this device to an [IKEA onf/off switch](https://www.zigbee2mqtt.io/devices/E1743.html#ikea-e1743). has anyone experienced some problems?