Koenkk/zigbee2mqtt #10776: [Problem]: GLEDOPTO GL-SD-001 brightness value not updated when changed via push switch

HellGL, 2022-01-14:
### What happened?

I can switch the light on and off with the push switch and the status changed accordingly. But when I press the switch for a longer time to change the brightness, the brightness value in the status report does not change. Even when I switch the light on and off it keeps the changed brightness, but the state report still shows the old number. Also the Dashboard shows the old value. The brightness value is only updated when I press the reload button in the "Exposes" tab of the device which sends a "/get" request to the device. 

### What did you expect to happen?

The brightness value in the status is updated whenever I change the brightness regardless if it is done via MQTT or via the push switch

### How to reproduce it (minimal and precise)

Change brightness to max via dashboard
Press the push swith for a longer time and wait that the brightness changes
Press the push switch short to switch light completely off
Press the push switch short to switch light on again
Press the reload button for brightness in the "Exposes" tab on the dashboard
See attached debug log

BTW: Why are the messages always published several times?

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20180815

### Adapter

TI CC2531 USB CDC

### Debug log

```debug 2022-01-14 02:32:39: Received MQTT message on 'zigbee/Dimmer_Arbeitszimmer/set' with data '{"brightness":255}'
debug 2022-01-14 02:32:39: Publishing 'set' 'brightness' to 'Dimmer_Arbeitszimmer'
info  2022-01-14 02:32:39: MQTT publish: topic 'zigbee/Dimmer_Arbeitszimmer', payload '{"brightness":254,"linkquality":31,"state":"ON"}'
debug 2022-01-14 02:32:39: Received Zigbee message from 'Dimmer_Arbeitszimmer', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2022-01-14 02:32:39: MQTT publish: topic 'zigbee/Dimmer_Arbeitszimmer', payload '{"brightness":254,"linkquality":5,"state":"ON"}'
debug 2022-01-14 02:32:39: Received Zigbee message from 'Dimmer_Arbeitszimmer', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
debug 2022-01-14 02:32:39: Received Zigbee message from 'Dimmer_Arbeitszimmer', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
debug 2022-01-14 02:32:39: Received Zigbee message from 'Dimmer_Arbeitszimmer', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
debug 2022-01-14 02:32:39: Received Zigbee message from 'Dimmer_Arbeitszimmer', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
debug 2022-01-14 02:32:39: Received Zigbee message from 'Dimmer_Arbeitszimmer', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
debug 2022-01-14 02:32:39: Received Zigbee message from 'Dimmer_Arbeitszimmer', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
debug 2022-01-14 02:32:39: Received Zigbee message from 'Dimmer_Arbeitszimmer', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2022-01-14 02:32:39: MQTT publish: topic 'zigbee/Dimmer_Arbeitszimmer', payload '{"brightness":254,"linkquality":5,"state":"ON"}'
info  2022-01-14 02:32:39: MQTT publish: topic 'zigbee/Dimmer_Arbeitszimmer', payload '{"brightness":254,"linkquality":5,"state":"ON"}'
info  2022-01-14 02:32:39: MQTT publish: topic 'zigbee/Dimmer_Arbeitszimmer', payload '{"brightness":254,"linkquality":5,"state":"ON"}'
info  2022-01-14 02:32:39: MQTT publish: topic 'zigbee/Dimmer_Arbeitszimmer', payload '{"brightness":254,"linkquality":5,"state":"ON"}'
info  2022-01-14 02:32:39: MQTT publish: topic 'zigbee/Dimmer_Arbeitszimmer', payload '{"brightness":254,"linkquality":5,"state":"ON"}'
info  2022-01-14 02:32:39: MQTT publish: topic 'zigbee/Dimmer_Arbeitszimmer', payload '{"brightness":254,"linkquality":5,"state":"ON"}'
info  2022-01-14 02:32:39: MQTT publish: topic 'zigbee/Dimmer_Arbeitszimmer', payload '{"brightness":254,"linkquality":5,"state":"ON"}'
debug 2022-01-14 02:33:03: Received Zigbee message from 'Dimmer_Arbeitszimmer', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2022-01-14 02:33:03: MQTT publish: topic 'zigbee/Dimmer_Arbeitszimmer', payload '{"brightness":254,"linkquality":0,"state":"OFF"}'
debug 2022-01-14 02:33:03: Received Zigbee message from 'Dimmer_Arbeitszimmer', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
debug 2022-01-14 02:33:03: Received Zigbee message from 'Dimmer_Arbeitszimmer', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
debug 2022-01-14 02:33:03: Received Zigbee message from 'Dimmer_Arbeitszimmer', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
debug 2022-01-14 02:33:03: Received Zigbee message from 'Dimmer_Arbeitszimmer', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
debug 2022-01-14 02:33:03: Received Zigbee message from 'Dimmer_Arbeitszimmer', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
debug 2022-01-14 02:33:03: Received Zigbee message from 'Dimmer_Arbeitszimmer', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
debug 2022-01-14 02:33:03: Received Zigbee message from 'Dimmer_Arbeitszimmer', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
info  2022-01-14 02:33:03: MQTT publish: topic 'zigbee/Dimmer_Arbeitszimmer', payload '{"brightness":254,"linkquality":0,"state":"OFF"}'
info  2022-01-14 02:33:03: MQTT publish: topic 'zigbee/Dimmer_Arbeitszimmer', payload '{"brightness":254,"linkquality":0,"state":"OFF"}'
info  2022-01-14 02:33:03: MQTT publish: topic 'zigbee/Dimmer_Arbeitszimmer', payload '{"brightness":254,"linkquality":0,"state":"OFF"}'
info  2022-01-14 02:33:03: MQTT publish: topic 'zigbee/Dimmer_Arbeitszimmer', payload '{"brightness":254,"linkquality":0,"state":"OFF"}'
info  2022-01-14 02:33:03: MQTT publish: topic 'zigbee/Dimmer_Arbeitszimmer', payload '{"brightness":254,"linkquality":0,"state":"OFF"}'
info  2022-01-14 02:33:03: MQTT publish: topic 'zigbee/Dimmer_Arbeitszimmer', payload '{"brightness":254,"linkquality":0,"state":"OFF"}'
info  2022-01-14 02:33:03: MQTT publish: topic 'zigbee/Dimmer_Arbeitszimmer', payload '{"brightness":254,"linkquality":0,"state":"OFF"}'
debug 2022-01-14 02:33:05: Saving state to file /app/data/state.json
debug 2022-01-14 02:33:07: Received Zigbee message from 'Dimmer_Arbeitszimmer', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2022-01-14 02:33:07: MQTT publish: topic 'zigbee/Dimmer_Arbeitszimmer', payload '{"brightness":254,"linkquality":31,"state":"ON"}'
debug 2022-01-14 02:33:07: Received Zigbee message from 'Dimmer_Arbeitszimmer', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
debug 2022-01-14 02:33:07: Received Zigbee message from 'Dimmer_Arbeitszimmer', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
debug 2022-01-14 02:33:07: Received Zigbee message from 'Dimmer_Arbeitszimmer', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
debug 2022-01-14 02:33:07: Received Zigbee message from 'Dimmer_Arbeitszimmer', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
debug 2022-01-14 02:33:07: Received Zigbee message from 'Dimmer_Arbeitszimmer', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
debug 2022-01-14 02:33:07: Received Zigbee message from 'Dimmer_Arbeitszimmer', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
debug 2022-01-14 02:33:07: Received Zigbee message from 'Dimmer_Arbeitszimmer', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2022-01-14 02:33:07: MQTT publish: topic 'zigbee/Dimmer_Arbeitszimmer', payload '{"brightness":254,"linkquality":31,"state":"ON"}'
info  2022-01-14 02:33:07: MQTT publish: topic 'zigbee/Dimmer_Arbeitszimmer', payload '{"brightness":254,"linkquality":31,"state":"ON"}'
[... cut ...]
debug 2022-01-14 02:33:21: Received MQTT message on 'zigbee/Dimmer_Arbeitszimmer/get' with data '{"brightness":""}'
debug 2022-01-14 02:33:21: Publishing get 'get' 'brightness' to 'Dimmer_Arbeitszimmer'
debug 2022-01-14 02:33:21: Received Zigbee message from 'Dimmer_Arbeitszimmer', type 'readResponse', cluster 'genLevelCtrl', data '{"currentLevel":10}' from endpoint 1 with groupID 0
info  2022-01-14 02:33:21: MQTT publish: topic 'zigbee/Dimmer_Arbeitszimmer', payload '{"brightness":10,"linkquality":31,"state":"ON"}'```

Koenkk, 2022-01-14:
Could you check if the following external converter fixes this?

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['GL-SD-001'],
    model: 'GL-SD-001',
    vendor: 'Gledopto',
    description: 'Zigbee triac AC dimmer',
    extend: gledoptoExtend.light_onoff_brightness({noConfigure: true}),
    meta: {disableDefaultResponse: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.endpoints[0];
        await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
    },
};

module.exports = definition;

```

See https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-adding-your-device how to add the external converter

HellGL, 2022-02-13:
Sorry for the late answer. But the solution doesn't work:

```
Using '/app/data' as data directory
(node:7) UnhandledPromiseRejectionWarning: ReferenceError: gledoptoExtend is not defined
(Use `node --trace-warnings ...` to show where the warning was created)
(node:7) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:7) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

Koenkk, 2022-02-13:
Made a mistake, try with:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['GL-SD-001'],
    model: 'GL-SD-001',
    vendor: 'Gledopto',
    description: 'Zigbee triac AC dimmer',
    extend: extend.light_onoff_brightness({noConfigure: true}),
    meta: {disableDefaultResponse: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.endpoints[0];
        await extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
    },
};

module.exports = definition;
```

closed 2022-03-23 by github-actions[bot]

Headhunter70, 2022-08-12:
Hi there, I had exactly the same issue and can confirm the external converter is working (second edition). We just had the amend the absolute path and force the device to reconfigure. I am using Iobroker with zigbee adapter. Cheers