Koenkk/zigbee2mqtt #10780: [Problem]: Aqara curtain motor Xiaomi ZNCLDJ11LM

hellcry37, 2022-01-14:
### What happened?

For all intents this device work's that is not the issue, the problem that I have is, I have to use the device in INVERTED mode.

For this reason (inverted mode ON) this device does not work correctly in the fallowing ways:
In Z2M:
open should become - close (atm when I open have to press close)
close should become - open (atm when I close have to press open)
slider is reported correctly

Because of this in Home Assistant:
Status of curtain in inverted mode is reported correctly but does not work. Clicking open will do nothing, clicking close will do nothing. Moving the slider will work as expected.

I mention if i switch the curtain motor on normal mode all works as expected.

### What did you expect to happen?

In inverted mode buttons open and close should be switched.

### How to reproduce it (minimal and precise)

sw mode to inverted

### Zigbee2MQTT version

1.22.2 commit: 414c51f

### Adapter firmware version

20211217

### Adapter

Electrolama zig-a-zig-ah! ZZH

### Debug log

 MQTT publish: topic 'zigbee2mqtt/Master Bedroom Curtain', payload '{"last_seen":"2022-01-14T08:52:33.067Z","linkquality":134,"position":100,"running":true,"update":{"state":"idle"},"update_available":false}'
info MQTT publish: topic 'zigbee2mqtt/Master Bedroom Curtain', payload '{"last_seen":"2022-01-14T08:52:33.072Z","linkquality":134,"position":0,"running":false,"update":{"state":"idle"},"update_available":false}'
info MQTT publish: topic 'zigbee2mqtt/Master Bedroom Curtain', payload '{"last_seen":"2022-01-14T08:53:18.630Z","linkquality":134,"position":0,"running":false,"update":{"state":"idle"},"update_available":false}'
info MQTT publish: topic 'zigbee2mqtt/Master Bedroom Curtain', payload '{"last_seen":"2022-01-14T08:53:18.633Z","linkquality":134,"position":0,"running":true,"update":{"state":"idle"},"update_available":false}'
info MQTT publish: topic 'zigbee2mqtt/Master Bedroom Curtain', payload '{"last_seen":"2022-01-14T08:53:18.874Z","linkquality":134,"position":0,"running":true,"update":{"state":"idle"},"update_available":false}'
info MQTT publish: topic 'zigbee2mqtt/Master Bedroom Curtain', payload '{"last_seen":"2022-01-14T08:53:36.452Z","linkquality":134,"position":0,"running":true,"update":{"state":"idle"},"update_available":false}'
info MQTT publish: topic 'zigbee2mqtt/Master Bedroom Curtain', payload '{"last_seen":"2022-01-14T08:53:36.462Z","linkquality":134,"position":100,"running":false,"update":{"state":"idle"},"update_available":false}'

invis-z, 2022-01-23:
According to [https://www.zigbee2mqtt.io/devices/ZNCLDJ11LM.html](https://www.zigbee2mqtt.io/devices/ZNCLDJ11LM.html), maybe use the `reverse_direction` option instead of `invert_cover`?

hellcry37, 2022-01-23:
Thank you, now that you said that to me I found out why my automation did not work and nor the reverse_direction option

From ha MQTT publish page.
```txt
payload must be a string. If you want to send JSON then you need to format/escape it properly.
```
From the page of the device **do not work**
```yaml
  - service: mqtt.publish
    data:
      topic: zigbee2mqtt/<FRIENDLY_NAME>/set
      payload: "{ 'options': { 'reverse_direction': true } }"
```
After reading very carefully https://www.home-assistant.io/docs/mqtt/service/ **new service that works and publish correctly**
```yaml
  - service: mqtt.publish
    data:
      topic: zigbee2mqtt/<FRIENDLY_NAME>/set
      payload: "{ \"options\": { \"reverse_direction\": true } }"
```

Quotes must be escaped if you want to send json payload!

closed 2022-01-23 by hellcry37