Koenkk/zigbee2mqtt #10781: [Problem]: TuYa TRV TS0601 _TZE200_hue3yfsn temp calibration range

Serios, 2022-01-14:
### What happened?

After updating to the latest 1.22.2 z2m my HA log gets bombarded with:
Invalid value for number...._local_temperature_calibration: -6.0 (range -5.0 - 5.0)

I have couple of _TZE200_hue3yfsn TuYa TRV's on which I set the local_temperature_calibration trough HA automation every 20 min based on external temp sensors, trough this calculation which value is send as payload to the corresponding MQTT topic:

`{{states(trigger.entity_id)|float(0) | round(1,'half') - state_attr('climate.0x...........','local_temperature')|float(0) | round(1, 'half') + state_attr('climate.0x.......................','local_temperature_calibration')|float(0) | round(1, 'half') }}`

Usually the values calculated are in the range of (-7 - 6.5) and everything was working fine before 1.22.2, but since updating the local_temperature_calibration apparently have new range limitation (-5.0 - 5.0) thus causing the error. Also it's very strange, but couple of the TRV's gets -429496720.0  as local_temperature_calibration value and I cant figure out where this is coming from.

### What did you expect to happen?

X

### How to reproduce it (minimal and precise)

Set local_temperature_calibration outside the (-5.0 - 5.0) -  zigbee2mqtt/FRIENDLY_NAME/set {"local_temperature_calibration": -8.5}

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

zStack12 20201127

### Adapter

CC2531 dongle 

### Debug log

X

Koenkk, 2022-01-15:
@vladi1234 could you check this?

vladi1234, 2022-01-15:
@Koenkk I will test.

vladi1234, 2022-01-15:
@Koenkk  I did the latest update and the thermostat works as it should.

@Serios  It is correct, since two weeks there is a limit of -5,0 to +5,0 for local_temperature_calibration, because TRV can only do -5,0 to +5,0 see manual excerpt:
[image]

> Invalid value for number...._local_temperature_calibration: -6.0 (range -5.0 - 5.0)

If you don't send the right number, but not a valid one like number (-6,0) to TRV, then you get an error with wrong number like that (-429496720.0).

@Serios  **Why do you want to send invalid number (-6,0)?**

Serios, 2022-01-15:
@vladi1234 This number (-6,0) came from calculation which take in consideration the temp from another external temp sensor. Since the local temperature reported from the TRV is a way too high from the real room temp, especially with two of the TRV's mounted on radiators, that are inside decorative enclosures. I know you cited the offset settings from the brochure, but all TRV's where working just great, with offsets below -5 and above +5, correcting the local temperature just fine, thus proper temp regulation was achieved  before 1.22.2. 

If this can't be reverted, because is based on general use case, can I somehow override the range limit so will work again just for my case?

vladi1234, 2022-01-15:
Yes, of course. You can change it individually for yourself as you want.
For this you need to make changes in a file `opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/tuya.js`

`exposes.climate().withPreset(['auto', 'manual', 'holiday']).withLocalTemperatureCalibration(-5, 5, 0.1, ea.STATE_SET)`

If you have adjusted accordingly, then restart z2m.

vladi1234, 2022-01-15:
If helps, please close this issue.
Thank you!

Serios, 2022-01-15:
Is there a way to override it, so it will be update persistent? Coming from web developing background I know that modifying core/system files is a big no no.

vladi1234, 2022-01-16:
I don't know of any other way.

Serios, 2022-01-16:
@vladi1234 Ok thanks, will think of some kind of script to update z2m and touch up the file after that. I'm closing this now since is about very specific use case and not a bug.

closed 2022-01-16 by Serios