Koenkk/zigbee2mqtt #10788: [Problem]: Moes TRV (BRT-100-TRV) is reporting local temperature in way too long intervals after upgrade to 1.22.2

ondrejbaranek, 2022-01-14:
### What happened?

Moes devices stopped reporting temperature change, so heating automation is broken due that and heating is then turned on for too long

### What did you expect to happen?

local temperature is reported when changed

### How to reproduce it (minimal and precise)

change temperature in the room to verify if change is reported
temperature is changed correctly only when manually waking up device by touching it

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20211217

### Adapter

zzh!

### Debug log

not needed, messages received contains wrong temperature, no errors or messages that could lead to cause

ondrejbaranek, 2022-01-14:
This happens for 4 TRVs, reset and repairing won't help

Koenkk, 2022-01-15:
@vladi1234 could you check this?

vladi1234, 2022-01-15:
@Koenkk I will test.

vladi1234, 2022-01-15:
My thermostat's worked correctly as always, nothing has changed.

What you describe is also correct, of course TRV also delivers local_temp again if you wake it up manually from standby mode.

My TRV's also automatically return local_temp if the temperature has changed significantly, which is also correct.

All right.

ondrejbaranek, 2022-01-18:
Mine doesn't report significant changes as well, not always but often. Sometimes not even after waking up the device manually. I will sniff zigbee traffic to try to find what's causing this. It was not really acting like this before.

ondrejbaranek, 2022-01-18:
@vladi1234 is your TRV sending so many Data requests as well?
[image]
This TRV is sending about 10% of zigbee packets. Same for other 3 TRVs as well. So TRV are sending 40% of the traffic. I have 23 devices and 10 are acting as routes, 13 are battery powered.
[image]

vladi1234, 2022-01-18:
Excuse me! I didn't understand your calculation.

My TRV always run differently, there are many factors that play a big role in how often this happens.

Factors:
  1. What's the weather like outside?
  2. How warm is the heating water?
  3. How good is house insulation?
  4. How is TVR installed on the radiator, at the entrance or exit from the radiator?

Factor four plays a very big role for TRV, how often does TRV open up that it needs to adjust temperature.
As soon as any of the first three factors change, the behavior of the TRV also changes.

See how different can look:

Bedroom TRV.
[image]

Flur TRV.
[image]

All right.

ondrejbaranek, 2022-01-19:
I ran zigbee sniffer and 40% of all packets were from TRVs. TRVs are 17% of all devices and as they are battery powered, packets level should be much lower. Those packets means route inactive, however i don't see that many packets like that for other battery powered devices.

ondrejbaranek, 2022-01-19:
I completely understand factors affecting TRVs as you described, but it was still reporting more often in similar outside temperatures and same water temperature.

marpasch, 2022-02-05:
I have the same problem with my two BRT-100-TVR. They don´t reporting temperature change. My english is too bad to explain more details.

vladi1234, 2022-02-05:
Can you make graph recording last 24 hours of TRV temperature? As in comment https://github.com/Koenkk/zigbee2mqtt/issues/10788#issuecomment-1015667897

marpasch, 2022-02-06:
Last 24 hours from TVR (red marks are wake up from hand)
[image]
Last Days from TVR
[image]
Last Days from another Sensor in the same room

[image]

vladi1234, 2022-02-06:
@marpasch I don't want to compare TRV with other sensors because it's not a thermometer!

In picture-1 and picture-2 I can see the TRV is working properly.
The temperature outside may have gotten warmer and as a result, it doesn't have to heat and send the temperature as often.

closed 2022-03-16 by github-actions[bot]

vladyspavlov, 2022-11-10:
It seems like it will fix this  https://github.com/Koenkk/zigbee-herdsman-converters/pull/4917