Koenkk/zigbee2mqtt #10792: [New device support]: Aqara WXKG14LM (single rocker wireless switch H1)

DamGranger, 2022-01-14:
### Link

https://www.aliexpress.com/i/1005003559498912.html

### Database entry

{"id":50,"type":"EndDevice","ieeeAddr":"0x54ef441000064a7f","nwkAddr":63980,"manufId":4447,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.remote.b18ac1","epList":[1,2,3,4,5,6],"endpoints":{"1":{"profId":260,"epId":1,"devId":259,"inClusterList":[0,3,1],"outClusterList":[3,6],"clusters":{"genBasic":{"attributes":{"modelId":"lumi.remote.b18ac1","appVersion":21}},"aqaraOpple":{"attributes":{"247":{"type":"Buffer","data":[1,33,139,12,3,40,25,4,33,168,19,5,33,31,1,6,36,1,0,0,0,0,8,33,21,1,10,33,226,15,12,32,1,100,32,1,102,32,3,103,32,1,104,33,168,0]}}},"genMultistateInput":{"attributes":{"presentValue":1}},"genPowerCfg":{"attributes":{"batteryVoltage":32}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"epId":2,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"epId":3,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"epId":4,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"epId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"epId":6,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":21,"stackVersion":2,"hwVersion":1,"dateCode":"20200907","swBuildId":"2019\u0000www.","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-983567100},"lastSeen":1642175622293,"useImplicitCheckin":true}

### Comments

Seems very similar to WXKG16LM but I had an error on the configure phase when reusing this definition. I ended up with the converter below, but I'm not very comfortable having zero knowledge of this. At least, it works without throwing any error message.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['lumi.remote.b18ac1'],
    model: 'WXKG14LM',
    vendor: 'Xiaomi',
    description: 'Aqara wireless remote switch H1 (single rocker)',
    fromZigbee: [fz.xiaomi_multistate_action, fz.aqara_opple],
    toZigbee: [tz.xiaomi_switch_click_mode],
    exposes: [e.battery(), e.battery_voltage(), e.action(['single', 'double', 'hold']),
        exposes.enum('click_mode', ea.SET, ['fast', 'multi'])
            .withDescription('Click mode, fast: only supports single click which will be send immediately after clicking.' +
                'multi: supports more events like double and hold')],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint1 = device.getEndpoint(1);
        await endpoint1.write('aqaraOpple', {'mode': 1}, {manufacturerCode: 0x115f, disableResponse: true});
    },
};

module.exports = definition;
```

Koenkk, 2022-01-15:
@z-x-f do you have this device?

z-x-f, 2022-01-15:
> @z-x-f do you have this device?

I don't have this device and plan to buy it soon.

Koenkk, 2022-01-19:
added in https://github.com/Koenkk/zigbee-herdsman-converters/pull/3706 thanks to @z-x-f 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)