Koenkk/zigbee2mqtt #10793: SEA802 Reports heating state as null

DominicWindisch, 2022-01-14:
### What happened?

I'm using 5 SAE802 thermostats. They are set to heating mode "heat" permanently and home assistant controls the respective heating set points via zigbee2mqtt. The devices report back their status occasionally (setpoint, Temperatur, child lock etc), however the "heating" property is always null.

### What did you expect to happen?

According to the documentation, "heating" should report "on" or "off".

### How to reproduce it (minimal and precise)

Pair SAE802, set to heating mode "heat", set a heating point and check status.

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20211217 

### Adapter

Sonoff zigbee dongle plus

zibous, 2022-01-15:
Same Problem here with `SEA801`:

```
Zigbee2MQTT version: 1.22.2-dev commit: 43ded111
Coordinator type: ConBee2/RaspBee2
Coordinator revision: 0x26660700
Frontend version: 0.6.68
```

```
{
    "anti_scaling": "ON",
    "away_mode": "OFF",
    "battery_low": false,
    "child_lock": "UNLOCK",
    "current_heating_setpoint": 23,
    "frost_detection": "ON",
    "last_seen": "2022-01-15T08:20:54+01:00",
    "linkquality": 255,
    "local_temperature": 22.6,
    "local_temperature_calibration": 0,
    "preset_mode": "none",
    "system_mode": "heat",
    "window_detection": "OFF",
    "device": {
        "applicationVersion": 85,
        "dateCode": "",
        "friendlyName": "thermostat-office",
        "hardwareVersion": 1,
        "ieeeAddr": "0x5c0272fffe8568d1",
        "manufacturerID": 4098,
        "manufacturerName": "_TZE200_yw7cahqs",
        "model": "SEA801-Zigbee/SEA802-Zigbee",
        "networkAddress": 62895,
        "powerSource": "Battery",
        "stackVersion": 0,
        "type": "EndDevice",
        "zclVersion": 3
    },
    "elapsed": 2759852,
    "heating": null  <----- ???
}
```

Koenkk, 2022-01-15:
this was added in https://github.com/Koenkk/zigbee-herdsman-converters/pull/3596 by @mgwoj  , @mgwoj  could you check this/

mgwoj, 2022-01-15:
> this was added in Koenkk/zigbee-herdsman-converters#3596 <https://github.com/Koenkk/zigbee-herdsman-converters/pull/3596>  by @mgwoj <https://github.com/mgwoj>  , @mgwoj <https://github.com/mgwoj>  could you check this?

Hello all, 

 

it is very likely that this is newest feature added to the recent models, like mine `_TZE200_zr9c0day`.

 

Thanks,

Michał

DominicWindisch, 2022-01-15:
I don't think that's the case. Using the same thermostat with ZHA properly reports heating on or off.

mgwoj, 2022-01-15:
Please activate DEBUG logging and check what you see in the log when you set temperature to low value or very high. 

 

You should see something like this:

 

Zigbee2MQTT:debug 2022-01-15 23:35:33: Received Zigbee message from 'valve', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[1],"type":"Buffer"},"datatype":4,"dp":3,"fn":0,"seq":6912}' from endpoint 1 with groupID 0                                                                                                                                                                                                                                      

Zigbee2MQTT:info  2022-01-15 23:35:33: MQTT publish: topic 'zigbee2mqtt2/valve', payload '{"anti_scaling":"ON","away_mode":"OFF","battery_low":false,"child_lock":"LOCK","current_heating_setpoint":27,"frost_detection":"ON","heating":"ON","

linkquality":120,"local_temperature":22.1,"local_temperature_calibration":0,"preset_mode":"none","schedule_away_or_vacation":"06:00/20.0°C 08:00/16.0°C 18:00/20.0°C 22:00/16.0°C","schedule_friday":"06:00/20.0°C 08:00/16.0°C 18:00/20.0°C 22:00/16.0°C","schedule_monday":"00:00/12.5°C 07:00/9.0°C 10:00/25.0°C 11:11/25.0°C","schedule_saturday":"06:00/20.0°C 08:00/16.0°C 18:00/20.0°C 22:00/16.0°C","schedule_thursday":"06:00/20.0°C 08:00/16.0°C 18:00/20.0°C 22:00/16.0°C","schedule_tuesday":"00:00/11.5°C 07:00/9.0°C 10:00/25.0°C 10:30/26.0°C","schedule_wednesday":"00:00/12.5°C 07:00/9.0°C 10:00/25.0°C 10:00/25.0°C","system_mode":"auto","weekly_schedule":{"days":["away_or_vacation"],"transitions":[{"heating_setpo

int":"20.0","time":360},{"heating_setpoint":"16.0","time":480},{"heating_setpoint":"20.0","time":1080},{"heating_setpoint":"16.0","time":1320}]},"window_detection":"ON"}'                                                                    

 

Please check what you have and send me back. Maybe you have different DP (data point) for this option in your model? You have to check how it is implemented in ZHA then.

P.S. Please ignore the entries for the scheduling options, I am working on exposing that on UI currently.

 

Thanks,

Michal

shutaranga, 2022-02-02:
Here is debug log when temperature is set to 5°C:

```
Zigbee2MQTT:debug 2022-02-02 18:34:35: Received Zigbee message from 'valve', type 'commandDataReport', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,50],"type":"Buffer"},"datatype":2,"dp":103}],"seq":6}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-02-02 18:34:35: MQTT publish: topic 'zigbee2mqtt/valve', payload '{"anti_scaling":"ON","away_mode":"OFF","battery_low":false,"child_lock":"UNLOCK","current_heating_setpoint":5,"frost_detection":"ON","linkquality":207,"local_temperature":26.8,"local_temperature_calibration":0,"preset_mode":"none","system_mode":"heat","window_detection":"ON"}'

Zigbee2MQTT:debug 2022-02-02 18:35:57: Received Zigbee message from 'valve', type 'commandDataResponse', cluster 'manuSpecificTuya', data '{"dpValues":[{"data":{"data":[0,0,0,238],"type":"Buffer"},"datatype":2,"dp":102}],"seq":512}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-02-02 18:35:57: MQTT publish: topic 'zigbee2mqtt/valve', payload '{"anti_scaling":"ON","away_mode":"OFF","battery_low":false,"child_lock":"UNLOCK","current_heating_setpoint":5,"frost_detection":"ON","linkquality":207,"local_temperature":23.8,"local_temperature_calibration":0,"preset_mode":"none","system_mode":"heat","window_detection":"ON"}'
```

this is for SAE802 thermostat (TS0601,  _TZE200_yw7cahqs), with zigbee2mqtt 1.23.0, firmware 20211217, adapter:  slaesh 

am missing "dp":3 in commandDataResponse and "heating":"ON" in mqtt msg.

also checked this evening logs for this valve and there is no contains "datatype":4 only "datatype":0, "datatype":1, "datatype":2 and "datatype":5

hope this helps

mgwoj, 2022-02-03:
If you do not receive any message for dp:3, then my modification for checking for heating will not work. You can try to reconnect the TRV to Zigbee2Mqtt and check messages at that time, because after reconnect device sends all the datapoints. If there will be no dp:3, then probably your device does not support it.

shutaranga, 2022-02-03:
I confirm Hama thermostat does not support dp:3, so we cant expect to see heating state. 

in addition, after reconnect i checked all sent datapoints and noticed that 4 more are missing.
The list of supported/unsupported datapoints of Hama thermostat against [zigbee-herdsman-converters](https://github.com/Koenkk/zigbee-herdsman-converters)/[lib/tuya.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/a0f9620683b90aa59986c75bb5aead507ee2b156/lib/tuya.js) (Latest commit 8d4a63f) is:
```
const dataPoints = {

.........

    // Saswell TRV
    saswellHeating: 3,                      // -- unsupported --
    saswellWindowDetection: 8,              //"dp":8
    saswellFrostDetection: 10,              //"dp":10
    saswellTempCalibration: 27,             //"dp":27
    saswellChildLock: 40,                   //"dp":40
    saswellState: 101,                      //"dp":101
    saswellLocalTemp: 102,                  //"dp":102
    saswellHeatingSetpoint: 103,            //"dp":103
    saswellValvePos: 104,                   // -- unsupported --
    saswellBatteryLow: 105,                 //"dp":105
    saswellAwayMode: 106,                   //"dp":106
    saswellScheduleMode: 107,               // -- unsupported --
    saswellScheduleEnable: 108,             //"dp":108
    saswellScheduleSet: 109,                // -- unsupported --
    saswellSetpointHistoryDay: 110,         //"dp":110
    saswellTimeSync: 111,                   // -- unsupported --
    saswellSetpointHistoryWeek: 112,        //"dp":112
    saswellSetpointHistoryMonth: 113,       //"dp":113
    saswellSetpointHistoryYear: 114,        //"dp":114
    saswellLocalHistoryDay: 115,            //"dp":115
    saswellLocalHistoryWeek: 116,           //"dp":116
    saswellLocalHistoryMonth: 117,          //"dp":117
    saswellLocalHistoryYear: 118,           //"dp":118
    saswellMotorHistoryDay: 119,            //"dp":119
    saswellMotorHistoryWeek: 120,           //"dp":120
    saswellMotorHistoryMonth: 121,          //"dp":121
    saswellMotorHistoryYear: 122,           //"dp":122
    saswellScheduleSunday: 123,             //"dp":123
    saswellScheduleMonday: 124,             //"dp":124
    saswellScheduleTuesday: 125,            //"dp":125
    saswellScheduleWednesday: 126,          //"dp":126
    saswellScheduleThursday: 127,           //"dp":127
    saswellScheduleFriday: 128,             //"dp":128
    saswellScheduleSaturday: 129,           //"dp":129
    saswellAntiScaling: 130,                //"dp":130

.........

}
```

closed 2022-03-14 by github-actions[bot]

iksrd, 2022-03-18:
I have the same problem.

evilpie, 2022-04-07:
Same problem here with the hama smart thermostat. How can I help debugging this? I sadly don't have the Hama central unit. @Koenkk please reopen.

mgwoj, 2022-04-07:
It is very likely that your thermostat does not support this information. Please read my earlier comments and check if after reconnect you receive info about `dp:3`. If not, it means it is not supported.

evilpie, 2022-04-07:
Unfortunate, the dp:3 message seems indeed to be missing. Looking at some YouTube videos for that device, the official app doesn't seem to show the heating state either. :disappointed:

iksrd, 2022-11-05:
I have the same problem

k3erg, 2023-02-28:
As I have obtained both, I can confirm that there are two versions of HAMAs SEA802: one supports "heating" and the other does not. Did not find any firmware updates yet, according to SASWELLs Website there are multiple signee 3.0 versions of the TVR.