Koenkk/zigbee2mqtt #10794: [Feature request]: Support for cloning device state to a group

dominicusmento, 2022-01-14:
### Is your feature request related to a problem? Please describe

I have a large hallway with 4 zigbee switches (only one is actually connected to the light wiring - main switch, the other 3 just have the phase and neutral to provide them with power - and they should serve just as relay switches) -> previously, it was 4-way switches installed there.

I have a need to make all those switches in sync, so whatever switch last provides the state -> whole group should follow the same state. From the docs, I can't see that I can use any property to achieve this.

Switches: 4x Moes ZTS-EU_1gang
Binding: seems that's not supported for those switches
Z2M Group: I think it all works as intended, group on -> all swiches follows state: on, bulbs are turned on, and vice versa..
(I'm also using 2x Hue motion sensors and HA automation to turn it on/off and that works quite fine cause I'm using the group to manage switches)

### Describe the solution you'd like

I really don't know how the zigbee2mqtt's groups are implemented, is it tied directly to zigbee group implementation or it has some management possibility? So, if my idea is not achievable, please just close the issue.

Ideally, there will be a new group configuration property, like 'optimistic: true' (multiswitch:true), which would for every change in state of any device in group -> change the state of the group so the rest of the devices would pick the change too.

### Describe alternatives you've considered

Tried to manage all of that with 2 separate automations in HA and I have achieved that but it's too slow and laggy and sometimes lights runs multiple time when automations are too slow to complete (as I have 4 switches that can be triggered, there are 4 triggers which are monitored - which seems 4 instances of automation to process on every change.. etc. Tried to create scenes for group too, but I don't see how that could be used in my case..

### Additional context

[image]
[image]
(if I try to unbind coordinator, I get 'Nothing to unbind' message)

SHxKM, 2022-01-17:
If I’m reading your request right (and without commenting on the feature request itself), this should be achievable in HA if all switches belong to one group. Why can’t something like this work? (pseudo-code)

``` yaml
trigger: 
   platform: state
   entity_id: switch.switch_n1
   to: “on”
actions:
  service: switch.turn_on
   target:
      entity_id: switch.SWITCH_GROUP
```

You can define multiple triggers of course.

dominicusmento, 2022-01-17:
Hi, thank you for a comment.. I've tried that already and it's not so simple because I have 4 triggers.. 
When first one is triggered I have a condition to check if it's 'TO' state is different from group.state at that moment.. if it is I than change the group state.. which now triggers automation 3 more times because the remaining switches has also changed the state. Now, if that's all.. everything works fine.. With a little bit of delay due to load but it's kind of ok. 
(I have 2 automations, one for turning from off -> on, and second for on -> off. 
My automations are 'single' and with a delay of 500ms at the end which prevents another instance of that automation for at least that period of time)

The problem is when someone turns on the first switch and then the other person turns it off in a half of second.. then the lights go crazy for couple of seconds because all the states gets messy and out of sync.
This also happens if the light is triggered by motion sensor (which is of course another automation) and then someone turns-off any of the switches in that same half of second..

So it's achievable with HA automation but very slow and buggy. I'd really like if it can be solved on a layer below the HA. And in general it seems to me that turning those switches by z2m group is faster than by HA light group or ordinary group (I tried both of them).

SHxKM, 2022-01-17:
> The problem is when someone turns on the first switch and then the other person turns it off in a half of second.. then the lights go crazy for couple of seconds because all the states gets messy and out of sync.

Can [Automation Modes](https://www.home-assistant.io/docs/automation/modes/)  help here? (Specifically “queued” or “restart”)

(Actually now that I’m thinking about it, if these individual switches all represent a group; why not expose them as a group, and treat them as a single entity in HA? So you’d listen to the group state change, and do actions based on that - with optimistic true the state of the individual devices is changing anyway…maybe I’m missing something)

> I'd really like if it can be solved on a layer below the HA.

Not disputing that, just trying to offer an alternative (maybe for the time being).

> And in general it seems to me that turning those switches by z2m group is faster than by HA light group or ordinary group (I tried both of them).

Definitely use a Zigbee group.

dominicusmento, 2022-01-25:
I finally found some time to check and play with the groups and switches again..
@SHxKM 
I finished with: 

- two automations, one for turning switches on, second one for turning them off
- having two groups used for triggering those automations: one zigbee optimistic for detecting ON, and one HA group to detect the OFF state (when only one swich is off, group is off; which doesn't work in zigbee group - or I haven't found I way)
- in both automations I am calling the service: light: on/off, for every HA light entity
(tried with calling mqtt publish first to publish the state for the whole group again thinking it would allow the rest of the swiches to pick up the 'new' state, but it seems that mqtt/z2m has some control of it and doesn't allow to send the state which is already active for the group)

'Double clicks' (fast on-off) still makes the system confused, but now only 1 additional automation sequence runs so, therefore, this is the best I've got now. Also, this is the fastest solution for me currently and it obviously is faster than previous solution, even if it's still not without the bugs. 
Thanks

@Koenkk do you maybe have the answer to initial question so the issue can be solved?

closed 2022-03-04 by github-actions[bot]