Koenkk/zigbee2mqtt #10797: [Feature request]: make mqtt dependency optionnal

roumano, 2022-01-14:
### Is your feature request related to a problem? Please describe

it is a bit odd z2m fails to start because if mqtt configuration

It'd make more sense if it would start and we could configure mqtt in the frontend

It's will also make z2m more robust ( if example mqtt start few seconds after, or the broker is on a another host and we get a temporary network issue ) and it's also be easier for testing as z2m will have less dependency.

### Describe the solution you'd like

accept to start z2m even if a mqtt broker is not started
Look regularly if the mqtt broker can be acceded

Also, it's can permit in the futur a help to configure the mqtt :
like add buttons : 
- test mqtt network parameters
- test login / password parameters

### Describe alternatives you've considered

.

### Additional context

Actually, you can change the mqtt on the UI of z2m, but if it's connect at the start of z2m, it's failling to start

roumano, 2022-01-15:
- on z2m, if a set a bad server address of the mqtt server : 
the z2m service is started but not working and the web ui can't be open
- if i use a bad login/password for mqtt server
z2m service failed to start

Koenkk, 2022-01-15:
This decision was made in https://github.com/Koenkk/zigbee2mqtt/issues/8956 , but might be worth to reconsider if enough people find this important.

closed 2022-02-23 by github-actions[bot]