Koenkk/zigbee2mqtt #10799: [New device support]: AVATTO X711A 1 gang switch no neutral

vladispavlov, 2022-01-14:
### Link

https://aliexpress.ru/item/1005002198458745.html

### Database entry

{"id":12,"type":"EndDevice","ieeeAddr":"0x60a423fffed9854e","nwkAddr":47678,"manufId":4098,"manufName":"_TZ3000_9hpxg80k","powerSource":"Mains (single phase)","modelId":"TS0011","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,4,5,6],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0011","manufacturerName":"_TZ3000_9hpxg80k","stackVersion":0,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001cdd2378","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1482240604},"lastSeen":1642196836349,"useImplicitCheckin":true}

### Comments

It is already supported, but i found `moesStartUpOnOff` attribute in `onOff` cluster, which does not exist in current converter. It partially works, i.e. only "off" and "previous" states. If i set "on" it doesn't turn on automatically power on behaviour.

### External converter

```shell
const fz = require("zigbee-herdsman-converters/converters/fromZigbee");
const tz = require("zigbee-herdsman-converters/converters/toZigbee");
const exposes = require("zigbee-herdsman-converters/lib/exposes");
const reporting = require("zigbee-herdsman-converters/lib/reporting");
const extend = require("zigbee-herdsman-converters/lib/extend");

module.exports = {
    fingerprint: [
        { modelID: "TS0011", manufacturerName: "_TZ3000_9hpxg80k" },
    ],
    zigbeeModel: ["TS0011"],
    model: "TS0011",
    vendor: "TuYa",
    description: "Smart light switch - 1 gang",
    toZigbee: extend.switch().toZigbee.concat([tz.moes_power_on_behavior]),
    fromZigbee: extend.switch().fromZigbee.concat([fz.moes_power_on_behavior]),
    exposes: [
        exposes.presets.switch(),
        exposes.presets.power_on_behavior(),
    ],
    whiteLabel: [{ vendor: "AVATTO", model: "X711A" }],
    configure: async (device, coordinatorEndpoint, logger) => {
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, [
        "genOnOff",
        ]);
        // Reports itself as battery which is not correct: https://github.com/Koenkk/zigbee2mqtt/issues/6190
        device.powerSource = "Mains (single phase)";
        device.save();
    },
};
```

kodifan, 2022-02-06:
hello ive got similar device:
https://www.aliexpress.com/item/1005003438720832.html

Database entry :
"manufId":4417,"manufName":"_TZ3000_ji4araar","powerSource":"Mains (single phase)","modelId":"TS0011"

"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":2}}

Also power on status is missing form device configuration but I can set moesStartUpOnOff attribute in dev console to execute needed behavior. 

Anyway my question is: why those attributes are missing from Exposes tab in device configuration in Z2M ? What should be changed in order to achieve that ?

vladispavlov, 2022-02-06:
> why those attributes are missing from Exposes tab in device configuration in Z2M ?

It's because there is no support for this in z2m. I did it in the external converter.

> EDIT2. The same problem. off and remember state is working, on option does not

I checked this switch in the tuya app and didn't find there any option to change power on behavior. It looks like in current firmware this functionality is not fully implemented. So it seems like the current states "off" and "previous" are only available.

kodifan, 2022-02-06:
previous state if fine (for now) but its nice to have an option.
i think i will test new sonoff switch module 
https://itead.cc/product/sonoff-zbmini-l-zigbee-smart-switch-no-neutral-wire-required/
this one have this option for sure

vladispavlov, 2022-02-07:
I fully tested device and power on behavior is not working as expected. If set option to "off" and try to repower the device, it turns on for a second and then turns off. Also it works only for the first time. If repower again it will turn on.
Also tested the indicator modes, and it seems like the change value of tuyaBacklightMode has no any effect.
So I think with the current appVersion it will not work normally and there are no reasons to do the external converters for this device.

closed 2022-02-07 by vladispavlov