Koenkk/zigbee2mqtt #10800: TERNCY-SD01 paired but all value=null

labactani, 2022-01-14:
### What happened?

I have just bought a TERNCY-SD01 smart dial and paired with Z2M with no issue in recognizing and configuring.
Problem is that all the value with the exception of link quality are null of NA

this is the state:
'''
{
    "linkquality": 255,
    "update": {
        "state": "idle"
    },
    "update_available": false,
    "action": null,
    "battery": null,
    "direction": null
}
'''

### What did you expect to happen?

Z2M to sense the actions

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

26720700

### Adapter

ConBee II

cracrama, 2022-01-18:
Same here but after stick firmware update to newest version. Changing firmware doesn't help.

labactani, 2022-02-15:
@cracrama did you solve tis finally? I have the latest firmware for my conbee2 but still nothing

cracrama, 2022-02-22:
Kind of - I have just repaired all my devices and now is ok.

labactani, 2022-03-01:
really? you have just repaired the Terncy and it worked? awesome?
just to be sure, can I ask you the following:
- Z2M version 
- Conbee II firware version
- do you do any particular action while Z2M interrogates the device (rotate the dial, push...rain dance?)

thanks a lot

closed 2022-04-09 by github-actions[bot]