Koenkk/zigbee2mqtt #10801: Suggestions on how to troubleshoot/workaround issue of all lights in group not always turning off

stcbus, 2022-01-15:
### What happened?

Via my Apple Home app (connected to Home Assistant) this evening I pushed the button to turn off my living room ceiling lights. It is a zigbee group of 4 Sengled bulbs. Only some of the lights turned off. I then subsequently turned the group on/off until finally after the 4th or 5th time, all of the lights went off.

Then, just as I was reporting this same thing is happening, but now even worse. Almost nothing is going off (around 08:00pm EST). See this, the state is received but...

`MQTT publish: topic 'zigbee2mqtt/Livingroom_window_lamp', payload '{"brightness":203,"color":{"h":30,"hue":30,"s":86,"saturation":86,"x":0.477,"y":0.4137},"color_mode":"color_temp","color_temp":400,"linkquality":84,"state":"ON","update":{"state":"idle"},"update_available":null}'
debug Received MQTT message on 'zigbee2mqtt/livingroom_ceiling/set' with data '{"state": "OFF"}'
debug Publishing 'set' 'state' to 'livingroom_ceiling'
info MQTT publish: topic 'zigbee2mqtt/0xb0ce181403531292', payload '{"brightness":203,"color":{"h":30,"hue":30,"s":86,"saturation":86,"x":0.477,"y":0.4137},"color_mode":"color_temp","color_temp":400,"linkquality":108,"state":"OFF","update":{"state":"idle"},"update_available":null}'
info MQTT publish: topic 'zigbee2mqtt/livingroom_ceiling', payload '{"brightness":203,"color":{"h":30,"hue":30,"s":86,"saturation":86,"x":0.477,"y":0.4137},"color_mode":"color_temp","color_temp":400,"state":"OFF"}'
info MQTT publish: topic 'zigbee2mqtt/0xb0ce181403531c71', payload '{"brightness":203,"color":{"h":30,"hue":30,"s":86,"saturation":86,"x":0.477,"y":0.4137},"color_mode":"color_temp","color_temp":400,"linkquality":147,"state":"OFF","update":{"state":"idle"},"update_available":null}'
info MQTT publish: topic 'zigbee2mqtt/0xb0ce181403531dae', payload '{"brightness":203,"color":{"h":30,"hue":30,"s":86,"saturation":86,"x":0.477,"y":0.4137},"color_mode":"color_temp","color_temp":400,"linkquality":111,"state":"OFF","update":{"state":"idle"},"update_available":null}'
info MQTT publish: topic 'zigbee2mqtt/0xb0ce18140353127f', payload '{"brightness":203,"color":{"h":30,"hue":30,"s":86,"saturation":86,"x":0.477,"y":0.4137},"color_mode":"color_temp","color_temp":400,"linkquality":72,"state":"OFF","update":{"state":"idle"},"update_available":null}'
debug Received Zigbee message from 'kitchen_ceiling_2', type 'readResponse', cluster 'genBasic', data '{"zclVersion":1}' from endpoint 1 with groupID 0`

### What did you expect to happen?

All lights to go off the first time.

In the end, I am wondering what is going on and how best to prevent this from happening (whether it's find ways to repeat the command or what).

### Zigbee2MQTT version

1.22.2-dev commit: 4f887a1

### Adapter firmware version

20210708

### Adapter

zzh

### Debug log

[zigbeelog.txt](https://github.com/Koenkk/zigbee2mqtt/files/7873981/zigbeelog.txt)

Koenkk, 2022-01-15:
could you check if upgrading to the latest adapter fw helps?

stcbus, 2022-01-15:
Updated and still really bad. Just was fighting with them all to turn off for like 5 minutes. Attached is the new log. Also, apologies, it is a Slaesh's CC2652RB stick not zzh. I have bought too many and can't keep them straight.
[debuglog2.txt](https://github.com/Koenkk/zigbee2mqtt/files/7875194/debuglog2.txt)

`
Zigbee2MQTT version
    1.22.2-dev commit: 4f887a1

Coordinator type
    zStack3x0

Coordinator revision
    20211217

Coordinator IEEE Address
    0x00124b0022988765

Frontend version
    0.6.68 
`

Koenkk, 2022-01-15:
Next step to debug this is to sniff when these commands are send, looks that there is some hickups on the network.

https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

Please share the `pcapng` file + your network key

stcbus, 2022-01-18:
That was fun to get working! lol. Attached is the capture when I turned on the light group and turned off a couple times and it wouldn't successfully turn off all of them. Also in there are commands from an automation that sets the light temperature and it didn't get to all of the lights either. Key is default
[sniff_lights.pcapng.zip](https://github.com/Koenkk/zigbee2mqtt/files/7890447/sniff_lights.pcapng.zip)
.

Koenkk, 2022-01-18:
@stcbus does this only happen when you toggle the light of groups fast after each other?

stcbus, 2022-01-18:
No, it has happened just when I wanted to turn them all off and it was on for awhile. They don’t seem to ever respond all together now. 

–Steve

Koenkk, 2022-01-18:
@stcbus I don't see anything strange in the log. Could you try to create a light group in HA? Does it work correctly after this? (the difference is that instead of a broadcast commands, a command is send to each bulb individually)

stcbus, 2022-01-18:
That seems to work perfectly when I put them in an HA group. What could
that mean? I really liked the efficiency of Zigbee groups. Could I have
network issues to address?

stcbus, 2022-01-18:
Or maybe the Sengled bulbs kinda suck following the protocol? Just odd it
works sometimes

Koenkk, 2022-01-19:
@stcbus I think they somehow ignore the message, but I don't know why. If other bulbs in your network aren't affected I expect this is a bug in the sengled bulbs.

closed 2022-02-26 by github-actions[bot]

Maxcrouz73, 2024-01-14:
Hello,
I have the same issue when i try to turn off all the lights thought the all off lights button with apple home. I received many time out in Z2m and some lights stay on .