Koenkk/zigbee2mqtt #10802: [Feature request]: Availability expansion ideas

createcandle, 2022-01-15:
### Is your feature request related to a problem? Please describe

I've been upgrading the Webthings Zigbee2MQTT addon, and have been trying to use availability better. My goal is to figure out when light bulbs are turned off with physical switches as quickly as possible.

While looking into it I developed some questions and ideas.

**Questions**
- Does availability actively ping devices/routers? In my tests it didn't seem that way. 
- I've set availability to three minutes, but it seems to take a minute or two longer than that for an availability message to show up after I switch of a bulb. Is a countdown started from the last communication? Or does the feature work by checking in a fixed X minute loop?
- Do battery powered end-devices generally send some kind of heartbeat once per day? Is that a convention? Or will things like leak and smoke sensors, which hopefully never need to send messages, always end up in an unavailable state after a day?
- If so, I was thinking I'd like to distinguish between such "always asleep" devices (e.g. smoke detectors) and sensors that send data a few times a day (e.g. temperature sensors). How could I best do that?

**Ideas**

- I noticed in the debug output that Z2M knows how long ago a device last responded. For example "last seen .13 hours ago". I was wondering if it would be possible to add this data to the availability message. Currently that message is a string, but perhaps it could be a json message instead? Something like:

```
availability: {'state': true, 'last_seen': 1254423935, 'threshold':180}
```

Another advantage: if the output is a json dictionary, then other bits of data could be added later. It's more future proof?

- I've created a pinging function that tries `get` ->`state` every three minutes. I noticed that in the debug output, after about 20 seconds, I can see that Z2M recognises that it didn't get a response from a powered off lamp. I'm assuming that at this point "under the hood" Z2M has been attempting to deliver the  mesage a few times?  

I would love to be able to hook into that awareness in my addon. If a router doesn't respond, then I'd like to immediately set the device to `disconnected`. From what I can tell this would give a much faster response to my UI. Perhaps there could be an MQTT respose to indicate this type or error, separately from the availability system?

** some more questions **

- A question. To my surprise I now own a device that is both a router _and_ battery powered (a smart lock). Is that common? It seems that for my pinging functionality I should check that a device is a router AND not battery powered, correct? Otherwise it might drain the battery?

- Another question: I've been thinking that I could ping each router in a round-robin fashion, where I loop through the list with a 10 second interval. Is that a good idea? In your experience, at what point does this type of pinging becomes detrimental to the network? Are there people/situations with so many Zigbee devices that this would start to interfere? Or are Zigbee message so quick and light that this never becomes an issue? And is there a frequency that you would recommend? What would happen if I ping a device every second?

### Describe the solution you'd like

See above

### Describe alternatives you've considered

See above

About distinguishing the EndDevices that regularly send data, and the ones that do not: I was thinking I could build an algorithm that figures out the "expected incoming data frequency" over time, and use that to figure out if a device has gone offline. 

For example, a smoke sensor could turn out to send 0 messages per day, and be excluded from being labeled "unavailable". 

At the same time, a temperature sensor could send an average number of messages per day. I could then set the threshold automatically based on that.

I was wondering if it would be interesting to have this type of data in the Z2M database, so that this type of functinality could move upstream.

### Additional context

See above

Koenkk, 2022-01-15:
Given the questions are about the availability feature, could you confirm that you are on at least z2m 1.22.0? The availability feature has been heavily refactored there.

Some answers to your questions can also be found here: https://www.zigbee2mqtt.io/guide/configuration/device-availability.html

createcandle, 2022-01-15:
yes I am on 1.22.2

Koenkk, 2022-01-16:
> Does availability actively ping devices/routers? In my tests it didn't seem that way.
yes

> I've set availability to three minutes, but it seems to take a minute or two longer than that for an availability message to show up after I switch of a bulb. Is a countdown started from the last communication? Or does the feature work by checking in a fixed X minute loop?

There is a timer and this timer is reset each time a message is received (so " Is a countdown started from the last communication")

> Do battery powered end-devices generally send some kind of heartbeat once per day? Is that a convention? Or will things like leak and smoke sensors, which hopefully never need to send messages, always end up in an unavailable state after a day?

Not a convention, we assume they send a message at least every 25 hours. I haven't seen any devices that don't (most e.g. report their battery status every day)

> I noticed in the debug output that Z2M knows how long ago a device last responded. For example "last seen .13 hours ago". I was wondering if it would be possible to add this data to the availability [...]

You can use the last_seen feature for this: https://www.zigbee2mqtt.io/guide/configuration/mqtt.html#mqtt-behaviour

> I've created a pinging function that tries get ->state every three minutes. I noticed that in the debug output, after about 20 seconds, I can see that Z2M recognises that it didn't get a response from a powered off lamp: failed (no response received. I'm assuming that at this point "under the hood" Z2M has been attempting to deliver the mesage a few times?

Yes z2m has some recovery built-in, I would not recommend pinging this way since the recovery mechanism can be quite heavy and slow down your network. The availability pinging mechanism has all these recovery disabled when a device is offline.

> To my surprise I now own a device that is both a router and battery powered (a smart lock). Is that common? It seems that for my pinging functionality I should check that a device is a router AND not battery powered, correct? Otherwise it might drain the battery?

Not very common, but each requests drains the battery a bit of course.

> I've been thinking that I could ping each router in a round-robin fashion, where I loop through the list with a 10 second interval. Is that a good idea? In your experience, at what point does this [...]

I don't understand why you want to built your own pinging functionality when the availability feature does exactly this.

Matthy1810, 2022-01-18:
To figure out when light bulbs are turned off with physical switches as quickly as possible you can set the availability timeout (in minutes) in the device section of the configuration file to a low value. In my tests even decimal values lower than 1 worked (e.g. 0.1), but it raises the load in the zigbee network.

uncle-fed, 2022-01-20:
I originally intended to open a separate issue for my concern but I see that it's sort of already being covered here.
It is about the fact that currently the `.../availability` topic payload is a plain text string, not a JSON object and not even a valid JSON string.

I have a lot of automation realized via Node Red that acts as an MQTT subscriber. In order for it to receive messages for particular groups of topics I rely heavily on subscribing via the '#' feature of MQTT. Thus, you subscribe once but to multiple things and decide how to handle certain cases inside Node Red. Unfortunately, if the 'availability' feature is enabled, the MQTT messages related to the same device have mixed payload type now. Most messages will be of course JSON, but then comes the availability message that cannot be decoded as JSON.

The MQTT client implementation in Node Red is able to receive MQTT messages and automatically decode their payload from JSON, however for this to work nicely, you have to tell Node Red MQTT client that you expect _all_ your messages to be JSON formatted. Which means that if you subscribe to multiple topics via '.../#', and the availability feature is ON, you are not able to guarantee that all messages will be JSON compliant anymore. This causes failures in message processing and warnings and errors in Node Red.  There are of course work arounds (like "expect a string instead of JSON", then "decide if it's JSON or not" and then "decide how to proceed") but all of this is rather silly, when you know that 99.99% of all your messages will be JSON. Totally unnecessary extra logic and additional code around every single NodeRed MQTT subscriber module. This becomes daunting very fast.

So my current "workaround" is to disable the availability feature :-) But I really would prefer to use it, so please consider harmonizing zigbee2mqtt payload to always be JSON. Perhaps, quoting `"online" / "offline"` would be a middle ground here because it will make them a valid JSON (but not necessarily an object). But then again, it provokes further questions like, should the other "online" / "offline" messages like the ones from the `zigbee2mqtt/bridge` topic be brought in line with the rest of it then?... So I suppose this is more a request of a general nature (to only use JSON compliant payloads) and not only about the availability messages...

Koenkk, 2022-01-25:
@uncle-fed to me it makes no sense to quote the value, it isn't meant to be JSON. Changing this would also be a breaking change. An alternative solution could be that the availability state is also included in the device state. E.g. topic `zigbee2mqtt/my_device`, payload `{"state": "ON", "available": true}`, but then you have to find a way to ignore all topics ending with `/availability` in nodeRED.

uncle-fed, 2022-01-25:
@Koenkk I understand what you're saying, but the bit about "ignoring all topics ending with /availability" is not possible, not in Node Red, not in any other MQTT client for that matter by the native means of MQTT client (if you're subscribing via '.../#'). Exclusions of topics are not possible AFAIK with MQTT. Which I am sure you're very well aware of.

Subscribing to each device topic individually is also not an option, for those of us that have 50+ devices and the devices are added / removed / names changed rather frequently.

So, I suppose, you mean to say that one should analyse the topic name (once the message has already been received in Node Red) and exclude (or custom-process) those topics which are not needed or not following JSON encoding. However, the "problem" with NodeRed implementation of MQTT client is that _before_ you can start analysing topics and/or payloads, it is already "too late". If you set your MQTT client to automatically decode incoming messages as JSON, then you will instantly get a "broken flow" / errors as soon as the availability message hits NodeRed. This happens directly inside the MQTT client of Node Red, _before_ you're able to do any user processing.

The only possible solution is to stop using JSON automatic deserialization, tell MQTT client to treat all incoming messages as text and then implement custom logic (plus the JSON decoding step, that will be required for 99% of all messages) after receiving each message. This can be done, of course, but instead of an elegant solution where your incoming message is already a deserialized object that you can immediately work with, you will have to implement a silly task of doing really low-level processing with additional multiple Node Red modules every single time, if you want to use the availability feature.

It is a breaking change, I understand. But from architectural point of view (and this GitHub issue is about that, I suppose), it makes it rather strange that z2m should carry its final output in two different "dialects" (JSON and plain text), don't you think? I actually quite like the idea from the first post here that suggest that the availability message should be a JSON object that can be easily extended with more information `{"available": true, "last_seen": "2022-01-25T15:23:02.848Z", "threshold": 3600}`. I think the "last_seen" feature should really belong together with the availability topic, instead of moving it together with the proposed availability state to the level of the device itself.

Yes, this is a very major change, but it is not a random request. Think about the interface in general, especially if you've already decided to go with a separate topic per device for its availability. Think about consolidating the output formatting across the solution .You're saying "the availability message is not supposed to be JSON". That's clear, that it isn't but the question is _why_? Is there some software out there that relies on this decision so heavily that it cannot be changed and that it also cannot decode JSON? I mean, the NodeRed thingy actually is not the real true reason here, it is only a trigger for a bigger picture.

The availability feature (per device) is rather recent and you may want consider still changing its layout before months and years pass and it becomes embedded deeply with other 3rd party software. Then it will of course become "a legacy implementation" that will take years to get rid of. Before it is "too late", please consider harmonisation. JSON is the way, all the way. ;-))

createcandle, 2022-01-26:
> This is not breaking

Just for the record: that would actually be 'breaking' to me. The Webthings addon automatically translates any incoming data into a property on the device. Filtering this out would be easy enough though.

I'm wondering: if you're going to create a `legacy_availability_topic` option, why not just output json to `availability`, and toggle that instead?

How about going 'all in' and creating an `wrap_in_json` advanced option that wraps any output that is currently text-only into a simple json structure? From what I can tell most Z2M topics already output json now (firmware update data used to be string and is now json, if I'm not mistaken). Then you get the advantage of expandability everywhere. In the long run you would need special advanced options less, and could just append new options into the existing data structure. It's hard to add new payloads to a string, but it's easy to add them to json.

uncle-fed, 2022-01-26:
I would be OK with either proposal as long as the consistent state (JSON everywhere) can be reached in the "non-legacy" mode.

Koenkk, 2022-01-26:
> Just for the record: that would actually be 'breaking' to me. The Webthings addon automatically translates any incoming data into a property on the device. Filtering this out would be easy enough though.

I don't see how this is breaking, new properties can be added to the payload all the time (e.g. if a new feature gets added for the device)

> I'm wondering: if you're going to create a legacy_availability_topic option, why not just output json to availability, and toggle that instead?

Because there is no reason anymore to have a separate availability topic (the only reason for this was that HA didn't support JSON for the availability topic but they do now)

uncle-fed, 2022-01-26:
> Include the availability in the state published to zigbee2mqtt/my_device (e.g. {"state": "ON", "available": true}). This is not breaking

On a second thought.. what about the 'retain' flag? At the moment, the availability message is always retained, how will this play together with other device properties published on the same topic / in the same message?

createcandle, 2022-01-26:
> I don't see how this is breaking, new properties can be added to the payload all the time (e.g. if a new feature gets added for the device)

Sure, except a user would expect device properties. A boolean property called "Availability" might not be immediately clear?

I'd prefer to filter it out, since (in this example) the webthings Gateway has a separate way of showing whether devices are online: through opacity of the device itself:

[image]

It's not 'breaking', since I could filter it out and redirect it in two lines of code. I would have to modify the addon to do this. Which I would be happy to do. I just wanted to point out that adding all kinds of new data to the device output would , in my current situation, result in it becoming visible to the user. I means that such changes in Z2M impact the UI and user experience, and I have to pay attention to such changes and 'filter them into the correct channes'.

Personally I quite like the availability topic because it keeps device properties and meta data separated a bit. Personally I would prefer more conceptual separation, not less.

**Device properties** - expected by the user as 'parts of their device'. Automaticallty get translated to device UI. Stable.
- brightness
- state
- tamper
- led_status
- temperature
- battery level
- basically all _expose_ data

**"Meta data"** - higher-level. Implementing it is 'optional'. The device from the perspective of the controller.
- last_seen
- availabilty
- update_available
- signal strength
- etc

It's not black and white.

As I'm building a UI on top of Z2M, then there is value in knowing that output remain stable and predictable. Mixing everything in one topic means I have to do the separation on my end. And it means I have to pay attention to avoid surprises.

To be clear, this is not a critique, and I don't have very strong opinions on this at all. Just sharing a perspective on why separate topics can be a feature for implementers.

> the only reason for this was that HA didn't support JSON for the availability topic but they do now

Ah, I see.

Koenkk, 2022-01-27:
> On a second thought.. what about the 'retain' flag? At the moment, the availability message is always retained, how will this play together with other device properties published on the same topic / in the same message, if I have retain set to false?

Good point, for HA this doesn't matter much because z2m sends all device states when HA comes online, however this is of course not a generic solution. There is another reason why the availability cannot be merged to `bridge/info`, z2m has a MQTT last will message configured on `bridge/state`. The payload of this message is determined upon connecting to the MQTT server (e.g. `{"available": true}`). This means the payload of this message is different then what is normally published to [`bridge/info`](https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html#zigbee2mqtt-bridge-info). 

So I think there is a valid reason to keep the availability topic separate since it would be good to have it always retained and to have a consistent payload for the last will message.

AFAIK the only remaining solution would be to use JSON payloads for `bridge/state` and `my_device/availability`. Although being valid JSON I really don't like the looks of `"online"`, therefore I propose `{"available":"online"}` and `{"available":"offline"}`. To prevent a breaking change an option should be added for this, I propose:

```yaml
advanced:
  # Default is true
  legacy_availability_payload: true
```

uncle-fed, 2022-01-28:
> I think there is a valid reason to keep the availability topic separate since it would be good to have it always retained and to have a consistent payload for the last will message.

> AFAIK the only remaining solution would be to use JSON payloads for bridge/state and my_device/availability. 

Fully agree with the proposal. The only other thought I have would be to move the `last_seen` property to the availability topic as well. My opinion only of course (and it won't change much in my processing routine if it stays or moves) but logically, the `last_seen` truly belongs together with the availability, doesn't it? It could go there only when `legacy_availability_payload` is set to `false`, for example.

Koenkk, 2022-01-28:
@uncle-fed the availability is indeed based on the `last_seen`, however `last_seen` is also available without the availability feature (see `last_seen` option: https://www.zigbee2mqtt.io/guide/configuration/mqtt.html#mqtt-behaviour)

Koenkk, 2022-01-29:
@uncle-fed given that the payload of the`/availability` topic is a JSON object, it can always be easily extended in the feature.  I will start on the `legacy_availability_payload` option implementation.

Koenkk, 2022-02-02:
@uncle-fed implemented this now, docs: https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/guide/configuration/device-availability.md#legacy-availability-payload , please let me know if it works.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

uncle-fed, 2022-02-03:
@Koenkk we got one caveat now. With the JSON payload enabled the web UI displays now '[object Object]' instead of 'online/offline', but I guess this is not what you are usually maintaining...

Koenkk, 2022-02-03:
@uncle-fed 

> Perhaps consider saying at the beginning that the state will be published to a topic without revealing anything about the payload specifics and refer to the section where the actual payload is explained in detail. Or leave it the way it is :-)

Improved

> we got one caveat now. With the JSON payload enabled the web UI displays now '[object Object]' instead of 'online/offline', but I guess this is not what you are usually maintaining...

@nurikk can you fix this in the frontend? TLDR; `bridge/state`/`MY_DEVICE/availability` can be a string or JSON object now  `online`/`offline`

uncle-fed, 2022-02-27:
@Koenkk FYI, the front end is still not ready for displaying the availability from JSON payload. I understand the release will happen shortly with the changes implemented here. Will it be possible to adjust the front end before the release as well? Thanks!

nurikk, 2022-02-27:
I really hate when a variable can be different types ex: string/number/object/whatever. Can we stick to one single type?

uncle-fed, 2022-02-27:
@nurikk we all hate it (hence the long discussion above). But in this case it is controlled by the configuration setting, if this helps.

nurikk, 2022-02-27:
I haven't read entire thread, but why can't we drop legacy format straight away? Availability feature is about two-three months old

uncle-fed, 2022-02-27:
My suggestion was similar (to have every single z2m MQTT message payload as JSON, including the availability messages) and I also argued that the availability feature is too new to already deserve a "legacy" option, but there were arguments to support 2 behaviours ... You can some of it starting from here:  https://github.com/Koenkk/zigbee2mqtt/issues/10802#issuecomment-1022353501
I think some of the HomeAssistant behaviour was built on "string" type already, so it was proposed to keep legacy format as an option. But then again, even Home Assistant now is able to parse JSON availability message, I understood from @Koenkk ...

Koenkk, 2022-02-28:
> I haven't read entire thread, but why can't we drop legacy format straight away? Availability feature is about two-three months old

The old availability feature used the same topic + structure (so it is almost as old as z2m is)

uncle-fed, 2022-03-23:
@nurikk, I see you've implemented Availability reading from JSON in the frontend. Thanks for that!
There is one issue that I get in the main listing of all devices - the sorting order of the "Availability" column in the table behaves strange. If you click on column title, the sorting order should change, but it does not order the items properly. Could you please check it using the JSON availability payload.

closed 2022-04-30 by github-actions[bot]