Koenkk/zigbee2mqtt #10812: [New device support]: Moes ZP-LZ-FR2U dual USB Plug

qp68, 2022-01-15:
### Link

https://www.domadoo.fr/fr/prises-connectees/5838-moes-prise-connectee-zigbee-30-2-ports-usb-pilotables-version-fr.html

### Database entry

{"id":36,"type":"Router","ieeeAddr":"0x04cd15fffe02cb59","nwkAddr":33636,"manufId":4098,"manufName":"_TZ3000_cymsnfvf","powerSource":"Mains (single phase)","modelId":"TS011F","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6,57344,57345],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS011F","manufacturerName":"_TZ3000_cymsnfvf","stackVersion":0,"dateCode":""}},"genOnOff":{"attributes":{"onOff":1}},"wiserDeviceInfo":{"attributes":{}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b002298926b","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":266,"inClusterList":[3,4,5,6,57344,57345],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-353372667},"lastSeen":1642244877940,"useImplicitCheckin":true}

### Comments

Similar to ZK-EU-2U but recognized as a TS011F simple plug (without USB controls) for now.

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {

    fingerprint: [
        {
            modelID: 'TS011F',
            manufacturerName: '_TZ3000_cymsnfvf'
        },
    ],
    model: 'ZP-LZ-FR2U',
    vendor: 'Moes',
    description: 'Zigbee 3.0 dual USB wireless socket plug',
    supports: 'switch',
    fromZigbee: [
        fz.on_off,
        fz.tuya_switch_power_outage_memory,
    ],
    toZigbee: [
        tz.on_off,
        tz.tuya_switch_power_outage_memory,
    ],
   exposes: [
       e.switch().withEndpoint('l1'),
       e.switch().withEndpoint('l2'),
       exposes.enum('power_outage_memory', ea.STATE_SET, ['on', 'off', 'restore']).withDescription('Recover state after power outage')
   ],
    endpoint: (device) => {
        return {'l1': 1, 'l2': 2};
    },
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        await reporting.bind(device.getEndpoint(1),coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        await reporting.onOff(device.getEndpoint(1));
		await reporting.onOff(device.getEndpoint(2));

    },

};

module.exports = definition;
```

Koenkk, 2022-01-15:
_TZ3000_cymsnfvf was already added in https://github.com/Koenkk/zigbee-herdsman-converters/pull/3393 by @toogooda  , can you confirm it is this device: https://www.domadoo.fr/fr/prises-connectees/5838-moes-prise-connectee-zigbee-30-2-ports-usb-pilotables-version-fr.html

qp68, 2022-01-15:
Yes, the data/database.db entry shown in the request correspond to the linked product. So, an outlet with 2 USB ports. Another confirmation on that can be seen there #10640

toogooda, 2022-01-16:
I don't use the USB ports, but didn't think they were smart... Are they controllable using Tuya?

toogooda, 2022-01-16:
Sorry never noticed the USB control as I don't use it...

There is a second endpoint:
[image]
It also has genOnOff so I assume its the USB switch control.
I don't know the code base well but will have a go.

toogooda, 2022-01-16:
OK think I have worked it out,

Made the following external converter:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{modelID: 'TS011F', manufacturerName: '_TZ3000_cymsnfvf'}],
    model: 'ZigbeeSmartPlug With USB', // Vendor model number, look on the device for a model number
    vendor: 'TuYa', // Vendor of the device (only used for documentation and startup logging)
    description: 'Zigbee Smart Pulg With USB', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.on_off, fz.tuya_switch_power_outage_memory],
    toZigbee: [tz.on_off, tz.tuya_switch_power_outage_memory],
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        const endpoint2 = device.getEndpoint(2);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(endpoint2, coordinatorEndpoint, ['genOnOff']);
    },
    exposes: [e.switch().withEndpoint('l1'), e.switch().withEndpoint('l2'), exposes.enum('power_outage_memory', ea.STATE_SET, ['on', 'off', 'restore'])
        .withDescription('Recover state after power outage')],
        whiteLabel: [{vendor: 'UseeLink', model: 'SM-SO306E/K/M'}],
    endpoint: (device) => {
        return {'l1': 1, 'l2': 2};
    },
}
module.exports = definition;
```
Works well but I am new to your code so possibly not  ideal :)

[image]

My only concern is that the two switches are call L1 and L2 by default although you can easily rename them in HA as I did above. I tried renaming 'l1' and 'l2' to 'AC' and 'USB' in the code but got a no converter error. Also tried .

Works well otherwise...

toogooda, 2022-01-16:
> Yes, the data/database.db entry shown in the request correspond to the linked product. So, an outlet with 2 USB ports. Another confirmation on that can be seen there #10640

Yes works on my device also

Koenkk, 2022-01-16:
thanks, added.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-16 by Koenkk

NeuerUser, 2022-02-17:
I have a similar issue. Also got this plug, which is incorrently identified as a Tuya device. But I have a different manufacturerName (_TZ3000_2xlvlnez), probably because mine has a Swiss power plug type (but otherwise identical).

NeuerUser, 2022-02-17:
OK, I used the above shown converter and exchanged the manufacturerName. This seems to work now. I can control both outputs (220V and USB). So, this can be added to the dev branch, I think.

However, the brief episode with the wrong identified Tuya device showed something interesting:
The Tuya device has an endpoint for "child unlock". And this actually did also work on this device!!! Setting this to lock, locks the button on the device. Now, this endpoint seems to be missing in the above shown converter. It would be great, if it could be added, too.