Koenkk/zigbee2mqtt #10813: [New device support]: Moes ZS-EU,  1-3 gang wall switch dimmer

ebcjaa, 2022-01-15:
### Link

https://www.moeshouse.com/collections/new-arrivals/products/new-zigbee-1-gang-eu-smart-light-dimmer-switch-independent-control-smart-life-tuya-app-relay-status-backlight-switch-off-rf-remote-control-works-with-alexa-google-voice-assistants

### Database entry

{"id":45,"type":"Router","ieeeAddr":"0xa4c13899a7ab80a0","nwkAddr":42540,"manufId":4417,"manufName":"_TZE200_vm1gyrso","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":" \u0007\u0000\u0000i\"\u0007\u0000\u0000i","65506":32,"65508":0,"appVersion":64,"modelId":"TS0601","manufacturerName":"_TZE200_vm1gyrso","powerSource":1,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b002298a5a1","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":821693351},"lastSeen":1642253952764,"useImplicitCheckin":true}

### Comments

	I tried to follow the supporting new device page but could not find a any similar converter supporting a 3-gang tuya dimmer... 
tried the  tuya_dimmer and the moes_105_dimmer.

I bought the 3-gan version here:
	https://www.aliexpress.com/item/1005003605347931.html?af=3233875_1&cv=33844362&cn=42r5qrnnhuqx6ijtl9q1v6l5j5u25r5x&dp=v5_42r5qrnnhuqx6ijtl9q1v6l5j5u25r5x&utm_source=epn&utm_medium=cpa&utm_campaign=3233875_1&utm_content=33844362&product_id=1005003605347931&afref=https%3A%2F%2Falipromo.com%2Fredirect%2Fcpa%2Fo%2Fpp9w5byi12y5mxvjc0zdsx8lkaladn1x%2F&aff_fcid=fc06105470e1490f9daad6cd0d258a63-1642233732003-09862-_vXaL3g&tt=API&aff_fsk=_vXaL3g&aff_platform=api-new-link-generate&sk=_vXaL3g&aff_trace_key=fc06105470e1490f9daad6cd0d258a63-1642233732003-09862-_vXaL3g&terminal_id=6a4cba29ac3440a39ba45368a02f972a

gio82c, 2022-01-17:
I have the same device, added from phoscom i have only 1 light on/off

Steefph, 2022-01-25:
Using the attached converter, I get to the following when using dimmer 1 (left button) (still needs more work)

- Debug from moes_105_dimmer, msg.data.dp=[2], msg.data.datatype=[2], value=[450]
- Debug from moes_105_dimmer, msg.data.dp=[2], msg.data.datatype=[2], value=[410]
- Debug from moes_105_dimmer, msg.data.dp=[2], msg.data.datatype=[2], value=[370]
- Debug from moes_105_dimmer, msg.data.dp=[2], msg.data.datatype=[2], value=[330]
- Debug from moes_105_dimmer, msg.data.dp=[2], msg.data.datatype=[2], value=[290]
- Debug from moes_105_dimmer, msg.data.dp=[2], msg.data.datatype=[2], value=[250]
- Debug from moes_105_dimmer, msg.data.dp=[2], msg.data.datatype=[2], value=[240]

[moes_3w_dimmer.txt](https://github.com/Koenkk/zigbee2mqtt/files/7931972/moes_3w_dimmer.txt)

Steefph, 2022-01-25:
The attached now gives a working feedback from input 1 & 2 (3 does not work yet). You can see the actual state but not control it. 

[moes_3w_dimmer.txt](https://github.com/Koenkk/zigbee2mqtt/files/7932097/moes_3w_dimmer.txt)

Steefph, 2022-02-02:
@gio82c do you mean you have 1 working? Which device exactly did you use?