Koenkk/zigbee2mqtt #10814: IKEA LED1546G12 shows as not supported

sebastianengman, 2022-01-15:
### What happened?

Adding IKEA LED1546G12 which is in the list of supported devices, but the bulb shows as not supported.

I figure it has something to do with the weird symbol in the model description?

**From the device page:**

Friendly name
Köket - Matbord
Last seen
N/A
Availability
Online
Device type
Router
Zigbee Model
TRADFRI bulb E27 WS�clear 950lm
Zigbee Manufacturer
IKEA of Sweden
Support status
Not supported

IEEE Address
0x000b57fffe0c994e
Network address
0x64AF
Firmware build date
20211102
Firmware version
2.3.087
Power
Interview completed
True

https://www.zigbee2mqtt.io/devices/LED1546G12.html
[image]

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

26720700

### Adapter

ConBee2

### Debug log

warn  2022-01-15 16:54:50: Received message from unsupported device with Zigbee model 'TRADFRI bulb E27 WS�clear 950lm' and manufacturer name 'IKEA of Sweden'
warn  2022-01-15 16:54:50: Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

Koenkk, 2022-01-15:
could you provide the data/database.db entry of this device?

sebastianengman, 2022-01-15:
Attached!

Koenkk, 2022-01-15:
I don't see it, could you copy paste it here?

sebastianengman, 2022-01-15:
> I don't see it, could you copy paste it here?

Can you see it now?

Koenkk, 2022-01-15:
yes, should be fixed now.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

sebastianengman, 2022-01-15:
> yes, should be fixed now.
> 
> Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

Thank you very much. How do I know when the dev branch update is available?

Also, do I need to delete the device in Z2M or will it get fixed by itself?

Thanks!

Koenkk, 2022-01-15:
> Thank you very much. How do I know when the dev branch update is available?

build is running , will be done in 10 mins for now, so update in 1 hour should be good

> Also, do I need to delete the device in Z2M or will it get fixed by itself?

no repair needed.

sebastianengman, 2022-01-15:
> > Thank you very much. How do I know when the dev branch update is available?
> 
> build is running , will be done in 10 mins for now, so update in 1 hour should be good
> 
> > Also, do I need to delete the device in Z2M or will it get fixed by itself?
> 
> no repair needed.

[image]

Koenkk, 2022-01-15:
perfect!

closed 2022-01-15 by Koenkk