Koenkk/zigbee2mqtt #10815: [New device support]: GIRIER Tuya ZigBee 3.0 Smart Dimmer Switch Module 10A/16A (model 'TS110E', manufacturer '_TZ3210_zxbtub8r')

Jn115759, 2022-01-15:
### Link

https://www.aliexpress.com/snapshot/0.html?orderId=8144137293264913

### Database entry

{"id":26,"type":"Router","ieeeAddr":"0xa4c1388ab6d8139d","nwkAddr":20995,"manufId":4417,"manufName":"_TZ3210_zxbtub8r","powerSource":"Mains (single phase)","modelId":"TS110E","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[4,5,3,6,8,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":"","zclVersion":3}},"genOnOff":{"attributes":{"61440":0,"onOff":0,"onTime":0,"moesStartUpOnOff":2}},"genLevelCtrl":{"attributes":{"61440":1000,"64514":0,"64515":11,"64516":1000,"currentLevel":255}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":2},"lastSeen":1642261130786,"useImplicitCheckin":true}

### Comments

Preamble : many thanks for your support. I am quite new and still learning. Keep up the good work ! Initially started with ZHA but dropped very soon and replaced by Zigbee2MQTT. Much more responsive.

**0. What I bought**

https://www.aliexpress.com/snapshot/0.html?orderId=8144137293264913
GIRIER Tuya ZigBee 3.0 Smart Dimmer Switch Module 10A/16A, Supports 2 Way Control Dimmable Switch
 

**1. Pairing the device with Zigbee2MQTT**

https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_1-pairing-the-device-with-zigbee2mqtt
[image]

 
[image]

 
[image]

Zigbee model 'TS110E', manufacturer name '_TZ3210_zxbtub8r'
Device '0xa4c1388ab6d8139d'

**2. Adding your device**
Tuya page : https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html#_2-adding-your-device

Based in this example below (looks a very similar dimmer switch)
 
[image]

on page https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js

I created the following .js file in the zigbee2mqtt folder

 
[image]

And enabled the external converter ts110e.js  by adding the following in my Zigbee2MQTT configuration.yaml.
 
[image]

Then I restarted Zigbee2MQTT. Now the module is listed as
 
[image]

This was in the log file
 
[image]

When I look for the Device the next is shown. It is possible to toggle the device ( works!) but I an not able to change brightness/dimm
[image]

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require("zigbee-herdsman-converters/lib/tuya");

const definition = {
    fingerprint: [{modelID: 'TS110E', manufacturerName: '_TZ3210_zxbtub8r'}],
    model: 'TS110E_dimmer_module_1ch',
    vendor: 'GIRIER',
    description: 'GIRIER Tuya ZigBee 3.0 1 Channel Smart Dimmer Switch Module 10A/16A',
    fromZigbee: extend.light_onoff_brightness().fromZigbee.concat([fz.tuya_min_brightness]),
    toZigbee: extend.light_onoff_brightness().toZigbee.concat([tz.tuya_min_brightness]),
    exposes: [e.light_brightness().withMinBrightness()],
    extend: extend.light_onoff_brightness(),
};
module.exports = definition;
```

bartekkw, 2022-01-26:
I have created custom converter based on "_TZ3000_xfs39dbf" device
[https://github.com/Koenkk/zigbee-herdsman-converters/blob/43bc1271a581b5f9132cbac381f5d948ca5eabf2/devices/tuya.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/43bc1271a581b5f9132cbac381f5d948ca5eabf2/devices/tuya.js)

It looks like it doesn't have min brightness, but I might be wrong as I have not much experience in mqtt.
So this should allow to toggle on / off and change the brightness. I don't create pull request because it would be better if someone more experienced review it.

```
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const libColor = require('zigbee-herdsman-converters/lib/color');
const utils = require('zigbee-herdsman-converters/lib/utils');

const definition = {
        fingerprint: [{modelID: 'TS110E', manufacturerName: '_TZ3210_zxbtub8r'}],
        model: 'TS110E_dimmer_module_1ch',
        vendor: 'TuYa',
        description: 'Zigbee dimmer module 1 channel',
        whiteLabel: [{vendor: 'OXT', model: 'SWTZ24'}],
        fromZigbee: extend.light_onoff_brightness().fromZigbee.concat([]),
        toZigbee: extend.light_onoff_brightness().toZigbee.concat([]),
        exposes: [e.light_brightness()],
        extend: extend.light_onoff_brightness(),
};

module.exports = definition;

```

Koenkk, 2022-01-27:
@bartekkw does turning on/off and changing brightness work? If  yes, this looks good to me, please create a pull request for it.

Koenkk, 2022-01-27:
Added in https://github.com/Koenkk/zigbee-herdsman-converters/pull/3755 by @kiat-huang 

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-27 by Koenkk

Jn115759, 2022-01-29:
Hi all, many thanks for the help and support. Appreciated !
I changed the content of my original local "ts110e.js" file with the content proposed by @bartekkw but still have no option to dimm the switch. New .js file was loaded correctly since now the device is recognized as Tuya replacing my original "girier" vendor specification.

[image]

Jn115759, 2022-01-29:
@Koenkk , @bartekkw , all : many thanks for the support. Just checked it out and it works. Great !
Lowest brightness (in my case) is at 30 % using custom:slider-entity-row

Ruubke, 2022-08-16:
I have an question about this device.
I got this device a few months ago, I paired it with home assistant and it did work.
But something happend to the firmware. It updated it self with the wrong bin file.
It still works in the sentence of the zigbee connection but lights don't work anymore.
Is there a way to get the bin file for this device?
I don't own a tuya hub

tvdbroeck, 2022-11-11:
I had the same problem after connecting my 1 gang dimmer from Lonsonho/Girier (TS110E) to my zigbee network. Z2M received updates FROM the dimmer, but I could not send commands TO the dimmer.
After creating a dummy group with 1 device, my dimmer, I can send commands to the group like I would to the dimmer. This seems to work fine now.

r4f4w, 2023-02-03:
Does anyone know if I will be able to use a regular on/off switch without using the Tuya app and using Zigbee2mqtt instead? I talked to GIRIER support and they said it will function if I'm using a tuya hub, but I wanted to connect directly to HA without losing any of the primary functions.

Suxsem, 2023-04-05:
@bartekkw thank you! I know about the led issue but I have bought specific dimmerable led bulbs and they work great with other dimmers. But I really hate the fade on effect...dont you have any video about it, do you?

bartekkw, 2023-04-05:
No, unfortunately I don't have :(