Koenkk/zigbee2mqtt #10816: When device are "reconnect" to the netwerk it's configuration is "forgoten" (e.g. prescaler for voltage/current/power)

Happyboy0980, 2022-01-15:
### What happened?

My Develco SPLZB-131 leave the network after 30 min .. few hours after pairing. In aprox 30 seconds the automatily join the netwerk again. 
During pairing the plugs are correctly configured with a prescalers. 

``
Debug Received Zigbee message from 'Koffie_0x0015bc002f000c69', type 'readResponse', cluster 'haElectricalMeasurement', data '{"acCurrentMultiplier":1,"acVoltageDivisor":100,"acVoltageMultiplier":1}' from endpoint 2 with groupID 0
Debug Received Zigbee message from 'Koffie_0x0015bc002f000c69', type 'readResponse', cluster 'haElectricalMeasurement', data '{"acCurrentDivisor":1000,"acPowerDivisor":1,"acPowerMultiplier":1}' from endpoint 2 with groupID 0
``

But when the plug leave the netwerk and rejoin, these prescaler are "lost". 
E.g. voltage is 233.19V becomes 23319 V, Power becomes "null".

When i press the reconfigure button in HA, the plugs correctly get reconfigerd with the right prescaler.
But also then they leave the netwerk after ~ 30 minutes.

In the previous HA / Z2M this issue never occured, also with ZHA this issue doesn't occure. 
Discussed this issue on Discord in the Z2M group seems more user experience the same issue.

### What did you expect to happen?

Would expect Z2M keeps the prescaler somewere stored, or it is reconfigerd during reconnect.

### How to reproduce it (minimal and precise)

1. Pair the plug with Z2M -> Gets correct prescaler
2. Wait 30min .. few hours Plug "leave" the netwerk and will re-join in ~30 sec
3. prescaler is gone

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20211217

### Adapter

SONOFF Zigbee 3.0 USB CC2652P

Happyboy0980, 2022-01-19:
Have "zigbee_herdsman_debug: true" now for 3 days, but don't anything different then above. 
1 thing i noticed after enabling herdsman, that the time it takes before the plugs leave (and re join) takes much longer, e.g. form 30 minutes it now easy takes 6...12 hours before it happens, even a few plugs where at least 24 hours ok.
But issue still is there.

Happyboy0980, 2022-01-24:
any one ? issue is still there.. after aprox 24 hrs all plug show valvue with out the pre-scaler .. (e.g. 23345V)

Happyboy0980, 2022-01-30:
@Koenkk any clue what this can be / how to solve it ?

Koenkk, 2022-01-31:
@Happyboy0980 could you provide the debug log when the device leaves?

Happyboy0980, 2022-02-03:
@Koenkk 

debug log when the plug leave the network, rejoin and paring fails:

```
debug 2022-02-02 16:26:33: Received Zigbee message from 'Vaatwasser_0x0015bc002f000c67', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":23012}' from endpoint 2 with groupID 0
info  2022-02-02 16:26:33: MQTT publish: topic 'zigbee2mqtt/Vaatwasser_0x0015bc002f000c67', payload '{"current":0,"energy":null,"linkquality":80,"power":0,"state":"ON","voltage":23012}'
warn  2022-02-02 16:26:37: Device 'Vaatwasser_0x0015bc002f000c67' left the network
info  2022-02-02 16:26:37: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Vaatwasser_0x0015bc002f000c67","ieee_address":"0x0015bc002f000c67"},"type":"device_leave"}'
info  2022-02-02 16:26:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0x0015bc002f000c67"},"type":"device_removed"}'
info  2022-02-02 16:26:56: Device 'Vaatwasser_0x0015bc002f000c67' joined
info  2022-02-02 16:26:56: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Vaatwasser_0x0015bc002f000c67","ieee_address":"0x0015bc002f000c67"},"type":"device_joined"}'
info  2022-02-02 16:26:56: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"Vaatwasser_0x0015bc002f000c67"},"type":"device_connected"}'
info  2022-02-02 16:26:56: Starting interview of 'Vaatwasser_0x0015bc002f000c67'
info  2022-02-02 16:26:57: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Vaatwasser_0x0015bc002f000c67","ieee_address":"0x0015bc002f000c67","status":"started"},"type":"device_interview"}'
info  2022-02-02 16:26:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"Vaatwasser_0x0015bc002f000c67"},"type":"pairing"}'
debug 2022-02-02 16:26:57: Device 'Vaatwasser_0x0015bc002f000c67' announced itself
debug 2022-02-02 16:26:58: Received Zigbee message from 'Vaatwasser_0x0015bc002f000c67', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":23072}' from endpoint 2 with groupID 0
info  2022-02-02 16:26:58: MQTT publish: topic 'zigbee2mqtt/Vaatwasser_0x0015bc002f000c67', payload '{"current":0,"energy":null,"linkquality":69,"power":0,"state":null,"voltage":23072}'
debug 2022-02-02 16:26:59: Retrieving state of 'Vaatwasser_0x0015bc002f000c67' after reconnect
debug 2022-02-02 16:27:03: Received Zigbee message from 'Vaatwasser_0x0015bc002f000c67', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":23328}' from endpoint 2 with groupID 0
info  2022-02-02 16:27:03: MQTT publish: topic 'zigbee2mqtt/Vaatwasser_0x0015bc002f000c67', payload '{"current":0,"energy":null,"linkquality":83,"power":0,"state":null,"voltage":23328}'
error 2022-02-02 16:27:07: Failed to interview 'Vaatwasser_0x0015bc002f000c67', device has not successfully been paired
info  2022-02-02 16:27:07: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Vaatwasser_0x0015bc002f000c67","ieee_address":"0x0015bc002f000c67","status":"failed"},"type":"device_interview"}'
info  2022-02-02 16:27:07: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"Vaatwasser_0x0015bc002f000c67"},"type":"pairing"}'
debug 2022-02-02 16:27:07: Received Zigbee message from 'Vaatwasser_0x0015bc002f000c67', type 'readResponse', cluster 'genOnOff', data '{"onOff":1}' from endpoint 2 with groupID 0
info  2022-02-02 16:27:07: MQTT publish: topic 'zigbee2mqtt/Vaatwasser_0x0015bc002f000c67', payload '{"current":0,"energy":null,"linkquality":80,"power":0,"state":"ON","voltage":23328}'
debug 2022-02-02 16:27:08: Received Zigbee message from 'Vaatwasser_0x0015bc002f000c67', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":23300}' from endpoint 2 with groupID 0
info  2022-02-02 16:27:08: MQTT publish: topic 'zigbee2mqtt/Vaatwasser_0x0015bc002f000c67', payload '{"current":0,"energy":null,"linkquality":72,"power":0,"state":"ON","voltage":23300}'
debug 2022-02-02 16:27:13: Received Zigbee message from 'Vaatwasser_0x0015bc002f000c67', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":23275}' from endpoint 2 with groupID 0
info  2022-02-02 16:27:13: MQTT publish: topic 'zigbee2mqtt/Vaatwasser_0x0015bc002f000c67', payload '{"current":0,"energy":null,"linkquality":72,"power":0,"state":"ON","voltage":23275}'
debug 2022-02-02 16:27:18: Received Zigbee message from 'Vaatwasser_0x0015bc002f000c67', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":23349}' from endpoint 2 with groupID 0
info  2022-02-02 16:27:18: MQTT publish: topic 'zigbee2mqtt/Vaatwasser_0x0015bc002f000c67', payload '{"current":0,"energy":null,"linkquality":80,"power":0,"state":"ON","voltage":23349}'
debug 2022-02-02 16:27:23: Received Zigbee message from 'Vaatwasser_0x0015bc002f000c67', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":23349}' from endpoint 2 with groupID 0
[... cut ...]
Info MQTT publish: topic 'zigbee2mqtt/WTW_0x0015bc002f000c19', payload '{"current":0.01,"energy":0.56,"linkquality":131,"power":1,"state":"ON","voltage":232.14}'
Debug Received Zigbee message from 'WTW_0x0015bc002f000c19', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":23214}' from endpoint 2 with groupID 0
Info MQTT publish: topic 'zigbee2mqtt/WTW_0x0015bc002f000c19', payload '{"current":0,"energy":0.56,"linkquality":123,"power":0,"state":"ON","voltage":232.14}'
```

Koenkk, 2022-02-04:
@timmikk do you still use z2m? If yes, can you check if this device still works correctly when pairing it to z2m 1.23.0?

timmikk, 2022-02-04:
@Koenkk 

Yes I have one develco plug in z2m but I have had older version of z2m in use. Just updated to 1.23.0 after saw your comment. Will be back if I notice any issues.

Happyboy0980, 2022-02-07:
It pairs correctly as far i can see, it works ok, the prescaler is set correctly as can see in the log. But after aprox 30 min..1 hr it is kicked out of the networks,  joins in approx 30 sec by it self and than paring fails / prescaler is not set. Can this be caused because paring is then disabled in Z2M ?

Koenkk, 2022-02-07:
> Can this be caused because paring is then disabled in Z2M ?

I don't expect so, device should stay in the network. Do you perhaps have another dongle to sniff your zigbee traffic, it would be interesting to see what happens when the device leaves: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html#with-cc2531

Happyboy0980, 2022-02-08:
Have everthing running. 

First press the "reconfigure button"  so that the reading are correct.
After his waited until the Develco "left"  the network and re-join.

Quite quickly found the "point"  where is seems to go wrong.

First the coordinator sends : 

1. Many-to-One Route Request, Dst: 0xfffc, Src: 0x0000 
2. Short time later coordinator Broadcast "ZCL Green Power : GP Proxy Commissioning Mode, Seq 226 
3. Coordinator send "Permit Join request" 
4. Only the Develco sends :  Permit Join Response, Status: Success

After this the presets are gone. 

Full log for reference :

[Log2.zip](https://github.com/Koenkk/zigbee2mqtt/files/8027369/Log2.zip)

Koenkk, 2022-02-09:
I think the device leaves because the coordinator does not expose the time cluster:

[image]

[image]

What setup are you running z2m on? Then I can provide some instructions to add this cluster.

Happyboy0980, 2022-02-09:
Nice catch.. spend couple hours yesterday reading and comparing, missed that one 

Running HA Core 2021-12.10 with Zigbee2mqtt 1.23.0-1 (1.22 and 1.23 have simulair results).
SONOFF Zigbee 3.0 USB CC2652P with 20211217.

Happyboy0980, 2022-02-11:
Implement above 12 hours ago, so far nothing left the network any more. And network "looks" less busy.. Prescallers are and stay correct. So looks you nailed it :), thanks !
Will keep monitor for 24 hours, think then we are very sure.

Incase somebody else have the same issue, in my case it was:

cd **app**/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter
vi endpoint**s**.js

just currious any clue why the coordinator didn't exposse the timecluster ?

Koenkk, 2022-02-11:
Great, added this now

> just currious any clue why the coordinator didn't exposse the timecluster ?

Because no device required this yet.

Happyboy0980, 2022-02-12:
More then running 24 hrs without any issue now. This one is solved and can be closed.
@Koenkk Thanks ! Learned a lot, and all working fine now.

closed 2022-02-12 by Happyboy0980