Koenkk/zigbee2mqtt #10820: [Feature request]: Could led_enable's display name be "led" to make it easier to work with voice control?

flatsiedatsie, 2022-01-15:
### Is your feature request related to a problem? Please describe

The Ikea air purifier has a propertiy called `led_enable`. It's `name is the same.

In the Webthings gateway users cannot change the names of properties of devices, which means the device has this:

[image]

The local voice control system allows users to interact with properties by using the display name. For example:

`What is the state of the air purifier?` or `enable the child lock of the air purifier`.

However, this gets a little weird with `Led enable`, since voice control sees `enable` as an actuator word. To turn on the led with their voice, the user could in theory say "enable led enable". In practise they'll probably say "turn on led enadble" or something to that effect, but it still a bit strange.

### Describe the solution you'd like

Would it be an idea to change the expose `name` from `led enable` to just `led`? If `led` is too short, how about `led indicator` or `status light` or...

### Describe alternatives you've considered

This is of course totally not important, and I can easily fix this in the Zigbee2MQTT addon. But I thought I'd share the idea. When I did a search I saw that the led_enable name occurs in multiple devices.

### Additional context

So far all the other properties I've seen in Z2M work great with voice control.

sjorge, 2022-01-15:
The cluster value actually got renamed to `controlPanelLight` from the old `ledEnabled`.

@Koenkk would it be considered a breaking change to just rename the property to `control_panel_light` ?

Koenkk, 2022-01-16:
@sjorge there are multiple properties that and with `_enable` (e.g. `pir_enable`) and I don't think we should rename those. This is a very specific problem, it sounds more logical to me if webthings supports a feature where you can rename those sensors and can use those in the voice recognition (in HA I also do it this way)

sjorge, 2022-01-16:
Fine by me, I think I originally had led_enabled and change it to _enable to match other like pir_enable (without d)

flatsiedatsie, 2022-01-18:
> it sounds more logical to me if webthings supports a feature where you can rename those sensors

I fully agree. Unfortunately that feature isn't there right now.

I'll try to fix it in the Z2M addon, no problem. I just wanted to make sure it wasn't better to do it upstream first.

closed 2022-02-20 by flatsiedatsie