Koenkk/zigbee2mqtt #10822: Neo NAS-PD07 incorrectly recognized as TuYa TS2020 "manufName":"_TZ3000_kmh5qpmb" "modelId":"TS0202"

E1m0W, 2022-01-15:
### What happened?

I just received 5pcs of [Neo NAS-PD07](https://www.zigbee2mqtt.io/devices/NAS-PD07.html)  motion,humidity and temperature sensors, but for some reason Z2M recognizes them as [TuYa TS2020](https://www.zigbee2mqtt.io/devices/TS0202.html) motion sensors (and thus I get no temp and humidity, and seems that the tampering might not work correctly when recognized as TS0202).

```
"manufName":"_TZ3000_kmh5qpmb"
"modelId":"TS0202"
```

I have never had any other device incorrectly recognized. 

There is Neo NAS-PD07 recognized in the converters, (the last one in [neo.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/neo.js), but the information to recognize the device is in the [tuya.js](https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/devices/tuya.js) TS0202 section.

I updated Z2M today with the master branch 1.22.2 (before that it was not recognized at all).

The database.db shows these items as:

`{"id":10,"type":"EndDevice","ieeeAddr":"0xa4c138171e4fc6a2","nwkAddr":53374,"manufId":4417,"manufName":"_TZ3000_kmh5qpmb","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,1280,3,0],"outClusterList":[4096,6,25,10],"clusters":{"genBasic":{"dir":{"value":1},"attributes":{"stackVersion":0,"dateCode":""}},"genPowerCfg":{"dir":{"value":1},"attributes":{}},"genIdentify":{"dir":{"value":1},"attributes":{}},"genOnOff":{"dir":{"value":2},"attributes":{}},"genTime":{"dir":{"value":2},"attributes":{}},"genOta":{"dir":{"value":2},"attributes":{}},"ssIasZone":{"dir":{"value":1},"attributes":{"61441":1,"iasCieAddr":"0x00124b00193685b6","zoneState":1,"currentZoneSensitivityLevel":1}},"lightLink":{"dir":{"value":2},"attributes":{}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1642272145699}`

`{"id":11,"type":"EndDevice","ieeeAddr":"0xa4c138c80243f2d4","nwkAddr":31192,"manufId":4417,"manufName":"_TZ3000_kmh5qpmb","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,1280,3,0],"outClusterList":[4096,6,25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}},"ssIasZone":{"attributes":{"61441":1,"iasCieAddr":"0x00124b00193685b6","zoneState":1,"currentZoneSensitivityLevel":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1642281335212,"useImplicitCheckin":true}`

`{"id":12,"type":"EndDevice","ieeeAddr":"0xa4c1387ecc1522e9","nwkAddr":17418,"manufId":4417,"manufName":"_TZ3000_kmh5qpmb","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,1280,3,0],"outClusterList":[4096,6,25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0202","manufacturerName":"_TZ3000_kmh5qpmb","powerSource":3,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}},"ssIasZone":{"attributes":{"61441":1,"iasCieAddr":"0x00124b00193685b6","zoneState":1,"currentZoneSensitivityLevel":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1642281492887,"useImplicitCheckin":true}`

`{"id":13,"type":"EndDevice","ieeeAddr":"0xa4c1389ab6f05762","nwkAddr":35207,"manufId":4417,"manufName":"_TZ3000_kmh5qpmb","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,1280,3,0],"outClusterList":[4096,6,25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0202","manufacturerName":"_TZ3000_kmh5qpmb","powerSource":3,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}},"ssIasZone":{"attributes":{"61441":1,"iasCieAddr":"0x00124b00193685b6","zoneState":1,"currentZoneSensitivityLevel":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1642281584023,"useImplicitCheckin":true}`

`{"id":14,"type":"EndDevice","ieeeAddr":"0xa4c138073451c367","nwkAddr":20215,"manufId":4417,"manufName":"_TZ3000_kmh5qpmb","powerSource":"Battery","modelId":"TS0202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[1,1280,3,0],"outClusterList":[4096,6,25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0202","manufacturerName":"_TZ3000_kmh5qpmb","powerSource":3,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b00193685b6","zoneState":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1642281650708,"useImplicitCheckin":true}`

### What did you expect to happen?

Them to be recognized as NAS-PD07 and have motion, temperature and humidity sensor data. 
Now i apparently only get the motion data, and possibly tampering and battery level. 

### How to reproduce it (minimal and precise)

After unboxing I set the zigbee2mqtt to allow join, plugged the sensors to usb power source and pressed reset button for 10s for each of the Neo NAS-PD07 sensors until they appeared in Homeassistant.

I have zigbee2mqtt running with Mosquitto on Raspberry PI 3. 
HA is running in docker
(unfortunately I'm not able to flash the adapter FW version to the latest at the moment.)
[image]
.

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20181024

### Adapter

CC2351 stick

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/7875931/log.txt)

Koenkk, 2022-01-16:
Could you check if this issue has been fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('../lib/tuya');

const definition = {
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_7hfcudw5'},
        {modelID: 'TS0202', manufacturerName: '_TZ3000_kmh5qpmb'}],
    model: 'NAS-PD07',
    vendor: 'Neo',
    description: 'Motion, temperature & humidity sensor',
    fromZigbee: [fz.neo_nas_pd07],
    toZigbee: [],
    onEvent: tuya.onEventSetTime,
    exposes: [e.occupancy(), e.humidity(), e.temperature(), e.tamper(), e.battery_low(),
        exposes.enum('power_type', ea.STATE, ['battery_full', 'battery_high', 'battery_medium', 'battery_low', 'usb'])],
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, go to the frontend, verify that the description of the device ends with CUSTOM (this means that external converter has been loaded correctly)
- press the yellow reconfigure button
- does it report the status now?

E1m0W, 2022-01-16:
I copied the code exactly to the ext_converter.js that is now in the /opt/zigbee2mqtt/data -folder with the configuration.yaml
`[USERNAME]@raspberrypi:/opt/zigbee2mqtt/data $ ls
configuration.yaml  coordinator_backup.json  database.db  database.db.backup  ext_converter.js  log  state.json
`

And I got these errors with an endless restart loop for Z2M:
`  JAN 16 12:32:29 raspberrypi- npm[3639]:     at Module.require (internal/modules/cjs/loader.js:692:17)
JAN 16 12:32:29 raspberrypi- npm[3639]:     at require (internal/modules/cjs/helpers.js:25:18)
JAN 16 12:32:29 raspberrypi- npm[3639]:     at /opt/zigbee2mqtt/dist/util/externally-loaded.js:8:14
JAN 16 12:32:29 raspberrypi- npm[3639]:     at Script.runInContext (vm.js:133:20)
JAN 16 12:32:29 raspberrypi- npm[3639]:     at Script.runInNewContext (vm.js:139:17)
JAN 16 12:32:29 raspberrypi- npm[3639]:     at Object.runInNewContext (vm.js:322:38)
JAN 16 12:32:29 raspberrypi- npm[3639]:     at loadModuleFromText (/opt/zigbee2mqtt/dist/util/utils.js:159:18)
JAN 16 12:32:29 raspberrypi- npm[3639]:     at loadModuleFromFile (/opt/zigbee2mqtt/dist/util/utils.js:165:12)
JAN 16 12:32:29 raspberrypi- npm[3639]: (node:3651) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
JAN 16 12:32:29 raspberrypi- npm[3639]: (node:3651) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
`

................................
( When I set the external_converters inside of advanced I got Z2M to start again

`advanced:
  external_converters:
    - ext_converter.js
`
but it did recognize them as the same Tuya models, with the same outputs, so I think that it is not the correct place in the code :D

` Zigbee2MQTT:debug 2022-01-16 12:46:27: Received MQTT message on 'homeassistant/sensor/0xa4c138073451c367/linkquality/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xa4c138073451c367"],"manufacturer":"TuYa","model":"Motion sensor (TS0202)","name":"0xa4c138073451c367"},"enabled_by_default":false,"entity_category` )

Koenkk, 2022-01-16:
@E1m0W think I found the issue, I've updated https://github.com/Koenkk/zigbee2mqtt/issues/10822#issuecomment-1013831157

E1m0W, 2022-01-16:
Thanks for the update. Now the sensors are showing in the logs with the CUSTOM e.g.
`debug 2022-01-16 18:59:27: Received MQTT message on 'homeassistant/sensor/0xa4c138171e4fc6a2/linkquality/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xa4c138171e4fc6a2"],"manufacturer":"Neo","model":"Motion, temperature & humidity sensor CUSTOM (NAS-PD07)","name":"0xa4c138171e4fc6a2"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0xa4c138171e4fc6a2","name":"0xa4c138171e4fc6a2 linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138171e4fc6a2","unique_id":"0xa4c138171e4fc6a2_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'`

but they do not carry any values from the sensors (I think that they did not do that previously either, but the initial state just gets changed and that made me think that they did). 

Here is a snip from the sensor in HA. You can see that only Battery low, Occupancy and Tamper show any values (and even they don't update).
[image]

Sorry about the messy log (have some devices in the config and db that are not currently in the network) but you get the idea.

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/7877488/log.txt)

Koenkk, 2022-01-16:
Then I'm afraid this device requires new converters since it uses different data points. Some documentation on how to add support for such devices: 
- https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html#instructions
- https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html#requirements-and-caveats

E1m0W, 2022-01-16:
This might actually be problem with my outdated coordinator (CC2531) as it is running Zigbee 1.2 and the devices are Zigbee 3.0.
I will get a new coordinator with Zigbee 3.0 and report back.

E1m0W, 2022-02-03:
I finally got my new Sonoff stick and the results are pretty much the same as before.
In the meantime I had a couple of them working as just motion detectors, and they did a decent job. A bit sluggish, but worked, until I upgraded my CC2531 to a Sonoff stick.

The only difference is that for some odd reason one of the sensors no longer trigger the motion in Homeassistant (or more accurately the message is not converted correctly, this is without the ext_converter.js):
From log
[log2.txt](https://github.com/Koenkk/zigbee2mqtt/files/7997406/log2.txt)

The one that works correctly (regarding to motion):
`debug 2022-02-03 20:56:25: Received Zigbee message from '0xa4c138073451c367', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
info  2022-0
[log2.txt](https://github.com/Koenkk/zigbee2mqtt/files/7997406/log2.txt)
2-03 20:56:25: MQTT publish: topic 'zigbee2mqtt/0xa4c138073451c367', payload '{"battery":null,"battery_low":false,"linkquality":72,"occupancy":true,"tamper":false}'
debug 2022-02-03 20:56:29: Received Zigbee message from '0xa4c138073451c367', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":0}' from endpoint 1 with groupID 0
info  2022-02-03 20:56:29: MQTT publish: topic 'zigbee2mqtt/0xa4c138073451c367', payload '{"battery":null,"battery_low":false,"linkquality":75,"occupancy":false,"tamper":false}'`

The one that does not anymore work correctly even regarding to motion.
debug 2022-02-03 20:55:45: Retrieving state of '0xa4c1387ecc1522e9' after reconnect
debug 2022-02-03 20:55:49: Received Zigbee message from '0xa4c1387ecc1522e9', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2022-02-03 20:55:50: Received Zigbee message from '0xa4c1387ecc1522e9', type 'attributeReport', cluster 'ssIasZone', data '{"61441":1}' from endpoint 1 with groupID 0
info  2022-02-03 20:55:50: MQTT publish: topic 'zigbee2mqtt/0xa4c1387ecc1522e9', payload '{"battery":null,"battery_low":false,"linkquality":54,"occupancy":false,"tamper":false}'
debug 2022-02-03 20:55:50: Received Zigbee message from '0xa4c1387ecc1522e9', type 'attributeReport', cluster 'ssIasZone', data '{"currentZoneSensitivityLevel":1}' from endpoint 1 with groupID 0
info  2022-02-03 20:55:50: MQTT publish: topic 'zigbee2mqtt/0xa4c1387ecc1522e9', payload '{"battery":null,"battery_low":false,"linkquality":51,"occupancy":false,"tamper":false}'

I don't see any traffic that could be traced to humidity or temperature, nor tampering. The only data comes when there is some motion. 

This is data is with the ext_converter.js mentioned earlier in this thread:

```
debug 2022-02-03 20:32:33: Received Zigbee message from '0xa4c138073451c367', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
debug 2022-02-03 20:32:33: No converter available for 'NAS-PD07' with cluster 'ssIasZone' and type 'commandStatusChangeNotification' and data '{"extendedstatus":0,"zonestatus":1}'
debug 2022-02-03 20:32:39: Received Zigbee message from '0xa4c138073451c367', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":0}' from endpoint 1 with groupID 0
debug 2022-02-03 20:32:39: No converter available for 'NAS-PD07' with cluster 'ssIasZone' and type 'commandStatusChangeNotification' and data '{"extendedstatus":0,"zonestatus":0}'
```
From log
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/7997312/log.txt)

Koenkk, 2022-02-04:
> The one that does not anymore work correctly even regarding to motion.

Does this one report also something after there has been no motion for some time?

E1m0W, 2022-02-04:
I recreated the situation last night and two of the sensors had that problem. Unfortunately I cleared the logs afterwards, but as I paired them all again (5pcs) (after a blackout that happened today) and tried to reproduce the situation, and now the motion seems to be working correctly with all of them.

I believe that in order to check the humidity and temperature (and tampering) detection I need to get a TuYa gateway and debug through that as there is no indication of any messages (besaids motion) coming through at the moment.

Rubquila, 2022-02-24:
I have the same problem.

> Could you check if this issue has been fixed with the following external converter:
> 
> ```js
> const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
> const tz = require('zigbee-herdsman-converters/converters/toZigbee');
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const reporting = require('zigbee-herdsman-converters/lib/reporting');
> const extend = require('zigbee-herdsman-converters/lib/extend');
> const e = exposes.presets;
> const ea = exposes.access;
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> 
> const definition = {
>     fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_7hfcudw5'},
>         {modelID: 'TS0202', manufacturerName: '_TZ3000_kmh5qpmb'}],
>     model: 'NAS-PD07',
>     vendor: 'Neo',
>     description: 'Motion, temperature & humidity sensor CUSTOM',
>     fromZigbee: [fz.neo_nas_pd07],
>     toZigbee: [],
>     onEvent: tuya.onEventSetTime,
>     exposes: [e.occupancy(), e.humidity(), e.temperature(), e.tamper(), e.battery_low(),
>         exposes.enum('power_type', ea.STATE, ['battery_full', 'battery_high', 'battery_medium', 'battery_low', 'usb'])],
> };
> 
> module.exports = definition;
> ```
> 
> * save this as file next to `configuration.yaml` as `ext_converter.js`
> * add it to `configuration.yaml`:
> 
> ```yaml
> external_converters:
>   - ext_converter.js
> ```
> 
> * start z2m, go to the frontend, verify that the description of the device ends with CUSTOM (this means that external converter has been loaded correctly)
> * does it report all values now?

I've tried this but it still doesn't work. Report this:
`Received Zigbee message from '0xa4c138c54c6cff74', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":0}' from endpoint 1 with groupID 0
Debug No converter available for 'NAS-PD07' with cluster 'ssIasZone' and type 'commandStatusChangeNotification' and data '{"extendedstatus":0,"zonestatus":0}'
Debug Received Zigbee message from '0xa4c138c54c6cff74', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":1}' from endpoint 1 with groupID 0
Debug No converter available for 'NAS-PD07' with cluster 'ssIasZone' and type 'commandStatusChangeNotification' and data '{"extendedstatus":0,"zonestatus":1}'`
(It seems that "zonestatus:1" it's movement detected)

Jordiete, 2022-03-05:
Hi, pleased to join you.
I had the same problem and  I bought a Tuya coordinator. 
After debugging I have found there is not data but PIR and Battery. No tamper, no temp, no humidity. 
[image]
I think some of NAS-PD07 sensors solded at Aliexpress are false.
The box said clearly it has temp and humidity sensor, but it doesn`t work in Tuya.
[image]

closed 2022-04-13 by github-actions[bot]

pelizz, 2022-06-02:
Jordiete...Have you open the device and compared?  I have the same device (from Aliexpress) with the same problem

pelizz, 2022-06-03:
> Hi, pleased to join you. I had the same problem and I bought a Tuya coordinator. After debugging I have found there is not data but PIR and Battery. No tamper, no temp, no humidity. [image] I think some of NAS-PD07 sensors solded at Aliexpress are false. The box said clearly it has temp and humidity sensor, but it doesn`t work in Tuya. [image] I have another that works so I will open them to compare.

Jordiete...Have you open the device and compared? I have the same device (from Aliexpress) with the same problem

Jordiete, 2022-06-03:
[image]
[image]
The second is the original one, the first as you can see is so simple compared with the original.

Jordiete, 2022-10-16:
https://es.aliexpress.com/item/1005002278127663.html?spm=a2g0o.order_detail.0.0.5a9139d3LyUsAm&gatewayAdapt=glo2esp

10der, 2023-06-02:
hi! issue still here.
a new brand NEO COOLCAM still as Neo NAS-PD07 incorrectly recognized as TuYa TS2020 "manufName":"_TZ3000_kmh5qpmb" "modelId":"TS0202"

i am trying to add the custom convertor but with no luck
if i am trying to use `fz.neo_nas_pd07` in 
`fromZigbee: [fz.neo_nas_pd07, fz.ignore_tuya_set_time, fz.ias_occupancy_alarm_1],`
it failed on the start like a 
```
TypeError: Cannot read properties of undefined (reading 'options')
```