Koenkk/zigbee2mqtt #10823: [New device support]: Livarno Smart LED Mood Light (14156506L)

quenthal, 2022-01-15:
### Link

https://www.lidl.fi/p/product-recommendation/livarno-home-smart-home-led-tunnelmavalo/p40581

### Database entry

{"id":59,"type":"Router","ieeeAddr":"0x04cd15fffea26043","nwkAddr":40474,"manufId":4098,"manufName":"_TZ3210_r0xgkft5","powerSource":"Mains (single phase)","modelId":"TS0505B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,4096,8,768,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505B","manufacturerName":"_TZ3210_r0xgkft5","stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":90,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":},"lastSeen":1642286386782,"useImplicitCheckin":true}

### Comments

From the looks of it, this seems identical to https://www.zigbee2mqtt.io/devices/14148906L.html.
Only difference I could make is the model name: 14148906L -> 14156506L. Also the manuf. name is different, altough this is also Lidl/Livarno device: "manufName":"_TZ3210_r0xgkft5".

There is battery with this in addition to use it from the mains.

(As this was the first Zigbee device ever that didn't work straight with z2m, I had some problems trying to add this device by hand. Whenever I made .js file, zigbee2mqtt simply didn't start whenever I added external_converters to configuration.yaml.)

quenthal, 2022-01-17:
I was still unable to add external converter. However, the light works as it should when I added following to zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/lidl.js:

    {
        fingerprint: [{modelID: 'TS0505B', manufacturerName: '_TZ3210_r0xgkft5'}],
        model: '14156506L',
        vendor: 'Lidl',
        description: 'Livarno Smart LED Mood Light',
        ...extend.light_onoff_brightness_colortemp_color({disableColorTempStartup: true}),
        meta: {applyRedFix: true, enhancedHue: false},
        configure: async (device, coordinatorEndpoint, logger) => {
            device.getEndpoint(1).saveClusterAttributeKeyValue('lightingColorCtrl', {colorCapabilities: 29});
        },
    },

Koenkk, 2022-01-17:
Could you provide the color temp range? https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

quenthal, 2022-01-17:
> Could you provide the color temp range? https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

Read result of 'lightingColorCtrl': {"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}

Koenkk, 2022-01-17:
thanks, added.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2022-01-17 by Koenkk