Koenkk/zigbee2mqtt #10828: Why motion detection ( aqara ) have 2 lines in logs ?

kevincaradant, 2022-01-16:
### What happened?

```
Zigbee2MQTT:info  2022-01-16 11:03:32: MQTT publish: topic 'zigbee2mqtt/motion_hotte_kitchen', payload '{"battery":100,"illuminance":29,"illuminance_lux":29,"linkquality":96,"occupancy":true,"temperature":24,"voltage":3015}'
Zigbee2MQTT:info  2022-01-16 11:03:32: MQTT publish: topic 'zigbee2mqtt/motion_hotte_kitchen', payload '{"battery":100,"illuminance":29,"illuminance_lux":29,"linkquality":99,"occupancy":true,"temperature":24,"voltage":3015}'
```

i know for some switches or devices with actions, we can have twice logs because we have one with `action: ""` and one with `action: 'single'` . It's for Home Assistant if my memory is good. 

But is it normal for motion detection to have twice logs with no difference in the display of the log? 

Thank you 

Rgd

### What did you expect to happen?

have only one line 

### Zigbee2MQTT version

1.22.2

### Adapter firmware version

20211210

### Adapter

Launchpad CC1352-P2

Koenkk, 2022-01-16:
To debug this a debug log of this is required.

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

kevincaradant, 2022-01-16:
Sorry I thought to have paste it in the debug section:

```
Zigbee2MQTT:debug 2022-01-16 16:55:40: Received Zigbee message from 'motion_desktop', type 'attributeReport', cluster 'msIlluminanceMeasurement', data '{"measuredValue":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-01-16 16:55:40: MQTT publish: topic 'zigbee2mqtt/motion_desktop', payload '{"battery":100,"illuminance":0,"illuminance_lux":0,"linkquality":72,"occupancy":true,"temperature":21,"voltage":3085}'
Zigbee2MQTT:debug 2022-01-16 16:55:40: Received Zigbee message from 'motion_desktop', type 'attributeReport', cluster 'msOccupancySensing', data '{"occupancy":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2022-01-16 16:55:40: MQTT publish: topic 'zigbee2mqtt/motion_desktop', payload '{"battery":100,"illuminance":0,"illuminance_lux":0,"linkquality":76,"occupancy":true,"temperature":21,"voltage":3085}'
```

Koenkk, 2022-01-16:
As can be seen in the log, the device sends 2 message, first is an update of the illuminance value, second is the occupancy detection message.

kevincaradant, 2022-01-16:
It's not possible to merge the message if it's in the same time?

If it's normal, ni problèm but before I was pretty sure to have only one ( maybe I'm wrong)

Koenkk, 2022-01-16:
It is not at exactly the same time, they are really 2 separate messages.

However you can merge them at the z2m side by using the debounce functionallity, see https://www.zigbee2mqtt.io/guide/configuration/devices-groups.html#common-device-options

closed 2022-01-16 by Koenkk