Koenkk/zigbee2mqtt #10850: Moes TRV model TS0601 /  _TZE200_zion52ef   support requested

olili, 2022-01-17:
### Link

https://zigbee.blakadder.com/Moes_SH4-ZB.html

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0x5c0272fffec20dfb","nwkAddr":29060,"manufId":4098,"manufName":"_TZE200_zion52ef","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,10,4,5,61184],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"�T�'\u0012�T�'\u0012�Y�'\u0011","65506":27,"modelId":"TS0601","manufacturerName":"_TZE200_zion52ef","powerSource":3,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1612951058691}

### Comments

reopen of https://github.com/Koenkk/zigbee2mqtt/issues/6211

Device still not supported by Z2M. 
(ZHA seems to support it already)

Can anybody help?

Thx in advance.

closed 2022-02-24 by github-actions[bot]

nwrkbiz, 2022-11-17:
I own the same device and can provide help to get this supported if needed.

gklamut, 2022-11-26:
I would appreciate : my model in tuya is SH4 Zigbee eTRV (fhn3negr) in HA it is _TZE200_zion52ef  I use Z2Mqtt.
Any  help?

olili, 2023-01-06:
Update: 

found this External Converter: https://gist.github.com/serrj-sv/af142b25de2d7ac54c3a2eb2623d9a6d

It is working also with _TZE200_zion52ef 
Needed only some smaller changes for running with current z2m version.
Id est:
line 335: _TZE200_zion52ef  
                INSTEAD _TZE200_fhn3negr
line 295:   const output= new Uint8Array(18); 
                INSTEAD   const output= new Buffer(17);
line 302:   if (Index < 8) { output[2+2*Index]=el.hour*4+Math.floor((el.minute/15)); } 
                INSTEAD output[2+2*Index]=el.hour*4+Math.floor((el.minute/15));
line 259:   const output= new Uint8Array(8);  
                INSTEAD  const output= new Buffer(8);

Simply modify accordingly, copy to data register and include as external converter.

Maybe someone can incorporate it into herdsman-converter, My skills were not sufficient. 

O.

lee-js, 2023-10-01:
Nearly 3 years later and this device is still not included in Z2M. Is there any way someone could help get this working? I have three of these which eventually worked in ZHA but I've now been forced to migrate to Z2M. These are now the only devices I have that are detected but not recognised.

I will give the suggestion above a go but I'd prefer it if it could just be auto-detected :(

olili, 2023-10-03:
attached external converter is working with the recent z2m for me. 

As told - my skills are not sufficient to properly include it to herdsmann lib. 

O.

[tuya_sh4_etrv.zip](https://github.com/Koenkk/zigbee2mqtt/files/12790891/tuya_sh4_etrv.zip)