Koenkk/zigbee2mqtt #10853: (node:2533) TimeoutOverflowWarning: 5400001000 does not fit into a 32-bit signed integer

khaidakin, 2022-01-17:
### What happened?

When I connect the WSDCGQ01LM temperature sensor, the error "(node:2533) TimeoutOverflowWarning: 5400001000 does not fit into a 32-bit signed integer.
Timeout duration was set to 1." 

### What did you expect to happen?

No errors

### How to reproduce it (minimal and precise)

Connect the WSDCGQ01LM sensor to the coordinator

### Zigbee2MQTT version

1.22.2-dev

### Adapter firmware version

20211222

### Adapter

cc2538, EFR32

### Debug log

https://pastebin.com/GdZnmpxR

khaidakin, 2022-01-31:
I installed the system and zigbee2mqtt (1.22.2-dev commit: 9fc413de) on another server from scratch. The error remains.
`Zigbee2MQTT:info  2022-01-31 16:07:03: MQTT publish: topic 'zigbee2mqtt-1/0x00158d0003a27999', payload '{"battery":74,"humidity":57.29,"last_seen":"2022-01-31T16:07:03+02:00","linkquality":255,"temperature":22.36,"voltage":2955}'
(node:905) TimeoutOverflowWarning: 5400001000 does not fit into a 32-bit signed integer.
Timeout duration was set to 1.
Zigbee2MQTT:info  2022-01-31 16:07:07: MQTT publish: topic 'zigbee2mqtt-1/0x00158d0003a27999', payload '{"battery":74,"humidity":57.29,"last_seen":"2022-01-31T16:07:07+02:00","linkquality":255,"temperature":22.01,"voltage":2955}'
(node:905) TimeoutOverflowWarning: 5400001000 does not fit into a 32-bit signed integer.
Timeout duration was set to 1.
Zigbee2MQTT:info  2022-01-31 16:07:07: MQTT publish: topic 'zigbee2mqtt-1/0x00158d0003a27999', payload '{"battery":74,"humidity":70.88,"last_seen":"2022-01-31T16:07:07+02:00","linkquality":252,"temperature":22.01,"voltage":2955}'
(node:905) TimeoutOverflowWarning: 5400001000 does not fit into a 32-bit signed integer.
Timeout duration was set to 1.
Zigbee2MQTT:info  2022-01-31 16:07:13: MQTT publish: topic 'zigbee2mqtt-1/0x00158d0003a27999', payload '{"battery":74,"humidity":70.88,"last_seen":"2022-01-31T16:07:13+02:00","linkquality":255,"temperature":21.61,"voltage":2955}'
(node:905) TimeoutOverflowWarning: 5400001000 does not fit into a 32-bit signed integer.
Timeout duration was set to 1.
Zigbee2MQTT:info  2022-01-31 16:07:13: MQTT publish: topic 'zigbee2mqtt-1/0x00158d0003a27999', payload '{"battery":74,"humidity":78.39,"last_seen":"2022-01-31T16:07:13+02:00","linkquality":252,"temperature":21.61,"voltage":2955}'
(node:905) TimeoutOverflowWarning: 5400001000 does not fit into a 32-bit signed integer.
Timeout duration was set to 1.
Zigbee2MQTT:info  2022-01-31 16:07:22: MQTT publish: topic 'zigbee2mqtt-1/0x00158d0003a27999', payload '{"battery":74,"humidity":78.39,"last_seen":"2022-01-31T16:07:22+02:00","linkquality":255,"temperature":21.59,"voltage":2955}'
(node:905) TimeoutOverflowWarning: 5400001000 does not fit into a 32-bit signed integer.
Timeout duration was set to 1.
Zigbee2MQTT:info  2022-01-31 16:07:22: MQTT publish: topic 'zigbee2mqtt-1/0x00158d0003a27999', payload '{"battery":74,"humidity":65.62,"last_seen":"2022-01-31T16:07:22+02:00","linkquality":255,"temperature":21.59,"voltage":2955}'
(node:905) TimeoutOverflowWarning: 5400001000 does not fit into a 32-bit signed integer.
Timeout duration was set to 1.
Zigbee2MQTT:info  2022-01-31 16:07:25: MQTT publish: topic 'zigbee2mqtt-1/0x00158d0003a27999', payload '{"battery":74,"humidity":65.62,"last_seen":"2022-01-31T16:07:25+02:00","linkquality":255,"temperature":21.64,"voltage":2955}'
(node:905) TimeoutOverflowWarning: 5400001000 does not fit into a 32-bit signed integer.
Timeout duration was set to 1.
Zigbee2MQTT:info  2022-01-31 16:07:25: MQTT publish: topic 'zigbee2mqtt-1/0x00158d0003a27999', payload '{"battery":74,"humidity":56.39,"last_seen":"2022-01-31T16:07:25+02:00","linkquality":255,"temperature":21.64,"voltage":2955}'
(node:905) TimeoutOverflowWarning: 5400001000 does not fit into a 32-bit signed integer.
Timeout duration was set to 1.
Zigbee2MQTT:info  2022-01-31 16:07:29: MQTT publish: topic 'zigbee2mqtt-1/0x00158d0003a27999', payload '{"battery":74,"humidity":56.39,"last_seen":"2022-01-31T16:07:29+02:00","linkquality":255,"temperature":21.69,"voltage":2955}'
(node:905) TimeoutOverflowWarning: 5400001000 does not fit into a 32-bit signed integer.
Timeout duration was set to 1.
Zigbee2MQTT:info  2022-01-31 16:07:29: MQTT publish: topic 'zigbee2mqtt-1/0x00158d0003a27999', payload '{"battery":74,"humidity":49.37,"last_seen":"2022-01-31T16:07:29+02:00","linkquality":255,"temperature":21.69,"voltage":2955}'
(node:905) TimeoutOverflowWarning: 5400001000 does not fit into a 32-bit signed integer.
Timeout duration was set to 1.
Zigbee2MQTT:info  2022-01-31 16:07:35: MQTT publish: topic 'zigbee2mqtt-1/0x00158d0003a27999', payload '{"battery":74,"humidity":49.37,"last_seen":"2022-01-31T16:07:35+02:00","linkquality":255,"temperature":21.72,"voltage":2955}'
(node:905) TimeoutOverflowWarning: 5400001000 does not fit into a 32-bit signed integer.
Timeout duration was set to 1.
Zigbee2MQTT:info  2022-01-31 16:07:35: MQTT publish: topic 'zigbee2mqtt-1/0x00158d0003a27999', payload '{"battery":74,"humidity":41.7,"last_seen":"2022-01-31T16:07:35+02:00","linkquality":255,"temperature":21.72,"voltage":2955}'
(node:905) TimeoutOverflowWarning: 5400001000 does not fit into a 32-bit signed integer.
`

Koenkk, 2022-01-31:
What node version are you using?

khaidakin, 2022-01-31:
rock@rockpis:~$ node --version
v14.18.3
rock@rockpis:~$ npm --version
6.14.15

khaidakin, 2022-01-31:
Here is the debug result with this error
https://pastebin.com/h6dWZq8y

khaidakin, 2022-01-31:
There is no such bug in the latest release.

Koenkk, 2022-01-31:
could you provide your configuration.yaml? I'm wondering if you have set a custom timeout for the passive device timeout.

khaidakin, 2022-01-31:
> could you provide your configuration.yaml? I'm wondering if you have set a custom timeout for the passive device timeout.

`homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt-1
  server: mqtt://localhost
  user: mqtt
  password: mqtt
serial:
  port: /dev/ttyS2
  adapter: ezsp
advanced:
  pan_id: 1111
  channel: 11
  cache_state: true
  log_level: info
  log_directory: data/
  last_seen: ISO_8601_local
  baudrate: 115200
  rtscts: false
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  ikea_ota_use_test_url: false
experimental:
  new_api: true
frontend:
  port: 8081
availability:
  active:
    timeout: 10
  passive:
    timeout: 1500
device_options:
  legacy: false
devices:
  '0x00158d0003a27999':
    friendly_name: '0x00158d0003a27999'
`

Koenkk, 2022-01-31:
Could you check if it is fixed in the latest dev now? ([912360c1f0c08aaf2116c8f3d63728c44077a95d](https://github.com/Koenkk/zigbee2mqtt/commit/912360c1f0c08aaf2116c8f3d63728c44077a95d))

khaidakin, 2022-01-31:
> Could you check if it is fixed in the latest dev now? ([912360c1f0c08aaf2116c8f3d63728c44077a95d](https://github.com/Koenkk/zigbee2mqtt/commit/912360c1f0c08aaf2116c8f3d63728c44077a95d))

No, not fixed. Here's what's inside availability.ts now:

        const key = this.isActiveDevice(device) ? 'active' : 'passive';
        const value = settings.get().availability[key]?.timeout;
        return key === 'active' ? utils.minutes(value) : utils.hours(value);
    }

    private isActiveDevice(device: Device): boolean {

Koenkk, 2022-01-31:
@khaidakin then you are not on the latest dev, it should have these changes: https://github.com/Koenkk/zigbee2mqtt/commit/912360c1f0c08aaf2116c8f3d63728c44077a95d#

khaidakin, 2022-01-31:
I just updated z2m and see the bug is fixed. Thanks for your help!

closed 2022-01-31 by khaidakin