Koenkk/zigbee2mqtt #10862: Button event issue with legrand Netatmo switch

fariouche, 2022-01-17:
### What happened?

The Legrand switch is working fine most of the time, it is a 067774 model.
But sometime, especially after some time of inactivity, the first button press does nothing.
Looking at the log, I see an activity, but the action is "null".
In the wireshark log, I can see a ZCL On/Off event with off (because I pressed the left off button)
So something is not interpreted properly in z2m. Also I can see in that case a latency (2s) and the log show two events that are not recognized (and maybe this is the reason of the latency)

### What did you expect to happen?

Button should always work and with no latency (or almost none)

### How to reproduce it (minimal and precise)

Just press a button after 1min without touching it and look at the mqtt events.
Press the button twice in a row and you will see two events, one with action: null and one with action off_left

### Zigbee2MQTT version

1.22.2-dev commit: 24d9065d

### Adapter firmware version

20210708

### Adapter

zzh!

### Debug log

Here is a log of what happens after one press after 1min of inactivity:
` Device 'Interrupteur Salle d'eau' announced itself
info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Interrupteur Salle d'eau","ieee_address":"0x0004740000b93428"},"type":"device_announce"}'
debug Received Zigbee message from 'Interrupteur Salle d'eau', type 'raw', cluster 'manuSpecificUbisysDimmerSetup', data '{"data":[21,33,16,1,9,40,52,185,0,0,116,4,0,1],"type":"Buffer"}' from endpoint 1 with groupID 0
debug No converter available for '067774' with cluster 'manuSpecificUbisysDimmerSetup' and type 'raw' and data '{"data":[21,33,16,1,9,40,52,185,0,0,116,4,0,1],"type":"Buffer"}'
debug Received Zigbee message from 'Interrupteur Salle d'eau', type 'raw', cluster 'manuSpecificUbisysDimmerSetup', data '{"data":[21,33,16,2,9,40,52,185,0,0,116,4,0,2],"type":"Buffer"}' from endpoint 1 with groupID 0
debug No converter available for '067774' with cluster 'manuSpecificUbisysDimmerSetup' and type 'raw' and data '{"data":[21,33,16,2,9,40,52,185,0,0,116,4,0,2],"type":"Buffer"}'
debug Received Zigbee message from 'Interrupteur Salle d'eau', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
debug Received Zigbee message from 'Interrupteur Salle d'eau', type 'readResponse', cluster 'genPowerCfg', data '{"batteryVoltage":30}' from endpoint 1 with groupID 0
info MQTT publish: topic 'zigbee2mqtt/Interrupteur Salle d'eau', payload '{"action":null,"battery":100,"brightness":65,"device":{"applicationVersion":0,"dateCode":" \u000020210904\u0000\u0000\u0000\u0000\u0000","friendlyName":"Interrupteur Salle d'eau","hardwareVersion":3,"ieeeAddr":"0x0004740000b93428","manufacturerID":4129,"manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","model":"067774","networkAddress":6376,"powerSource":"Battery","softwareBuildID":"0006","stackVersion":66,"type":"EndDevice","zclVersion":2},"linkquality":99,"voltage":3000}'
debug Retrieving state of 'Interrupteur Salle d'eau' after reconnect
debug Received Zigbee message from 'Volet Balcon', type 'attributeReport', cluster 'closuresWindowCovering', data '{"57360":0}' from endpoint 5 with groupID 0
debug Received Zigbee message from 'Volet Balcon', type 'attributeReport', cluster 'closuresWindowCovering', data '{"57363":0}' from endpoint 5 with groupID 0
debug Saving state to file /mnt/raid/zigbee2mqtt/state.json`

Here is a log of what happens after two presses (the second one worked)
`Device 'Interrupteur Salle d'eau' announced itself
info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Interrupteur Salle d'eau","ieee_address":"0x0004740000b93428"},"type":"device_announce"}'
debug Received Zigbee message from 'Interrupteur Salle d'eau', type 'raw', cluster 'manuSpecificUbisysDimmerSetup', data '{"data":[21,33,16,1,9,40,52,185,0,0,116,4,0,1],"type":"Buffer"}' from endpoint 1 with groupID 0
debug No converter available for '067774' with cluster 'manuSpecificUbisysDimmerSetup' and type 'raw' and data '{"data":[21,33,16,1,9,40,52,185,0,0,116,4,0,1],"type":"Buffer"}'
debug Received Zigbee message from 'Interrupteur Salle d'eau', type 'raw', cluster 'manuSpecificUbisysDimmerSetup', data '{"data":[21,33,16,2,9,40,52,185,0,0,116,4,0,2],"type":"Buffer"}' from endpoint 1 with groupID 0
debug No converter available for '067774' with cluster 'manuSpecificUbisysDimmerSetup' and type 'raw' and data '{"data":[21,33,16,2,9,40,52,185,0,0,116,4,0,2],"type":"Buffer"}'
debug Received Zigbee message from 'Interrupteur Salle d'eau', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
debug Received Zigbee message from 'Interrupteur Salle d'eau', type 'readResponse', cluster 'genPowerCfg', data '{"batteryVoltage":30}' from endpoint 1 with groupID 0
info MQTT publish: topic 'zigbee2mqtt/Interrupteur Salle d'eau', payload '{"action":null,"battery":100,"brightness":65,"device":{"applicationVersion":0,"dateCode":" \u000020210904\u0000\u0000\u0000\u0000\u0000","friendlyName":"Interrupteur Salle d'eau","hardwareVersion":3,"ieeeAddr":"0x0004740000b93428","manufacturerID":4129,"manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","model":"067774","networkAddress":6376,"powerSource":"Battery","softwareBuildID":"0006","stackVersion":66,"type":"EndDevice","zclVersion":2},"linkquality":69,"voltage":3000}'
debug Received Zigbee message from 'Interrupteur Salle d'eau', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
info MQTT publish: topic 'zigbee2mqtt/Interrupteur Salle d'eau', payload '{"action":"off_left","battery":100,"brightness":65,"device":{"applicationVersion":0,"dateCode":" \u000020210904\u0000\u0000\u0000\u0000\u0000","friendlyName":"Interrupteur Salle d'eau","hardwareVersion":3,"ieeeAddr":"0x0004740000b93428","manufacturerID":4129,"manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","model":"067774","networkAddress":6376,"powerSource":"Battery","softwareBuildID":"0006","stackVersion":66,"type":"EndDevice","zclVersion":2},"linkquality":72,"voltage":3000}'
info MQTT publish: topic 'zigbee2mqtt/Interrupteur Salle d'eau', payload '{"action":"","battery":100,"brightness":65,"device":{"applicationVersion":0,"dateCode":" \u000020210904\u0000\u0000\u0000\u0000\u0000","friendlyName":"Interrupteur Salle d'eau","hardwareVersion":3,"ieeeAddr":"0x0004740000b93428","manufacturerID":4129,"manufacturerName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","model":"067774","networkAddress":6376,"powerSource":"Battery","softwareBuildID":"0006","stackVersion":66,"type":"EndDevice","zclVersion":2},"linkquality":72,"voltage":3000}'
info MQTT publish: topic 'zigbee2mqtt/Interrupteur Salle d'eau/action', payload 'off_left'
debug Received MQTT message on 'zigbee2mqtt/Lumiere Salle d'eau/set' with data '{"state":"OFF"}'
debug Publishing 'set' 'state' to 'Lumiere Salle d'eau'
info MQTT publish: topic 'zigbee2mqtt/lumieres salle d'eau', payload '{"brightness":226,"color":{"h":210,"hue":210,"s":8,"saturation":8,"x":0.3143,"y":0.3244},"color_mode":"color_temp","color_temp":155,"state":"OFF"}'
info MQTT publish: topic 'zigbee2mqtt/Lumiere Salle d'eau', payload '{"brightness":226,"color":{"h":210,"hue":210,"s":8,"saturation":8,"x":0.3143,"y":0.3244},"color_mode":"color_temp","color_temp":155,"device":{"applicationVersion":1,"dateCode":"20131206","friendlyName":"Lumiere Salle d'eau","hardwareVersion":1,"ieeeAddr":"0x00124b00226fcfa5","manufacturerID":0,"manufacturerName":"GLEDOPTO","model":"GL-C-008-1ID","networkAddress":6070,"powerSource":"Mains (single phase)","softwareBuildID":"3.0.1","stackVersion":2,"type":"Router","zclVersion":1},"linkquality":87,"state":"OFF"}'
debug Retrieving state of 'Interrupteur Salle d'eau' after reconnect
debug Saving state to file /mnt/raid/zigbee2mqtt/state.json`

rdorys, 2022-01-18:
Hi, I have the same issue with 067773. This issue doesn't occurs when the device has been bound to a group (onoff cluster)... Strange behavior... But still have the error as soon as I unbound it...
`No converter available for '067773' with cluster 'manuSpecificUbisysDimmerSetup' and type 'raw' and data '{"data":[21,33,16,0,5,128,163,195,0,0,116,4,0],"type":"Buffer"}'`

Can you try to bind it to a dummy group to see if it fix it ?

For the latency, legrand switches don't seem that good...Compared to ikea remote which are responding instantly... I don't know if it is a firmware issue or something due to Z2M. 

I noticed ikea remote never announce themselves.  I don't know if it can explain anything.

fariouche, 2022-01-19:
Yes, it works when binded, but unfortunately I cannot use it as I'm using geledpto module that doesn't behave well when dimming it directly. I need to use domoticz to workaround that.

Also binding works here because it bypass z2m. But there is still an issue in z2m about some received frames... I didn't had the time to look at what z2m receive from the dongle to understand why

rdorys, 2022-01-20:
You don't need to bind it to the device you want to interact with.

Mine is bound to a 'test' group which contains no device, and it triggers an automation in home assistant.

The point is you won't have that glitch from the moment it is bound to anything.

Does it help?

fariouche, 2022-01-20:
Oooh! I see. Will try. If it works it will be a simple workaround lol

fariouche, 2022-01-20:
ok, so I've done the test: Created a dummy group with no device inside, bind the switch to the dummy group... and still the exact same issue :(
Your setup does not get the switch actions from the group, right? (but directly from the switch itself?)

rdorys, 2022-01-20:
This is my configuration for the device.
[image]

Is that yours ? It's also bound to coordinator on OnOff and LevelCtrl clusters... you too ?

I get the action directly from the switch.

fariouche, 2022-01-20:
Yes, same except that my switch has two buttons so multiply by 2 every lines but same logic :(

fariouche, 2022-01-20:
@Koenkk , is there an option to see what zigbee-herdsman receives from the dongle exactly? I would like to see if this issue is from the firmware of the dongle or from herdsman driver itself as I can see the off command with the sniffer (so I know it's not the legrand switch)

Koenkk, 2022-01-21:
@fariouche you can run z2m with herdsman debug logging enabled: https://www.zigbee2mqtt.io/guide/usage/debug.html#enabling-logging , everything that starts with `zigbee-herdsman:adapter:zStack:adapter` comes from the adpater

fariouche, 2022-01-21:
For info, I tried to bind the switch directly to the lights.... it worked, for some time then it crashed my lamps (not responding anymore)
Tried to unbind, no response. I had to remove it from z2m, reset the switch , power cycle the lamps and re-add the switch.
So, I advise to never directly bind Legrand switches... buggy as hell. A shame.

I will try to see what z2m receives just to learn how to debug, but not sure I want to rely on this switch. Legrand Netatmo = ban for me :)

I wish OTA was available easily

fariouche, 2022-01-22:
Ok, here is the debug log when it does not work... we can see that it received the off command, and then issue an mqtt null action
```
zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,144,248,40,52,185,0,0,116,4,0,72,190,200] +51s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,144,248,40,52,185,0,0,116,4,0,72,190,200] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [144,248,40,52,185,0,0,116,4,0,72,190] - 200 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":63632,"extaddr":"0x0004740000b93428","parentaddr":48712} +51s
  zigbee-herdsman:controller:log Device '0x0004740000b93428' joined +51s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:controller:log Device '0x0004740000b93428' accepted by handler +0ms
  zigbee-herdsman:controller:log Not interviewing '0x0004740000b93428', completed 'true', in progress 'false' +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,193,144,248,144,248,40,52,185,0,0,116,4,0,128,220] +165ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,193,144,248,144,248,40,52,185,0,0,116,4,0,128,220] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 193 - [144,248,144,248,40,52,185,0,0,116,4,0,128] - 220 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":63632,"nwkaddr":63632,"ieeeaddr":"0x0004740000b93428","capabilities":128} +166ms
  zigbee-herdsman:controller:log Device announce '0x0004740000b93428' +165ms
Zigbee2MQTT:debug 2022-01-22 19:14:19: Device 'Interrupteur Salle d'eau' announced itself
Zigbee2MQTT:info  2022-01-22 19:14:20: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Interrupteur Salle d'eau","ieee_address":"0x0004740000b93428"},"type":"device_announce"}'
  zigbee-herdsman:controller:endpoint Read 0x0004740000b93428/1 genPowerCfg(["batteryVoltage"], {"sendWhenActive":true,"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4129,"transactionSequenceNumber":null,"writeUndiv":false}) +52s
  zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x0004740000b93428:63632/1 (0,0,1) +52s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":63632,"destendpoint":1,"srcendpoint":1,"clusterid":1,"transid":3,"options":0,"radius":30,"len":7,"data":{"type":"Buffer","data":[20,33,16,4,0,32,0]}} +52s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,17,36,1,144,248,1,1,1,0,3,0,30,7,20,33,16,4,0,32,0,70] +52s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +161ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,144,248,1,72,190,27,254,5,69,196,144,248,1,72,190,27,254,34,68,129,0,0,1,252,144,248,1,1,0,105,0,98,39,228,0,0,14,21,33,16,1,9,40,52,185,0,0,116,4,0,1,72,190,18,168,254,34,68,129,0,0,1,252,144,248,1,1,0,105,0,98,43,228,0,0,14,21,33,16,2,9,40,52,185,0,0,116,4,0,2,72,190,18,164] +4ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,144,248,1,72,190,27,254,5,69,196,144,248,1,72,190,27,254,34,68,129,0,0,1,252,144,248,1,1,0,105,0,98,39,228,0,0,14,21,33,16,1,9,40,52,185,0,0,116,4,0,1,72,190,18,168,254,34,68,129,0,0,1,252,144,248,1,1,0,105,0,98,43,228,0,0,14,21,33,16,2,9,40,52,185,0,0,116,4,0,2,72,190,18,164] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [144,248,1,72,190] - 27 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":63632,"relaycount":1,"relaylist":[48712]} +165ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,144,248,1,72,190,27,254,34,68,129,0,0,1,252,144,248,1,1,0,105,0,98,39,228,0,0,14,21,33,16,1,9,40,52,185,0,0,116,4,0,1,72,190,18,168,254,34,68,129,0,0,1,252,144,248,1,1,0,105,0,98,43,228,0,0,14,21,33,16,2,9,40,52,185,0,0,116,4,0,2,72,190,18,164] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [144,248,1,72,190] - 27 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":63632,"relaycount":1,"relaylist":[48712]} +3ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,34,68,129,0,0,1,252,144,248,1,1,0,105,0,98,39,228,0,0,14,21,33,16,1,9,40,52,185,0,0,116,4,0,1,72,190,18,168,254,34,68,129,0,0,1,252,144,248,1,1,0,105,0,98,43,228,0,0,14,21,33,16,2,9,40,52,185,0,0,116,4,0,2,72,190,18,164] +3ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 34 - 2 - 4 - 129 - [0,0,1,252,144,248,1,1,0,105,0,98,39,228,0,0,14,21,33,16,1,9,40,52,185,0,0,116,4,0,1,72,190,18] - 168 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":64513,"srcaddr":63632,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":105,"securityuse":0,"timestamp":14952290,"transseqnumber":0,"len":14,"data":{"type":"Buffer","data":[21,33,16,1,9,40,52,185,0,0,116,4,0,1]}} +1ms
  zigbee-herdsman:controller:log Received 'raw' data '{"clusterID":64513,"data":{"type":"Buffer","data":[21,33,16,1,9,40,52,185,0,0,116,4,0,1]},"address":63632,"endpoint":1,"linkquality":105,"groupID":0,"wasBroadcast":false,"destinationEndpoint":1}' +170ms
Zigbee2MQTT:debug 2022-01-22 19:14:20: Received Zigbee message from 'Interrupteur Salle d'eau', type 'raw', cluster 'manuSpecificUbisysDimmerSetup', data '{"data":[21,33,16,1,9,40,52,185,0,0,116,4,0,1],"type" [... cut ...] \u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","model":"067774","networkAddress":63632,"powerSource":"Battery","softwareBuildID":"0006","stackVersion":66,"type":"EndDevice","zclVersion":2},"linkquality":105,"voltage":3000}'
Zigbee2MQTT:info  2022-01-22 19:14:43: MQTT publish: topic 'zigbee2mqtt/Interrupteur Salle d'eau/action', payload 'off_right'
Zigbee2MQTT:debug 2022-01-22 19:14:45: Retrieving state of 'Interrupteur Salle d'eau' after reconnect
```

Koenkk, 2022-01-23:
Could you check if this issue has been fixed with the following external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const command_off = {
    cluster: 'genOnOff',
    type: 'commandOff',
    convert: (model, msg, publish, options, meta) => {
        if (hasAlreadyProcessedMessage(msg)) return;
        const payload = {action: postfixWithEndpointName('off', msg, model)};
        addActionGroup(payload, msg, model);
        return payload;
    },
}

const definition = {
    zigbeeModel: [' Double gangs remote switch\u0000\u0000\u0000\u0000'],
    model: '067774',
    vendor: 'Legrand',
    description: 'Wireless double remote switch CUSTOM',
    fromZigbee: [fz.identify, fz.command_on, command_off, fz.command_toggle, fz.command_move, fz.command_stop, fz.battery],
    exposes: [e.battery(),
        e.action(['identify', 'on', 'off', 'toggle', 'brightness_move_up', 'brightness_move_down', 'brightness_stop'])],
    toZigbee: [],
    meta: {multiEndpoint: true, battery: {voltageToPercentage: '3V_2500'}},
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'genOnOff', 'genLevelCtrl']);
        const endpoint2 = device.getEndpoint(2);
        await reporting.bind(endpoint2, coordinatorEndpoint, ['genPowerCfg', 'genOnOff', 'genLevelCtrl']);
    },
    endpoint: (device) => {
        return {left: 1, right: 2};
    },
};

module.exports = definition;
```

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, go to the frontend, verify that the description of the device ends with CUSTOM (this means that external converter has been loaded correctly)

rdorys, 2022-01-23:
hi @Koenkk  I did the same for the 067773 model with your external converter (adapted for 067773) :
I had to add utils or hasAlreadyProcessedMessage, postfixWithEndpointName and addActionGroup wouldn't be recognized (and I am a total noob in node.js).

I don't know for the 067774 but the 067773 only generate toggle command on single clic (and only "off" command when double-clic, which is not handy at all when using binding.) So I modified the command_toggle function instead off command_off.

```
const fz = {... require('zigbee-herdsman-converters/converters/fromZigbee') , legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const utils = require('zigbee-herdsman-converters/lib/utils');

const command_toggle = {
    cluster: 'genOnOff',
    type: 'commandToggle',
    convert: (model, msg, publish, options, meta) => {
        if (utils.hasAlreadyProcessedMessage(msg)) return;
        const payload = {action: utils.postfixWithEndpointName('toggle', msg, model)};
        utils.addActionGroup(payload, msg, model);
        return payload;
    },
}

const definition = {zigbeeModel: [
            ' Remote switch\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000'],
        model: '067773',
        vendor: 'Legrand',
        // led blink RED when battery is low
        description: 'Wireless remote switch custom 3',
        fromZigbee: [fz.identify, fz.command_on, fz.command_off, command_toggle, fz.legacy.cmd_move, fz.legacy.cmd_stop, fz.battery],
        exposes: [e.battery(), e.action(['identify', 'on', 'off', 'toggle', 'brightness_move_up',
            'brightness_move_down', 'brightness_stop'])],
        toZigbee: [],
        meta: {battery: {voltageToPercentage: '3V_2500'}},
        configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg', 'genOnOff', 'genLevelCtrl']);}
            
};

module.exports = definition;

```

But I still got the weird message after announcement :
```
Device '0x0004740000c3a380' announced itself
Received Zigbee message from '0x0004740000c3a380', type 'raw', cluster 'manuSpecificUbisysDimmerSetup', data '{"data":[21,33,16,0,5,128,163,195,0,0,116,4,0],"type":"Buffer"}' from endpoint 1 with groupID 0
Received Zigbee message from '0x0004740000c3a380', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
Retrieving state of '0x0004740000c3a380' after reconnect
```
Any Idea ?
It did'nt work either with a "command_off" version I also did.

Here after my  zigbee-herdsman logs : 

```
2022-01-23T15:32:51.147Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,13,69,193,161,71,161,71,128,163,195,0,0,116,4,0,128,153]

2022-01-23T15:32:51.147Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,13,69,193,161,71,161,71,128,163,195,0,0,116,4,0,128,153]

2022-01-23T15:32:51.148Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 13 - 2 - 5 - 193 - [161,71,161,71,128,163,195,0,0,116,4,0,128] - 153

2022-01-23T15:32:51.148Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - endDeviceAnnceInd - {"srcaddr":18337,"nwkaddr":18337,"ieeeaddr":"0x0004740000c3a380","capabilities":128}

2022-01-23T15:32:51.151Z zigbee-herdsman:adapter:zStack:adapter Discovering route to 18337
[... 51 lines cut ...]
2022-01-23T15:32:51.414Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 5 - 69 - [0] - 33

2022-01-23T15:32:51.414Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- ZDO - extRouteDisc - {"status":0}

2022-01-23T15:32:51.414Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []

2022-01-23T15:32:51.617Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,12,69,202,161,71,128,163,195,0,0,116,4,0,0,0,245]

2022-01-23T15:32:51.617Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,12,69,202,161,71,128,163,195,0,0,116,4,0,0,0,245]

2022-01-23T15:32:51.617Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 12 - 2 - 5 - 202 - [161,71,128,163,195,0,0,116,4,0,0,0] - 245

2022-01-23T15:32:51.617Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":18337,"extaddr":"0x0004740000c3a380","parentaddr":0}

2022-01-23T15:32:51.618Z zigbee-herdsman:controller:log Device '0x0004740000c3a380' joined

2022-01-23T15:32:51.619Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []

2022-01-23T15:32:51.619Z zigbee-herdsman:controller:log Device '0x0004740000c3a380' accepted by handler

2022-01-23T15:32:51.619Z zigbee-herdsman:controller:log Not interviewing '0x0004740000c3a380', completed 'true', in progress 'false'

Zigbee2MQTT:debug 2022-01-23 15:32:53: Retrieving state of '0x0004740000c3a380' after reconnect

```

fariouche, 2022-01-23:
@rdorys , Thanks for the fix... I was testing when I got this error and was about to fix it too ;)

From my point of view (067774), the external converter fixes the problem I initially reported! thanks @Koenkk !
In my case, one click = one command (no need to double click anymore to have the off action)

Koenkk, 2022-01-23:
Added the fix to the dev branch now, could you check wether the issue is fixed here? (without having any external converter)

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

fariouche, 2022-01-24:
yes, sure!
Before that I need the #10019 fix (missing tzLocal.lift_duration for NHPB/SHUTTER) and if possible the fix i proposed for casaia thermometer to handle negative temperatures (#9943).
That way I can remove 3 external converters and to one test :)
Thank you!

Crocmagnon, 2022-01-24:
@Koenkk I'm using the stable HA addon.
After reading https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html#home-assistant-addon and https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/tree/master/zigbee2mqtt-edge, here's what I understand about using the edge version:

1. stop the stable HA addon
2. Copy its config
3. Install the edge HA addon
4. Use the previously copied config

What about rolling back to the stable version? Is it as easy as:
1. Stop (and uninstall) the edge addon
2. Start the stable addon

rdorys, 2022-01-24:
Hi, yes this is the right way to do it.

It doesn't fix the issue for 067773 with 002a / 20210218 firmware. Hope it will fix it for your version.

Crocmagnon, 2022-01-24:
Looks like I have an older firmware: 001e / 20191030.
I'll try the edge addon then :)

rdorys, 2022-01-24:
> Looks like I have an older firmware: 001e / 20191030.
> I'll try the edge addon then :)

Please, never update ! 😄

Crocmagnon, 2022-01-24:
> Please, never update ! 😄

Haha I don't intend to now that I have seen this issue 😁 

> Hope it will fix it for your version.

@Koenkk After a couple of initial tests, it looks like the issue is fixed on the edge version. I'll leave it running for a few more hours in order to get a better picture but it's promising! Thanks 🙏🏻

rdorys, 2022-01-24:
@Koenkk switching to dev branch seems to also actually fix it for my fw version (I guess my external converter custom version was pretty bad)!
There is still the :

No converter available for '067773' with cluster 'manuSpecificUbisysDimmerSetup' and type 'raw' and data '{"data":[21,33,16,0,5,128,163,195,0,0,116,4,0],"type":"Buffer"}'

in debug log bug the issue is fixed as we don't need to wake it up before being detected in home assistant.

I will test it for few days and keep you in touch. Thx a lot  !

Koenkk, 2022-01-24:
So if I understand correctly, both `067773` and `067774` work as expected now in the latest-dev?

rdorys, 2022-01-24:
@Koenkk  Yeah, I assume it works on 067773. @Crocmagnon , is it ok for you ?  If eventually it failed again I will re-open an issue.

Crocmagnon, 2022-01-24:
Fine with me 👍🏻 @Koenkk, do you have an idea when the next release will happen? ("no" is an acceptable answer 😉 )

rdorys, 2022-01-24:
@Crocmagnon 

https://github.com/Koenkk/zigbee2mqtt/releases/

Every months on 1st day :)

Koenkk, 2022-01-25:
Next release is on 1 February indeed 🎉

closed 2022-01-25 by Koenkk

mkapuscinski, 2022-05-18:
Hi, 

I am having couple of those netatmo switches with similar issues, and I am running 1.25.1-1 addon version

Loic691, 2022-12-07:
Hi all
I have the issue to
Since a last version of Z2M, I don't wich exactly, I have a big latence when using 067774
The zigbee message action was received several seconds (0, 10, 20, 60, 120,...) after pushed the button.
I have no issue with the roller shutter wireless switch.
Do you have this issue with latest version of Z2M  and the 0677774 ?