Koenkk/zigbee2mqtt #10864: ikea control outlet toggles automaticaly

Thuurke, 2022-01-17:
### What happened?

I have 3 ikea control outlets and 3 ikea shortcut buttons. I tried several combinations but problem stays.
I use a conbee 2 stick and previously in deconz this situation worked.

Issue: Without pressing one of the buttons the outlet toggles unexpected.
The logging shows first and second the normal button trigger behaviour. The third section is of unexpected toggle.

### What did you expect to happen?

pressing one of the buttons should toggle the outlet. 

### How to reproduce it (minimal and precise)

I don't now. 

### Zigbee2MQTT version

1.22.2 commit: 414c51f

### Adapter firmware version

0x26580700

### Adapter

ConBee2

### Debug log

```
Received Zigbee message from 'IKEA Shortcut Button 03', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/IKEA Shortcut Button 03', payload '{"action":"on","battery":60,"linkquality":159,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/IKEA Shortcut Button 03', payload '{"action":"","battery":60,"linkquality":159,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/IKEA Shortcut Button 03/action', payload 'on'
Debug Received MQTT message on 'zigbee2mqtt/IKEA Control Outlet 03/set' with data 'OFF'
Debug Publishing 'set' 'state' to 'IKEA Control Outlet 03'
Info MQTT publish: topic 'zigbee2mqtt/IKEA Control Outlet 03', payload '{"linkquality":151,"power_on_behavior":null,"state":"OFF","update":{"state":"idle"},"update_available":false}'
Debug Received Zigbee message from 'IKEA Control Outlet 03', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/IKEA Control Outlet 03', payload '{"linkquality":159,"power_on_behavior":null,"state":"OFF","update":{"state":"idle"},"update_available":false}'

Received Zigbee message from 'IKEA Shortcut Button 03', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/IKEA Shortcut Button 03', payload '{"action":"on","battery":60,"linkquality":159,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/IKEA Shortcut Button 03', payload '{"action":"","battery":60,"linkquality":159,"update":{"state":null},"update_available":null}'
Info MQTT publish: topic 'zigbee2mqtt/IKEA Shortcut Button 03/action', payload 'on'
Debug Received MQTT message on 'zigbee2mqtt/IKEA Control Outlet 03/set' with data 'ON'
Debug Publishing 'set' 'state' to 'IKEA Control Outlet 03'
Info MQTT publish: topic 'zigbee2mqtt/IKEA Control Outlet 03', payload '{"linkquality":143,"power_on_behavior":null,"state":"ON","update":{"state":"idle"},"update_available":false}'
Debug Received Zigbee message from 'IKEA Control Outlet 03', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/IKEA Control Outlet 03', payload '{"linkquality":127,"power_on_behavior":null,"state":"ON","update":{"state":"idle"},"update_available":false}'

Debug Received Zigbee message from 'IKEA Shortcut Button 03', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":74}' from endpoint 1 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/IKEA Shortcut Button 03', payload '{"action":null,"battery":74,"linkquality":167,"update":{"state":null},"update_available":null}'
Debug Received MQTT message on 'zigbee2mqtt/IKEA Control Outlet 03/set' with data 'OFF'
Debug Publishing 'set' 'state' to 'IKEA Control Outlet 03'
Info MQTT publish: topic 'zigbee2mqtt/IKEA Control Outlet 03', payload '{"linkquality":183,"power_on_behavior":null,"state":"OFF","update":{"state":"idle"},"update_available":false}'
Debug Received Zigbee message from 'IKEA Control Outlet 03', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null
Info MQTT publish: topic 'zigbee2mqtt/IKEA Control Outlet 03', payload '{"linkquality":183,"power_on_behavior":null,"state":"OFF","update":{"state":"idle"},"update_available":false}'

Koenkk, 2022-01-18:
As can be seen in the log, and MQTT message is send to turn of the device: `Debug Received MQTT message on 'zigbee2mqtt/IKEA Control Outlet 03/set' with data 'OFF'`. The issue is in your automations, not on the z2m side.

closed 2022-01-18 by Koenkk

Thuurke, 2022-01-18:
The automation I have is this one.      
```
- alias: shortcut_lamp_toggle
  initial_state: true
  trigger:
  - platform: state
    entity_id: sensor.ikea_shortcut_button_03_action
  action:
    - service: switch.toggle
      entity_id: switch.ikea_control_outlet_03
```
I don't have anything with a time and the spooky toggle is random.

It just toggled again. It seems to occur after a "attributeReport"?

```
debug 2022-01-18 12:13:47: Received Zigbee message from 'IKEA Shortcut Button 03', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":87}' from endpoint 1 with groupID null
info  2022-01-18 12:13:47: MQTT publish: topic 'zigbee2mqtt/IKEA Shortcut Button 03', payload '{"action":null,"battery":87,"linkquality":167,"update":{"state":null},"update_available":null}'
debug 2022-01-18 12:13:47: Received MQTT message on 'zigbee2mqtt/IKEA Control Outlet 03/set' with data 'ON'
debug 2022-01-18 12:13:47: Publishing 'set' 'state' to 'IKEA Control Outlet 03'
```

Koenkk, 2022-01-18:
The toggle is not random, this automation will be triggered every time something is published for the device (in this case the battery value), instead it should only trigger when the action != null. 

See https://www.zigbee2mqtt.io/guide/usage/integrations/home_assistant.html#responding-to-button-clicks how to fix this

Thuurke, 2022-01-18:
Thank you missed that in the documentation.