Koenkk/zigbee2mqtt #10871: Xiaomi MCCGQ11LM reports incorrect payload

jsherman256, 2022-01-18:
### What happened?

The payloads from my Xiaomi MCCGQ11LM contact sensor are incorrect.

When the contacts are slid apart, these messages are seen:

```
Zigbee2MQTT:info  2022-01-18 03:56:08: MQTT publish: topic 'zigbee2mqtt/window-a', payload '{"battery":100,"contact":false,"linkquality":93,"temperature":23,"voltage":3135}'
Zigbee2MQTT:info  2022-01-18 03:56:08: MQTT publish: topic 'zigbee2mqtt/window-a', payload '{"battery":100,"contact":true,"linkquality":93,"temperature":23,"voltage":3135}'
```

When the contacts are slid back together, these messages are seen:

```
Zigbee2MQTT:info  2022-01-18 03:56:12: MQTT publish: topic 'zigbee2mqtt/window-a', payload '{"battery":100,"contact":false,"linkquality":123,"temperature":23,"voltage":3135}'
Zigbee2MQTT:info  2022-01-18 03:56:12: MQTT publish: topic 'zigbee2mqtt/window-a', payload '{"battery":100,"contact":true,"linkquality":123,"temperature":23,"voltage":3135}'
Zigbee2MQTT:info  2022-01-18 03:56:12: MQTT publish: topic 'zigbee2mqtt/window-a', payload '{"battery":100,"contact":false,"linkquality":123,"temperature":23,"voltage":3135}'
Zigbee2MQTT:info  2022-01-18 03:56:12: MQTT publish: topic 'zigbee2mqtt/window-a', payload '{"battery":100,"contact":true,"linkquality":123,"temperature":23,"voltage":3135}'
```

Note that in both cases, the final message contains `contact: true`, so debounce will not fix this. There appear to be 2 messages for "contact opened" and 4 messages for "contact closed", so perhaps the encoding Xiaomi uses has changed

### How to reproduce it (minimal and precise)

1. Pair Xiaomi MCCGQ11LM
2. Slide contacts apart
3. Slide contacts back together

zigbee2mqtt reports the Zigbee model of the sensor as `lumi.sensor_magnet.aq2` with sensor firmware date and version `20161128` and `3000-0001` respectively

No other issues are observed with zigbee2mqtt or any other sensors. This issue is seen both on the `latest` and `latest-dev` tags of the zigbee2mqtt docker image

### Zigbee2MQTT version

1.22.1

### Adapter firmware version

20210708

### Adapter

zzh (CC2652R1)

### Debug log

[MCCGQ11LM-herdsman-debug.log](https://github.com/Koenkk/zigbee2mqtt/files/7888898/MCCGQ11LM-herdsman-debug.log)

Koenkk, 2022-01-18:
Looks like https://github.com/Koenkk/zigbee2mqtt/issues/8663

jsherman256, 2022-01-18:
> Looks like #8663

I wouldn't not have guessed that, as all other zigbee devices have been working fine, but you're absolutely right. Upgraded the Ubuntu kernel from 5.4.0-89 to 5.4.0-94 and the sensor is working. Thanks!!

closed 2022-01-18 by jsherman256