Koenkk/zigbee2mqtt #10888: IKEA TRADFRI bulb switches on every 4 hours

Veldkornet, 2022-01-19:
### What happened?

I have two IKEA TRADFRI bulb E27 WS opal 1000lm bulbs within 1m of each other. The further away of the two (from the controller) seems to spontaneously switch on every 4 hours. I've even created an automation that will turn the light off if it gets turned on (without the door being open).

- I can say that the lights have not change position since installation.
- The issue does not occur in ZHA or deconz (I don't know what event occurs exactly every 4 hours in zigbee2mqtt)
- If I remove power from the bulb and add it back, it will sometime stop the issue (sometimes)
- Similarly, if I tell zigbee2mqtt to reconfigure the device, it will correct the issue (sometimes)
- However, some time later (could be days for example) it will start up again

### What did you expect to happen?

That the light does not turn on randomly every 4 hours out of it's own

### How to reproduce it (minimal and precise)

Honestly, I wish I knew. I don't see anything specific in the logs as yet.

Will update with the debug log when I can get it...

### Zigbee2MQTT version

1.22.2-1

### Adapter firmware version

 0x26720700

### Adapter

ConBee2

Koenkk, 2022-01-19:
Please provide the debug log at the moment the light turns on

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

Veldkornet, 2022-01-19:
I've updated the post with the debug.

It randomly turned on at 20:24:42.

I see at that time it received an MQTT message... but.... ???

Veldkornet, 2022-01-19:
I noticed that via a MQTT explorer, that the messages of the device were set to retain; where as for the other device (the working one) they were not. Even though on the settings it was not set to retain. 

So I've deleted the retained messages now. 

Would that have had any effe t on it?

Koenkk, 2022-01-19:
If the retained message was to `zigbee2mqtt/Schuurlamp/set`, yes. Note that if this happens after seeing `Received MQTT message on 'zigbee2mqtt/Schuurlamp/set' with data '{"state": "ON"}'` in your log it isn't a z2m issue. z2m receives a request to turn on the device

Veldkornet, 2022-01-28:
I think that was probably the issue. At least, it hasn't happened since I discovered and deleted the mqtt message.

Strange! :)

closed 2022-01-28 by Veldkornet