Koenkk/zigbee2mqtt #10889: BlitzWolf BW-SHP13 - After re-pairing power, voltage and current aren't exposed

nohn, 2022-01-19:
### What happened?

I removed and rejoined a BlitzWolf BW-SHP13. After rejoin, the plug that worked before doesn't report power, voltage and current (but does report energy).

Action starts in log @ 2022-01-19T13:59:06.806Z

### What did you expect to happen?

After rejoin, the plug reports power, voltage, current and energy.

### How to reproduce it (minimal and precise)

Remote a BlitzWolf BW-SHP13 from the network and re-join it.

### Zigbee2MQTT version

1.22.2 

### Adapter firmware version

20210708

### Adapter

Slaesh's CC2652RB Stick

### Debug log

[0x84ba20fffe77b372.log](https://github.com/Koenkk/zigbee2mqtt/files/7898336/0x84ba20fffe77b372.log)

nohn, 2022-01-19:
Update, I got it to work:

1. Remove the device via GUI
2. Stop Zigbee2MQTT
3. When trying to remove the entry from database.db and state.json files, nothing about this device was left in the files.
4. Start Zigbee2MQTT
5. Re-Join the device

So I would guess, something is cached in working memory that prevents this device from re-joining with all it's properties corretly after removing. A restart of zigbee2mqtt between remove and re-join seems to clear that cache.

jrmax, 2022-01-21:
I just had the same issue and can confirm that the workaround works 👍
Thx @nohn

aIexus, 2022-01-23:
I have 3 such devices. And only one has the same problem: current, power and voltage are null.
Issue appeared after flashing firmware on stick to 20211217 revision.

Zigbee2MQTT version: 1.22.2
Coordinator type: zStack3x0
Coordinator revision: 20211217
Frontend version: 0.6.67

{
    "energy": 17.97,
    "last_seen": "2022-01-23T15:40:59.866Z",
    "linkquality": 61,
    "power_outage_memory": "off",
    "state": "OFF",
    "current": null,
    "power": null,
    "voltage": null
}

After workaraound (without step 3! ) - issue is dissappeared:

{
    "current": 0,
    "energy": 17.97,
    "last_seen": "2022-01-23T15:48:17.038Z",
    "linkquality": 112,
    "power": 0,
    "power_outage_memory": "off",
    "state": "OFF",
    "voltage": 197
}

closed 2022-03-03 by github-actions[bot]