Koenkk/zigbee2mqtt #10894: Sonoff Zigbee 3.0 USB Dongle Plus: have to unscrew antenna to start NPM and show devices in frontend

gvt70, 2022-01-19:
### What happened?

New install of Zigbee2mqtt on Pi4

Using Sonoff Zigbee 3.0 USB Dongle Plus with CC2652 P1F
Update to latest firmware version CC1352P2_CC2652P_launchpad_coordinator_20211217

Problems:
- I need to unscrew the antenna in order to be able to start NPM 
- I need to unscrew the antenna in order to discover devices in the frontend.
Sometimes devices are recognized under the correct brand and type. Sometimes as 'unsupported'

I have read this problem somewhere but can't find it anymore. 
Hopefully someone knows the solution

### What did you expect to happen?

correct use with antenna

### How to reproduce it (minimal and precise)

use adapter with or without antenna

### Zigbee2MQTT version

1.22.2 commit: 1a0a9a6e

### Adapter firmware version

20211217

### Adapter

Sonoff Zigbee 3.0 USB Dongle Plus with CC2652 P1F

Tonio16, 2022-01-19:
Hello

 I have the same adaptor on the same z2m. No problem at all. Did you check without the case of the key?

Antoine

gvt70, 2022-01-19:
Yes, Makes no difference

> Hello
> 
> I have the same adaptor on the same z2m. No problem at all. Did you check without the case of the key?
> 
> Antoine

Koenkk, 2022-01-19:
What is logged on z2m startup when starting with antenna? If it logs `Coordinator failed to start, probably the panID is already in use, try a different panID or channel` see https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html#error-coordinator-failed-to-start-probably-the-panid-is-already-in-use-try-a-different-panid-or-channel

closed 2022-02-26 by github-actions[bot]