Koenkk/zigbee2mqtt #10903: [New device support]: Door sensor I think TS0203 _TYZB01_xph99wvr

jocamane, 2022-01-19:
### Link

I don't know

### Database entry

info  2022-01-19 20:19:44: Zigbee: allowing new devices to join. info  2022-01-19 20:19:44: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"time":254,"value":true},"status":"ok","transaction":"wh03c-38"}' info  2022-01-19 20:20:19: Device '0x00124b001b2f46c3' joined info  2022-01-19 20:20:19: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b001b2f46c3","ieee_address":"0x00124b001b2f46c3"},"type":"device_joined"}' info  2022-01-19 20:20:19: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00124b001b2f46c3"},"type":"device_connected"}' info  2022-01-19 20:20:19: Starting interview of '0x00124b001b2f46c3' info  2022-01-19 20:20:19: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b001b2f46c3","ieee_address":"0x00124b001b2f46c3","status":"started"},"type":"device_interview"}' info  2022-01-19 20:20:19: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00124b001b2f46c3"},"type":"pairing"}' debug 2022-01-19 20:20:19: Received Zigbee message from '0x00124b001b2f46c3', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryVoltage":30}' from endpoint 1 with groupID null debug 2022-01-19 20:20:19: Skipping message, definition is undefined and still interviewing debug 2022-01-19 20:20:19: Received Zigbee message from '0x00124b001b2f46c3', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200}' from endpoint 1 with groupID null debug 2022-01-19 20:20:19: Skipping message, definition is undefined and still interviewing debug 2022-01-19 20:20:19: Received Zigbee message from '0x00124b001b2f46c3', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryVoltage":30}' from endpoint 1 with groupID null debug 2022-01-19 20:20:19: Skipping message, definition is undefined and still interviewing debug 2022-01-19 20:20:21: Saving state to file /config/zigbee2mqtt/state.json error 2022-01-19 20:20:48: Failed to interview '0x00124b001b2f46c3', device has not successfully been paired info  2022-01-19 20:20:48: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b001b2f46c3","ieee_address":"0x00124b001b2f46c3","status":"failed"},"type":"device_interview"}' info  2022-01-19 20:20:48: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x00124b001b2f46c3"},"type":"pairing"}' debug 2022-01-19 20:21:13: Received Zigbee message from 'Tomada 3 ZG', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID null info  2022-01-19 20:21:13: MQTT publish: topic 'zigbee2mqtt/Tomada 3 ZG', payload '{"linkquality":255,"state":"OFF"}'

### Comments

I tried in edge version and prodution version

Koenkk, 2022-01-26:
what z2m version are you using? Make sure you are on the latest z2m (1.22.2) and follow https://www.zigbee2mqtt.io/guide/faq/#interview-fails

jocamane, 2022-01-26:
thanks for your response.

The version of z2m is the last. (1.22.2)
in edge version is the same.

Try pairing the device closer to the coordinator.
**DONE. SAME**
There can be too much interference, try connecting the coordinator USB through an USB extension cable. This problem occurs a lot when used in combination with a Raspberry Pi 4.
**DONE. SAME**
If it’s a battery powered device, try replacing the battery. Try to keep the device awake by pressing the button of the device (if any) every 3 seconds.
**DONE. SAME**
Try repairing the device again for 2 or 3 times.
**DONE. 5 AND 6 TIMES. SAME**
This might be a Zigbee2MQTT bug, Create a new issue with the zigbee-herdsman debug logging attached to it. How to enable zigbee-herdsman debug logging.
**DONE. I CREATE THIS ISSUE**
If device joins with 0x000000000000000 as ieeeAddress (you will see: Starting interview of '0x0000000000000000' in the Zigbee2MQTT log) your CC253X might be broken. See issue #2761.
**IT'S NOT THE CASE. MY COORDENATOR IS CONBEE II**
In case the device is a bulb, try resetting it through Touchlink
**IT'S NOT THE CASE. IS A DOOR SENSOR**
Try pairing close to a bulb (light) router instead of the coordinator.
** I HAVEN'T TRIED YET. I WILL TEST TODAY  **

Thanks

jocamane, 2022-01-26:
Try pairing close to a bulb (light) router instead of the coordinator.
**Test and don't work**

Koenkk, 2022-01-27:
Could you provide the herdsman debug log when pairing this device?

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

jocamane, 2022-01-27:
sorry, but i don't understand.
I start my z2m with gui in home assistant.
in the instructions speak of npm start.
I don't understand how to do it. sorry 
help me

jocamane, 2022-01-27:
and when i see the log ?

thanks

jocamane, 2022-01-27:
I tried to do it but when I got to the end I couldn't do the "docker ps" to know what the container ID is.
gives error: docker : command not found

jocamane, 2022-01-27:
2022-01-27T20:20:19.346Z zigbee-herdsman:controller:log Interview '0x00124b001b2f46c3' start
2022-01-27T20:20:19.515Z zigbee-herdsman:controller:device:log Interview - start device '0x00124b001b2f46c3'
2022-01-27T20:20:19.521Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x2f67 EP:0 SeqNr. 83 request id: 19
2022-01-27T20:20:19.522Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2022-01-27T20:20:19.522Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2022-01-27T20:20:19.522Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2022-01-27T20:20:19.522Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2022-01-27T20:20:19.523Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2022-01-27T20:20:19.523Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2022-01-27T20:20:19.527Z zigbee-herdsman:deconz:driver query aps data indication
2022-01-27T20:20:19.532Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 19 status: 0
2022-01-27T20:20:19.533Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2022-01-27T20:20:19.533Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 84
2022-01-27T20:20:19.535Z zigbee-herdsman:deconz:driver query aps data indication
2022-01-27T20:20:19.538Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 84 srcAddr: 0x2f67 destAddr: 0x0 profile id: 0x104 cluster id: 0x1 lqi: 255
2022-01-27T20:20:19.538Z zigbee-herdsman:deconz:frameParser response payload: 24,4,10,32,0,32,30
2022-01-27T20:20:19.540Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":4,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":32,"dataType":32,"attrData":30}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":12135,"endpoint":1,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}'
2022-01-27T20:20:19.540Z zigbee-herdsman:controller:log 'zcl' data is from unknown endpoint '1' from device with network address '12135', creating it...
2022-01-27T20:20:19.547Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2022-01-27T20:20:19.548Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 85
2022-01-27T20:20:19.548Z zigbee-herdsman:deconz:driver query aps data indication
2022-01-27T20:20:19.550Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 85 srcAddr: 0x2f67 destAddr: 0x0 profile id: 0x104 cluster id: 0x1 lqi: 255
2022-01-27T20:20:19.550Z zigbee-herdsman:deconz:frameParser response payload: 24,5,10,33,0,32,120
2022-01-27T20:20:19.552Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":5,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":33,"dataType":32,"attrData":120}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":12135,"endpoint":1,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}'
2022-01-27T20:20:19.555Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2022-01-27T20:20:19.556Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 86
2022-01-27T20:20:19.557Z zigbee-herdsman:deconz:driver query aps data indication
2022-01-27T20:20:19.560Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 86 srcAddr: 0x2f67 destAddr: 0x0 profile id: 0x104 cluster id: 0x1 lqi: 255
2022-01-27T20:20:19.560Z zigbee-herdsman:deconz:frameParser response payload: 24,7,10,33,0,32,120
2022-01-27T20:20:19.562Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":7,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":33,"dataType":32,"attrData":120}],"Command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}},"address":12135,"endpoint":1,"linkquality":255,"groupID":null,"wasBroadcast":false,"destinationEndpoint":1}'
2022-01-27T20:20:19.563Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2022-01-27T20:20:19.564Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 87
2022-01-27T20:20:20.196Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
[... cut ...]
2022-01-27T20:21:04.944Z zigbee-herdsman:deconz:frameParser device state: 10100010
2022-01-27T20:21:04.944Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2022-01-27T20:21:14.944Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 110
2022-01-27T20:21:14.947Z zigbee-herdsman:deconz:frameParser device state: 10100010
2022-01-27T20:21:14.948Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1

Koenkk, 2022-01-28:
These are the correct logs indeed. Could you provide them starting from the point where you factory reset the device until you see interview failed in the z2 logs?

jocamane, 2022-01-28:
sorry. this logs is only choise i can copy paste to you in screen.
if it's not everything, it was because in the meantime it passed and I couldn't copy it.
I can't extract the log to file.

jocamane, 2022-02-12:
help me

STKI55, 2022-03-03:
Hi there, 
I also got nine of the TS0203 devices. They work nicely at the tuya gateway but with zigbee2mqtt all values return the state N/A except the linkquality. I tested with 2 of them. Both with the same result.

[image]
[image]

How can I support you fixing this issue? Don't want to buy new ones :)

STKI55, 2022-03-04:
I have to update my last post. It was necessary to hold the pairing button a lot more than 3 seconds for normal pairing. After holding it for about 10 seconds a kind of factory reset started. Now I get some more values but not everything.

One sensor give me the correct battery level but not more information. 
[image]

The other one return all information exclude the battery level.
[image]

Quite hard for me to understand what these sensors are doing and why there's a different.

jocamane, 2022-04-04:
Koenkk, you can help again ?
Thanks

Koenkk, 2022-04-04:
@jocamane can you try with the following external converter?

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0203'],
    model: 'TS0203',
    vendor: 'TuYa',
    description: 'Door sensor CUSTOM',
    fromZigbee: [fz.ias_contact_alarm_1, fz.battery, fz.ignore_basic_report, fz.ias_contact_alarm_1_report],
    toZigbee: [],
    exposes: [e.contact(), e.battery_low(), e.tamper(), e.battery(), e.battery_voltage()],
    whiteLabel: [{vendor: 'CR Smart Home', model: 'TS0203'}],
    configure: async (device, coordinatorEndpoint, logger) => {
        try {
            const endpoint = device.getEndpoint(1);
            await endpoint.read('genBasic', ['manufacturerName', 'zclVersion', 'appVersion', 'modelId', 'powerSource', 0xfffe]);
            await reporting.bind(endpoint, coordinatorEndpoint, ['genPowerCfg']);
            await reporting.batteryPercentageRemaining(endpoint);
            await reporting.batteryVoltage(endpoint);
        } catch (error) {/* Fails for some*/}
    },
};

module.exports = definition;

```

jocamane, 2022-04-04:
Hi,
No don't work

[image]

I try 4 sensors and all don't work

jocamane, 2022-04-04:
Another issue.
I noticed however that the "allow all" button when it is pressed and goes to "Disable Join" the seconds go by very fast .
it has nothing to do with this issue but it wasn't like that in the other version of z2m

Koenkk, 2022-04-05:
@jocamane it seems something went wrong during the interview. Could you try to force remove the device, restart zigbee2mqtt and pair the device again? If it still fails please provide me the herdsman debug log when pairing this device, "Interview completed" should be "True". 

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable the herdsman debug logging. Note that this is only logged to STDOUT and not to log files.

jocamane, 2022-04-07:
[New Text Document (2).txt](https://github.com/Koenkk/zigbee2mqtt/files/8444270/New.Text.Document.2.txt)

jocamane, 2022-04-07:
Zigbee2MQTT, even without having the "permit all" option active, is adding devices, that is, it is not possible to disable the option to add devices.

[image]

but add devices

jocamane, 2022-04-07:
if you press "permit join" the seconds go by very fast

jocamane, 2022-04-07:
and now you see anything ?
[New Text Document (2).txt](https://github.com/Koenkk/zigbee2mqtt/files/8444745/New.Text.Document.2.txt)

thanks

Koenkk, 2022-04-07:
Yes but somehow nothing is happening, can you try to:
- make sure you are running zigbee2mqtt 1.25.0
- force remove the device through the frontend
- restart zigbee2mqtt
- try pairing the device close to the coordinator
- provide the log

jocamane, 2022-04-07:
Yes of course, but for that I need a way for the log to be saved in a file automatically. if it goes to the foot of the coordinator, I can't copy/paste the log because I'm away from the pc

jocamane, 2022-04-07:
[image]

jocamane, 2022-04-07:
This procedure is what I do every time I try again. I've also tested it with the device leaning against the conbee. I'm about 5 meters from the conbee on most tests

closed 2022-05-16 by github-actions[bot]

soubhik-khan, 2022-11-01:
did anyone got this sensor working? I'm seeing "N/A" for "contact"!

jocamane, 2022-11-01:
No, sensors don't work. I have 8 sensors in this situation